chipsalliance/firrtl #622: `Can't emit firrtl.ir.DoPrim as PrimOp argument`

donggyukim, 2017-07-03:
Is there any reason not to allow it here?
https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/Emitter.scala#L285

There a couple of issues we can think about:
* This makes it harder to write a pass because we need to create a separate node for each `DoPrim`.
* This should be eventually gone for more readable Verilog. I think one way to increase readability is to inline `DoPrim`s in a single line.

donggyukim, 2017-07-03:
To continue, the following is what I want to emit:
```
circuit name_of_sram_module :
  module name_of_sram_module :
    input clock : Clock
    input RW0A : UInt<11>
    input RW0I : UInt<8>
    output RW0O : UInt<8>
    input RW0E : UInt<1>
    input RW0W : UInt<1>
    input RW0M : UInt<1>

    inst mem_0_0 of vendor_sram
    mem_0_0.clock <= clock
    mem_0_0.RW0A <= RW0A
    node RW0O_0_0 = bits(mem_0_0.RW0O, 7, 0)
    mem_0_0.RW0I <= bits(RW0I, 7, 0)
    mem_0_0.RW0M <= bits(RW0M, 0, 0)
    mem_0_0.RW0W <= and(RW0W, eq(bits(RW0A, 10, 10), UInt<1>("h0")))
    mem_0_0.RW0E <= and(RW0E, eq(bits(RW0A, 10, 10), UInt<1>("h0")))
    inst mem_1_0 of vendor_sram
    mem_1_0.clock <= clock
    mem_1_0.RW0A <= RW0A
    node RW0O_1_0 = bits(mem_1_0.RW0O, 7, 0)
    mem_1_0.RW0I <= bits(RW0I, 7, 0)
    mem_1_0.RW0M <= bits(RW0M, 0, 0)
    mem_1_0.RW0W <= and(RW0W, eq(bits(RW0A, 10, 10), UInt<1>("h1")))
    mem_1_0.RW0E <= and(RW0E, eq(bits(RW0A, 10, 10), UInt<1>("h1")))
    RW0O <= mux(eq(bits(RW0A, 10, 10), UInt<1>("h0")), RW0O_0_0, mux(eq(bits(RW0A, 10, 10), UInt<1>("h1")), RW0O_1_0, UInt<1>("h0")))

  extmodule vendor_sram :
    input clock : Clock
    input RW0A : UInt<10>
    input RW0I : UInt<8>
    output RW0O : UInt<8>
    input RW0E : UInt<1>
    input RW0W : UInt<1>
    input RW0M : UInt<1>

    defname =
```

However, this is rejected by the current Verilog emitter. To make it work, I should have the following instead:
```
circuit name_of_sram_module :
  module name_of_sram_module :
    input clock : Clock
    input RW0A : UInt<11>
    input RW0I : UInt<8>
    output RW0O : UInt<8>
    input RW0E : UInt<1>
    input RW0W : UInt<1>
    input RW0M : UInt<1>

    inst mem_0_0 of vendor_sram
    mem_0_0.clock <= clock
    mem_0_0.RW0A <= RW0A
    node _GEN_0 = bits(mem_0_0.RW0O, 7, 0)
    node RW0O_0_0 = _GEN_0
    node _GEN_1 = bits(RW0I, 7, 0)
    mem_0_0.RW0I <= _GEN_1
    node _GEN_2 = bits(RW0M, 0, 0)
    mem_0_0.RW0M <= _GEN_2
    node _GEN_3 = bits(RW0A, 10, 10)
    node _GEN_4 = eq(_GEN_3, UInt<1>("h0"))
    node _GEN_5 = and(RW0W, _GEN_4)
    mem_0_0.RW0W <= _GEN_5
    node _GEN_6 = bits(RW0A, 10, 10)
    node _GEN_7 = eq(_GEN_6, UInt<1>("h0"))
    node _GEN_8 = and(RW0E, _GEN_7)
    mem_0_0.RW0E <= _GEN_8
    inst mem_1_0 of vendor_sram
    mem_1_0.clock <= clock
    mem_1_0.RW0A <= RW0A
    node _GEN_9 = bits(mem_1_0.RW0O, 7, 0)
    node RW0O_1_0 = _GEN_9
    node _GEN_10 = bits(RW0I, 7, 0)
    mem_1_0.RW0I <= _GEN_10
    node _GEN_11 = bits(RW0M, 0, 0)
    mem_1_0.RW0M <= _GEN_11
    node _GEN_12 = bits(RW0A, 10, 10)
    node _GEN_13 = eq(_GEN_12, UInt<1>("h1"))
    node _GEN_14 = and(RW0W, _GEN_13)
    mem_1_0.RW0W <= _GEN_14
    node _GEN_15 = bits(RW0A, 10, 10)
    node _GEN_16 = eq(_GEN_15, UInt<1>("h1"))
    node _GEN_17 = and(RW0E, _GEN_16)
    mem_1_0.RW0E <= _GEN_17
    node _GEN_18 = bits(RW0A, 10, 10)
    node _GEN_19 = eq(_GEN_18, UInt<1>("h0"))
    node _GEN_20 = bits(RW0A, 10, 10)
    node _GEN_21 = eq(_GEN_20, UInt<1>("h1"))
    RW0O <= mux(_GEN_19, RW0O_0_0, mux(_GEN_21, RW0O_1_0, UInt<1>("h0")))

  extmodule vendor_sram :
    input clock : Clock
    input RW0A : UInt<10>
    input RW0I : UInt<8>
    output RW0O : UInt<8>
    input RW0E : UInt<1>
    input RW0W : UInt<1>
    input RW0M : UInt<1>

    defname =
```
which is much less readable.

aswaterman, 2017-07-03:
Emitting multiple operations in a single Verilog assign statement is actually somewhat complicated because of Verilog's unusual rules for signedness and width of compound statements.  Since Firrtl's rules don't precisely match Verilog's rules, you'd need to check whether a Firrtl compound statement's semantics match Verilog's, and conditionally break the compound statement into multiple simple statements if not.

jackkoenig, 2017-07-03:
We *could* come up with a set of compound expressions that are legal to emit in order to emit better Verilog, but this would need to be done very carefully and should involve some amount of formal verification to ensure we don't introduce really subtle bugs.

Per @donggyukim's point that this makes it harder to write passes, not really, just run the SplitExpressions pass after your custom pass.

jackkoenig, 2017-07-03:
Better yet the Verilog emitter should just run SplitExpressions itself

donggyukim, 2017-07-03:
I forgot the existence of `SplitExpressions`, so I ended up with my own... Why is it not included in lowering compilers?

jackkoenig, 2017-07-03:
It's included in LowFirrtlOptimization but definitely should be run in the Verilog Emitter as well

donggyukim, 2017-07-04:
@jackkoenig BTW, am I not supposed to get`SplitExpression` from `getLoweringTransforms(HighForm, LowForm)`? It doesn't return `LowFirrtlOptimization`:
https://github.com/freechipsproject/firrtl/blob/fde7a6a7c8f08984095827cdbe0e3e4df5f1fab8/src/main/scala/firrtl/Compiler.scala#L292

kevintownsend, 2018-03-16:
Have there been any thoughts about using width casting to address this? e.g.:
  wire[2:0] a = 3'H7;
  wire[2:0] b = 3'H1;
  wire[2:0] c = (a + b) >> 1'H1; // Results in 0
  wire[2:0] d = 4'(a + b) >> 1'H1; // Results in 4

jackkoenig, 2018-03-20:
@kevintownsend That is an interesting idea. The main impetus for development on this is that it is the kind of thing that can really bite you if you mess up, and it's hard to test for corner cases that could be problematic. That being said, width casting looks promising.

I've think combining Firrtl circuit fuzzing and LEC could help test a pass that combines expressions in the emitted Verilog (as well as some of the existing transforms, especially optimizations).

albert-magyar, 2020-03-09:
Closed due to non-activity. A ground-up re-examination of the Emitter's mix of context-determined and self-determined widths is planned for the indefinite future.

closed 2020-03-09 by albert-magyar