chipsalliance/firrtl #623: spec: memory syntax differs from chisel3 reality

sorear, 2017-07-03:
Chisel3 represents memories using the `smem` and `cmem` keywords, e.g. https://github.com/freechipsproject/chisel3/blob/aaee58d374dc3f3a088856da13a6a59ecffb2cac/src/main/scala/chisel3/internal/firrtl/Emitter.scala#L58

The FIRRTL spec does not mention  smem or cmem; instead it defines a single `mem` keyword with a `read-latency` that identifies combianational or sequential behavior: https://github.com/freechipsproject/firrtl/blob/b60c31806e9220d63ac2dae98ef4b54c37122491/spec/spec.tex#L755-L766

edwardcwang, 2017-07-04:
In this case, it seems that the spec might be out of date with the current implementation of FIRRTL.

jackkoenig, 2017-07-06:
This is due to a poorly documented mismatch between Chisel IR (Chirrtl) and Firrtl. The purpose of Chirrtl was to help a simple lean Chisel3 frontend use it's looser memory notions with Firrtl. The first thing that the Firrtl compiler does is convert `cmem` and `smem` to `mem` (ie. we run the Chirrtl -> High Firrtl transform).

Solution: Chirrtl as a construct should be moved into chisel3 so that only legal Firrtl is emitted. Since chisel3 depends on firrtl this shouldn't be too difficult to do.

edwardcwang, 2017-07-06:
Or perhaps we should document chirrtl ("lean Chisel3 frontend use it's looser memory notions with Firrtl")?

azidar, 2018-01-03:
This will be fixed with new resolution of #702

albert-magyar, 2020-02-03:
To be fixed in #1210

closed 2020-02-03 by albert-magyar