chipsalliance/firrtl #624: Primop cat requires 2 expression arguments

donggyukim, 2017-07-03:
Can we release this restriction? I believe we can easily write a helper pass to break them down.

jackkoenig, 2017-07-03:
Rather than loosening the restrictions on the IR, I would propose instead providing utilities to help users construct valid IR in a more convenient way.

For example, add a helper method to object Primops.Cat
```scala
def apply(exprs: Expression*): Expression =
  if (exprs.size == 1) exprs.head
  else DoPrim(Cat, Seq(exprs.head, apply(exprs.tail: _*)), Seq.empty, ...)
```

donggyukim, 2017-07-03:
I'm also with that in the short term. Related with #622, we will want to emit it in a single line for readability.

jackkoenig, 2017-07-03:
Cat does seem like a relatively easy primop to combine for emission. As mentioned in #622, the Verilog emitter definitely needs to run split expressions, but then it could run a second "recombine expressions" to merge ones that we know can be safely merged (like Cat)

albert-magyar, 2017-11-09:
There is seqCat in Utils.scala

jackkoenig, 2017-11-09:
I've been thinking about writing apply methods for the various PrimOps to make it easier for pass writers to write stuff like `Cat(a, b, c)` or `Add(in0, in1)`. It's partially done but is also nice to be able to do like `WRef(reg)` and stuff just to make IR construction a bit more concise.

albert-magyar, 2020-03-11:
This has the same idea as #827, so I'll just copy the resolution from there.

>The emission concerns were resolved in #851, so I'm closing this for now, as that resolves the immediate incentive to increase arity in primops.

closed 2020-03-11 by albert-magyar