chipsalliance/firrtl #627: Invalid firrtl from const prop

donggyukim, 2017-07-14:
It seems like const prop generates invalid firrtl with this change: a43486b65506620f89f3e171101353b2dde65351 (e.g. It uses references before declared by node or wire.)

Thus, I couldn't use it before running custom transforms although it's just fine before feeding it into the emitter.

jackkoenig, 2017-07-14:
Can you check if this is fixed by https://github.com/freechipsproject/firrtl/pull/626?

donggyukim, 2017-07-14:
No, it doesn't solve my problem. Here's an example.

Without const prop:
```
    wire hit : UInt<1>
    node _T_287 = or(_T_398, is_alloc_reg)
    ...
    node _T_398 = and(_T_396, _T_397)
    ...
    hit <= _T_398
```

With const prop:
```
    wire _T_398 : UInt<1>
    node _T_287 = or(hit, is_alloc_reg)
    ...
    node hit = and(_T_396, _T_397)
    ...
    _T_398 <= hit
    ...
```

jackkoenig, 2017-07-14:
Good catch, looking into it.

donggyukim, 2017-07-18:
#630

PR #630 (2017-07-17): Do not swap wire names with node names
This resolves #627. Can we merge it to #626?
closed 2017-07-18 by donggyukim