chipsalliance/firrtl #632: [Feature Request] Instancing primitives as modules

leonardt, 2017-07-28:
Convenience feature request.

Right now FIRRTL primitives require a different syntax than instancing Module. Would it be possible to add support for instancing and wiring up primitives like a Module? 

Reasoning: our language infrastructure treats primitives like user-defined modules, thus emitting FIRRTL either requires special casing primitive modules or defining wrapped version of primitives as modules.

Are there any downsides to this approach? Presumably one could write a pass that replaces instances of primitive modules using the function-call-style syntax. Or the other way around (the special primitive syntax would then be considered sugar on top of normal module instantiation). Could this affect any FIRRTL passes?

Example:
```
circuit Example :
  module my_module : 
    input a : UInt
    input b : UInt
    output c : UInt

    inst my_adder of add
    my_adder.in0 <= a
    my_adder.in1 <= b
    c <= my_adder.out
```

edwardcwang, 2017-07-28:
Is it not possible to recognize those primitives and just emit the appropriate firrtl primitive?

leonardt, 2017-07-28:
@edwardcwang 
> Convenience feature request.

> Reasoning: our language infrastructure treats primitives like user-defined modules, thus emitting FIRRTL either requires special casing primitive modules or defining wrapped version of primitives as modules.

It is possible, just inconvenient. More importantly, we have to duplicate this logic (magma and coreir both need to do this to emit FIRRTL). It would be nice to have this done once in FIRRTL and be usable by both systems.

edwardcwang, 2017-07-28:
Well there is an inlining pass https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/Inline.scala - perhaps you could just wrap your primitives in a firrtl module and emit the appropriate annotation?

leonardt, 2017-07-28:
@edwardcwang Yes, we can do that. Again, the issue here isn't that we can't do this, it's that we need to implement this logic twice in two separate systems.

azidar, 2017-07-28:
I agree with @edwardcwang that probably the best way to do it is to emit a module with an inline annotation or to special case it in the Magma/Coreir backends.

The problem with simply added the inst syntax is that FIRRTL primops are polymorphic in their input/output types. Thus, at declaration of ```inst my_adder of add```, we don't know the input or output types of my_adder.

leonardt, 2017-07-28:
@azidar thanks, that reasoning makes sense.

closed 2017-07-28 by leonardt