chipsalliance/firrtl #635: Use adversarial behavior for SRAM models

aswaterman, 2017-08-08:
In two cases, Firrtl's behavioral synchronous-write RAMs are unrealistically optimistic:

- reading & writing the same row on the same cycle should make the read result undefined
- turning off the read-enable should do the same

We should generate simulation models that are a bit more pessimistic.

aswaterman, 2017-08-08:
(BTW, the current workaround we're using is to use ```--repl-seq-mem```, even for behavioral simulation, so we can inject this behavior.)

edwardcwang, 2017-08-08:
Seems like there is a readUnderWrite option for this but isn't implemented at the moment...?

jackkoenig, 2017-08-08:
The issue isn't that readUnderWrite isn't implemented (the current "implementation" is that all read-under-write behavior is undefined). The issue is that this "undefined" behavior has been implemented in an optimistic way that users can accidentally rely on. The undefined behavior should be more pessimistic.