chipsalliance/firrtl #642: Limit length of names/identifiers

aswaterman, 2017-08-24:
Some backend tools limit the lengths of identifiers, module names, etc. to implementation-specific values as small as 1024 chars.  If we eventually add features to Chisel to name modules better (e.g., by placing argument values to parameterized modules in the concretized module name), then Firrtl's Verilog output will eventually run afoul of these limits.

I don't think this needs to be a new command-line argument; the Verilog emitter could unconditionally truncate names to something much more conservative, like 200 chars.

jackkoenig, 2017-11-30:
I think 200 characters is pretty reasonable. Do you think there should be an error if a name that is marked `dontTouch` gets truncated? Or perhaps some future annotation "force name" is what people should use for enforcing names in the Verilog if they want to rely on it?

seldridge, 2020-03-31:
Just reiterating Andrew's statement, this is an issue of aligning with the Verilog spec. Verilog 2005 (we emit Verilog 2001, but I couldn't find a copy of that spec) has a minimum limit of 1024: 

> Implementations may set a limit on the maximum length of identifiers, but the limit shall be at least
1024 characters.

However, tools may have other limits, so configurable is likely best.