chipsalliance/firrtl #647: Use Lowest Common Ancestor for Wiring Transform (Relax Explicit Top Requirement?)

seldridge, 2017-08-30:
I'm reading through [`WiringTransform`](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/wiring/WiringTransform.scala) and it's unclear to me why this requires that the user provides a [`TopAnnotation`](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/wiring/WiringTransform.scala#L41). This currently creates one problem and one annoyance:
  * __Problem__: Incorrectly specifying the `TopAnnotation` as being higher than the lowest common ancestor (LCA) of sinks and sources will result in the addition of source signals going all the way up to the specified top
  * __Annoyance__: The user has to specify the `TopAnnotation` correctly as the LCA. The user should know what this is, but I've always envisioned this type of pass as computing it for you and allowing you to place sinks and sources arbitrarily.

__Proposed Solution__: 
  1. Remove the requirement on a top annotation being specified
  2. Construct all data structures from the circuit-level top module
  3. Compute the [`sharedParent`](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/wiring/WiringUtils.scala#L115) as the LCA when running [`setFields`](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/wiring/WiringUtils.scala#L104) and remove the `setSharedParent` method

__Questions__:
  * Is the exposure of the source all the way up to the top and not using the LCA a bug or a feature? 
  * Is there a reasoning why the top is expected to be specified explicitly? The only thing that comes to mind here is that this is intended to solve the problem of multiple instances of a source existing in the hierarchy and the explicit top module designation is used for scoping (e.g., you have a dual core rocket and there are pins that need to be wired internal to both that are named the same thing). In this case, I think it's equivalent to say that all sinks should be connected to all named sources via their respective LCAs.

I can get a PR together. The modifications for LCA in `3` are easy, but doing doing the `1` and `2` changes are slightly more involved.

seldridge, 2017-08-30:
In better explanation: fdb9f562 provides a test that demonstrates the problem. This checks that specifying the `TopAnnotation` as the LCA or as one module higher than the LCA produce equivalent output. 

1416a837 includes a fix that computes the LCA and gets this test to pass. 

I'll wait on some guidance on whether or not the `TopAnnotation` is actually needed before further updates to the PR (#648).

azidar, 2017-08-30:
I don't think TopAnnotation is actually needed, it was just a hack to not have the wiring pass to go all the way to the top. In general, I think this pass should be rewritten to use the `InstanceGraph` library, as this pass was written before that. However, if you don't want to make a huge change, I'd be happy with minor modifications to make it better.

seldridge, 2017-08-30:
I can give it a shot with `InstanceGraph`.

Removing `TopAnnotation` smells right to me. However, this will invalidate a lot of the tests that are intentionally constructed to hit corner cases involving multiple module instantiations with multiple source instantiations that are ignored due to scoping via `TopAnnotation`. Without `TopAnnotation` all the source to sink connections are assumed to be greedily determined based on their LCA (to hit what I think is the correct solution to a dual core rocket with internal sources and sinks).

closed 2018-01-15 by jackkoenig