chipsalliance/firrtl #654: Instance Graph Merges Structurally Equivalent Instances

seldridge, 2017-09-07:
### Problem
Instance Graph merges multiple instances with the same `WDefInstance(info, name, module, tpe)` signature. This stems from the fact that `WDefInstance` (a case classes) is using structural comparison when being added to a `Set` (which is arguably how it should work for `WIR`/`IR` equality!).

### Example

This is a simplified version of the [`"Wiring from r to X"` test of `WiringTests.scala`](https://github.com/freechipsproject/firrtl/blob/master/src/test/scala/firrtlTests/WiringTests.scala#L36).

__Circuit with two instances of `D` named `d`__
```firrtl
circuit Top :
  module Top :
    input clock: Clock
    inst a of A
    a.clock <= clock
  module A :
    input clock: Clock
    inst b of B
    b.clock <= clock
    inst d of D      @ 'd of D'
    d.clock <= clock
  module B :
    input clock: Clock
    inst c of C
    c.clock <= clock
    inst d of D      @ 'd of D' conflicts with above
    d.clock <= clock
  module C :
    input clock: Clock
    reg r: UInt<5>, clock
  module D :
    input clock: Clock
```

This produces the following edges:
```
[info]   (c,C) -> Set()
[info]   (a,A) -> Set((d,D), (b,B))
[info]   (d,D) -> Set()
[info]   (Top,Top) -> Set((a,A))
[info]   (b,B) -> Set((d,D), (c,C))
```

Both `a` and `b` are connected to `d`. While not incorrect, this breaks how I expected Instance Graph to work and gives you something between an Instance Graph and a Module Graph.

__Circuit with two instances of `D` with different names__
```
circuit Top :
  module Top :
    input clock: Clock
    inst a of A
    a.clock <= clock
  module A :
    input clock: Clock
    inst b of B
    b.clock <= clock
    inst d1 of D     @ renaming d to d1 produces the expected result
    d1.clock <= clock
  module B :
    input clock: Clock
    inst c of C
    c.clock <= clock
    inst d of D
    d.clock <= clock
  module C :
    input clock: Clock
    reg r: UInt<5>, clock
  module D :
    input clock: Clock
```

This produces the following edges:
```
[info]   (c,C) -> Set()
[info]   (a,A) -> Set((d1,D), (b,B))
[info]   (d1,D) -> Set()
[info]   (d,D) -> Set()
[info]   (Top,Top) -> Set((a,A))
[info]   (b,B) -> Set((d,D), (c,C))
```

### Possible Solutions
The Inference Graph needs to store unique things. These can be meaningful or not. Instance Graph could just store instance IDs and have some way to dereference what they mean (e.g., `Int` with a companion `Map`, tuples of `(Int, WDefInstance)`, or some one-off case class).

It seems wrong that `WDefInstance` and `WIR`/`IR` types should use reference equality. A change like this will break everything... and just make things very difficult to work with.

Uniquification via the `info` attribute is possible, but seems very kludgy.

As I'm already neck-deep in mucking around with this, I'm happy to get a PR together (possibly slowly), but some guidance on the approach would be great.

albert-magyar, 2017-09-08:
While I'd definitely be open to changing it, this is intentional.

The basic rationale: ultimately, referring to a specific instance in the full circuit hierarchy (say Top.a.d) will require a _path_ to uniquely identify it. This is true whether structurally-equivalent instances collide or not, since a module can be instantiated multiple times. Given that this fact necessitates remembering paths for most use cases, the instances `Top.a.d` and `Top.a.b.d` would be differentiated in that manner.

seldridge, 2017-09-08:
Ah, okay... Thanks for the explanation. That rationale makes sense. I had incorrectly assumed that the Instance Graph must be a tree or should be a tree. 

I see your point about structurally-equivalent instances and that hadn't occurred to me---the underlying issue is that `WDefInstance` will never be a unique identifier. I forget that `WDefInstance` is really a language construct and has no implied connection to an instance identifier (unless somebody wanted to write their own data structure to record this).

I think that it is cleaner to keep it the way you have it as that doesn't imply some other semantic that differs from what `WDefInstance` is expressing.

I'll leave this open while I verify that there's nothing else here that prevents what I'm trying to do with the wiring refactor.

seldridge, 2017-09-12:
Possibly another point of confusion on my part... is the intent of `pathsInDAG` to produce a flattened hierarchy of instance definitions or actual instances? In other words, is `pathsInDAG` supposed to flatten a digraph into a tree?

The comment for `pathsInDAG` led me to think yes ("return a Map[T,Seq[Seq[T]]] where the value associated with v is the Seq of all paths from start to v"), but nodes are defined as visited based on their instance definition and not their full path.

What is the intent here?

For reference, the following circuit, similar to above, produces a hierarchy that shows this. It finds multiple paths to `d`, but only a single path to `x1` and `x2`. I'd expect this to either find only one path to every node ("find all paths") or all paths to all nodes ("find all possible paths").

```verilog
circuit Top :
  module Top :
    input clock: Clock
    inst a of A
    a.clock <= clock
  module A :
    input clock: Clock
    inst b of B
    b.clock <= clock
    inst x of X
    x.clock <= clock
    inst d of D
    d.clock <= clock
  module B :
    input clock: Clock
    inst c of C
    c.clock <= clock
    inst d of D
    d.clock <= clock
  module C :
    input clock: Clock
    reg r: UInt<5>, clock
  module D :
    input clock: Clock
    inst x1 of X
    x1.clock <= clock
    inst x2 of X
    x2.clock <= clock
  extmodule X :
    input clock: Clock
```

```
[info] hierarchy:
[info]   - c <- ArrayBuffer(List(Top, a, b, c))
[info]   - a <- ArrayBuffer(List(Top, a))
[info]   - x1 <- ArrayBuffer(List(Top, a, d, x1))
[info]   - x <- ArrayBuffer(List(Top, a, x))
[info]   - x2 <- ArrayBuffer(List(Top, a, d, x2))
[info]   - d <- ArrayBuffer(List(Top, a, b, d), List(Top, a, d))
[info]   - Top <- ArrayBuffer(List(Top))
[info]   - b <- ArrayBuffer(List(Top, a, b))
```

seldridge, 2017-09-26:
Closing this as #655 fixed it.

closed 2017-09-26 by seldridge