chipsalliance/firrtl #669: Tie floating wires to zero bug

jchang0, 2017-09-30:
This featured in introduced in 
https://github.com/freechipsproject/firrtl/pull/605

However, it doesn't seem to apply to bundles where part of the bundle is connected and and part of it is not.
(e.g.
    reg reg_mip : {lip : UInt<1>[48], zero2 : UInt<1>, debug : UInt<1>, zero1 : UInt<1>, rocc : UInt<1>, meip : UInt<1>, heip : UInt<1>, seip : UInt<1>, ueip : UInt<1>, mtip : UInt<1>, htip : UInt<1>, stip : UInt<1>, utip : UInt<1>, msip : UInt<1>, hsip : UInt<1>, ssip : UInt<1>, usip : UInt<1>}, clock @[CSR.scala 275:20]
)

I think it also doesn't work for vectors where part one index is not connected and another index is not.

If this is unexpected, can try to provide example.

jackkoenig, 2017-11-29:
Is this still an issue?

jchang0, 2017-11-29:
Haven't done anything about it, so probably yes.

seldridge, 2018-12-18:
Closing, re-open if it's still an issue.

closed 2018-12-18 by seldridge