chipsalliance/firrtl #678: [question] Custom Transforms Outside of FIRRTL?

seldridge, 2017-10-18:
__Background__:

I have a RoCC that requires the use of custom FIRRTL transforms. I would like to package the custom transforms inside the project for the RoCC as opposed to maintaining a separate branch of FIRRTL with custom transforms included. Clarification: the "project" is an unmanaged source directory and I'm not using `project-template`.

The custom FIRRTL transform is then packaged under Rocket Chip as `firrtl/passes/MyTransform` inside of Rocket Chip. I'm not deviating from the standard Rocket Chip approach of running the FIRRTL compiler directly, i.e., `java -cp firrtl/utils/bin/firrtl.jar`.

__Problem__:

There are orthogonal ways that this transform can be packaged and exposed to FIRRTL. 

_I could use some guidance on the best approach for this._

Orthogonal approaches that I'm aware of:

  * The custom transform can be part of the FIRRTL project and is then available in `firrtl.jar`. This only then requires that the build flow generates the correct `--custom-transform` command line argument for the FIRRTL compiler. This is fine, but it seems to scale badly if I need custom branches of FIRRTL for each RoCC and there's then no way to mix in multiple branches.

  * When calling `firrtl.jar`, additional class paths can be specified via `java -cp firrt.jar:$ROCKET_DIR/target/scala-2.11/firrtl/passes/MyTransform`. This will allow the custom transforms that are not inside `firrtl.jar` to get picked up. Command line `--custom-transform` arguments still need to be generated.

  * (Not thought out) Using a generator in Rocket Chip to get everything working correctly, determine annotations, etc. and then call the FIRRTL compiler here. This may provide the necessary class visibility.

It would seem like a custom class path is the way to go, but there may be something better?

Note: this may be very specific to the specific build flow used by Rocket as opposed to a general FIRRTL question.

colinschmidt, 2017-10-18:
Haven't fully processed this yet but one way we have done this is with https://github.com/ucb-bar/barstools

seldridge, 2017-10-18:
Thanks for the pointer! I've seen this, but hadn't dug into what it's intended purpose was (other than what I assumed to be some way of applying passes for a tape out).

After a quick look, and for the sake of documentation (for non-Berkeley folks/me), `barstools` is similar to, but different from, the last approach above. `barstools` is giving you a wrapper around `firrtl.Driver` that's reads FIRRTL, generates annotations, causes custom transforms to be run, and generates Verilog. This effectively makes it a drop in replacement for `firrtl.jar` that adds functionality. This would likely change the approach of packaging transforms inside disjoint hardware module projects, but that's arguably a logistical/organizational decision.

azidar, 2018-03-09:
I've been thinking about how to simplify adding custom transforms to a FIRRTL compiler, I'll submit a new issue discussing this, then close this one.

seldridge, 2018-12-17:
The general way to handle this is to add to the classpath of the FIRRTL compiler. The library registration that the annotations/options refactor will make this easier to deal with.

closed 2018-12-17 by seldridge