chipsalliance/firrtl #683: Option class-log-level should be allowed more than once

jackkoenig, 2017-11-09:
I think it would be better if I could write:
`-cll firrtl.transforms.CheckCombLoops:Info -cll firrtl.transforms.DeadCodeElimination:Info` instead of having to write `-cll firrtl.transforms.CheckCombLoops:Info,firrtl.transforms.DeadCodeElimination:Info`

chick, 2017-11-09:
I think that would be better to but AFAIK scopt does not support that idiom, I've tried and could not find a way.

jackkoenig, 2017-11-29:
It looks like there's an `unbounded()` option: https://javadoc.io/doc/com.github.scopt/scopt_2.12/3.7.0. I think we should switch to allowing arguments multiples times rather than using commas

seldridge, 2018-06-15:
FWIW, this is marked as `unbounded` in the options refactor of #764/#765. In fact, _every option is marked as unbounded_ for the reason that annotation/option validation only happens when a view is attempted and it seems logical to not do any type of early checking (which is technically what a lack of `unbounded` does).

Note that currently both styles are supported as the option argument for this is `Seq[String]` (which scopt will split on `,` and we further, manually subdivide by `:`) _and this is marked as `unbounded`_.

Is there a convention that we want to adopt here or do we want to support both?

seldridge, 2019-04-26:
Fixed via #1005.

closed 2019-04-26 by seldridge