chipsalliance/firrtl #687: Emission of Verilog Inline Attributes

davidbiancolin, 2017-11-16:
I had a short conversation with Jack over slack and there doesn't seem to be a general way to do this. I'd like to write firrtl passes that add specific Inline Annotations like 
```(* mark_debug = "true" *) wire my_wire;``` or ``` (* ram_style = "block") reg [] my_ram [];```

doofin, 2018-08-02:
I also need this feature,which lets us annotate at chisel and then get those annotations at verilog.What's the current status? Glad to help!

jackkoenig, 2018-08-27:
A bit late, but https://github.com/freechipsproject/firrtl/pull/874 add annotations to let users put comments in the emitted Verilog. During the developers meeting last Friday, there was wide agreement with the approach and I believe we can easily extend that mechanism to support Inline Attributes as well! I don't have an exactly timeline but I believe we will have support for this the next few weeks.

jwright6323, 2019-02-26:
Is this still in the works? Would be nice to have this! cc @colinschmidt

grebe, 2019-02-26:
See also #941

jackkoenig, 2020-10-14:
Implemented in https://github.com/freechipsproject/firrtl/pull/1643

closed 2020-10-14 by jackkoenig