chipsalliance/firrtl #691: "sbt assembly" fails: Java version check failure?

ccelio, 2017-11-30:
I'm trying to build firrtl on a clean OSX machine.

`sbt compile` and `test` work, but `assembly` does not.

Is this a Java version failure? I'm using Java **9.0.1**.

```bash
~/git/firrtl$sbt assembly
[info] Loading project definition from /Users/CCelio/git/firrtl/project
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
[info] Set current project to firrtl (in build file:/Users/CCelio/git/firrtl/)
[info] Compiling 77 Scala sources and 4 Java sources to /Users/CCelio/git/firrtl/target/scala-2.11/classes...
[warn] there were two deprecation warnings; re-run with -deprecation for details
[warn] there were 25 feature warnings; re-run with -feature for details
[warn] two warnings found
[warn] bootstrap class path not set in conjunction with -source 1.7
java.lang.ExceptionInInitializerError
	at scala.collection.parallel.ParIterableLike$class.$init$(ParIterableLike.scala:166)
	at scala.collection.parallel.immutable.ParVector.<init>(ParVector.scala:43)
	at scala.collection.immutable.Vector.par(Vector.scala:88)
	at sbtassembly.Assembly$.assembleMappings(Assembly.scala:181)
	at sbtassembly.Assembly$$anonfun$assembledMappingsTask$1.apply(Assembly.scala:252)
	at sbtassembly.Assembly$$anonfun$assembledMappingsTask$1.apply(Assembly.scala:250)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:237)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:514)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.base/java.lang.Thread.run(Thread.java:844)
Caused by: java.lang.NumberFormatException: Not a version: 9
	at scala.util.PropertiesTrait$class.parts$1(Properties.scala:176)
	at scala.util.PropertiesTrait$class.isJavaAtLeast(Properties.scala:180)
	at scala.util.Properties$.isJavaAtLeast(Properties.scala:16)
	at scala.collection.parallel.package$.getTaskSupport(package.scala:45)
	at scala.collection.parallel.package$.<init>(package.scala:48)
	at scala.collection.parallel.package$.<clinit>(package.scala)
	at scala.collection.parallel.ParIterableLike$class.$init$(ParIterableLike.scala:166)
	at scala.collection.parallel.immutable.ParVector.<init>(ParVector.scala:43)
	at scala.collection.immutable.Vector.par(Vector.scala:88)
	at sbtassembly.Assembly$.assembleMappings(Assembly.scala:181)
	at sbtassembly.Assembly$$anonfun$assembledMappingsTask$1.apply(Assembly.scala:252)
	at sbtassembly.Assembly$$anonfun$assembledMappingsTask$1.apply(Assembly.scala:250)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:237)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:514)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.base/java.lang.Thread.run(Thread.java:844)
[error] (*:assembly::assembledMappings) java.lang.ExceptionInInitializerError
[error] Total time: 31 s, completed Nov 30, 2017, 11:16:36 AM
````

ucbjrl, 2017-11-30:
This appears to be a problem with Scala and JDK 9: [Support JDK 9](https://github.com/scala/scala-dev/issues/139)

jackkoenig, 2017-11-30:
It looks like sbt supports JDK 9 though? https://github.com/sbt/sbt-launcher-package/pull/143

Try building with Scala 2.12 just to see: `+assembly`

Also checkout my branch update-sbt and try with sbt version 1.0.4 (Note: if you're in an interactive sbt session, reloading is **not** sufficient, you need to quit and restart sbt to update the version)

ccelio, 2017-11-30:
I concur. I'm running into similar issues with building rocket-chip elsewhere too:

````bash
CCelio@Kaby:~/git/boom-template/verisim$make
/Library/Developer/CommandLineTools/usr/bin/make -C /Users/CCelio/git/boom-template/rocket-chip/firrtl SBT="java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /Users/CCelio/git/boom-template/rocket-chip/sbt-launch.jar" root_dir=/Users/CCelio/git/boom-template/rocket-chip/firrtl build-scala
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /Users/CCelio/git/boom-template/rocket-chip/sbt-launch.jar "assembly"
Java HotSpot(TM) 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
error: error while loading package, Missing dependency 'object java.lang.Object in compiler mirror', required by /Users/CCelio/.sbt/boot/scala-2.10.6/lib/scala-library.jar(scala/package.class)
error: error while loading package, Missing dependency 'object java.lang.Object in compiler mirror', required by /Users/CCelio/.sbt/boot/scala-2.10.6/lib/scala-library.jar(scala/runtime/package.class)
scala.reflect.internal.MissingRequirementError: object java.lang.Object in compiler mirror not found.
````

This is unfortunately an incredibly alarming issue for attempting to evangelize Chisel. :( 

Since this is almost certainly out of our court, I think it is important for our READMEs to reflect the appropriate workaround. I'll try installing Java 8 and verifying that cleans this up.

ucbjrl, 2017-11-30:
I wonder if we still need the `sbt assembly` for `firrtl`? Does anything try and execute the jar as a standalone application?

ccelio, 2017-11-30:
Thanks Jack, it looks like your branch works.

seldridge, 2017-12-01:
Rocket chip uses the standalone jar. Nothing precludes moving to Chisel's driver handling everything, though.

jackkoenig, 2017-12-01:
@seldridge is correct so long as it is possible for rocket-chip to use the local checkout of Firrtl with it's local checkout of Chisel. We should probably take another look at your work @ucbjrl to enable that.

jackkoenig, 2017-12-20:
chisel, firrtl, and rocket-chip have all updated (on master at least) to sbt 1.0.4, @ccelio can you see if the problem is fixed?

azidar, 2018-01-05:
Assuming this is fixed. Please reopen issue if it isn't.

closed 2018-01-05 by azidar