chipsalliance/firrtl #695: Negedge-Triggered Things (memories, registers, etc.)

seldridge, 2017-12-13:
Tracking issue related to a discussion of adding support for negative edge triggered things in FIRRTL. Motivated by @ccelio's Chisel3 issue: https://github.com/freechipsproject/chisel3/issues/736.

Roughly, I see this as:
  - [ ] Extending clocked IR statements (`DefRegister`, `DefMemory`) to specify "_edgedness_"
  - [ ] Define edgedness, assumedly as `Set(Posedge, Negedge, PosNegEdge)`
  - [ ] Extending the Verilog emitter to use the appropriate edgedness when generating statements
  - [ ] Update the specification document to include edgedness

Couple of questions:
  - Thoughts on the approach?
  - Is this excluded from the FIRRTL specification intentionally and is there some motivation for not addressing negative edge clocking?

Possible problems/alternative solutions:
  - The original intent is to only provide the capability to emit negedge triggered SRAMs. This may be better handled as just some tweaks to the `ReplSeqMem` transform or a similar transform that corrects clocking. However, by my rough understanding a full solution would likely require extending the IR.
  - This could be viewed as entirely out of scope and should be handled via blackboxing (a view that would likely hinder adoption).

azidar, 2018-01-05:
A few more thoughts to consider. Adding more to the IR requires all transformations to need to consider it. For example, all MIDAS transformations would have to address Negedge, or explicitly disallow it. Before we add more burden to transformation writers, I think its important to understand (1) how common this feature really is in practice, (2) whether the current practice is *best* practice and not an anti-pattern, and (3) how difficult it is to support in the toolchain (testers, interpreters). We shouldn't half-support a feature.

azidar, 2018-01-23:
Any updates on examples? We discuss this during our research meeting, and came up with the following that would need negedge support, as well as explicit latches.

- JTAG block
- Physical design of register file
- DDR signals
- SRAM (BOOM)

seldridge, 2018-01-23:
Maybe @ccelio has some thoughts on additional examples.

This seems to cover it. I'd be more inclined to limit the motivation to two groups: "negedge latches" and "negedge memories". I think everything else falls into one of these two categories (which aligns with the IR).

No updates, nor have I had a chance to work out a sketch of a PR. This isn't something that's affecting me in any way currently.

Towards your points above:
  1) This seems to indicate that it's not that common other than being an available HDL design primitive (in Verilog). Most of the examples are what I'd consider "hard IP" which could be reasonably relegated to one-time design blackboxes.
  2) What are you referring to here as it relates to the current practice?
  3) This is the crux... If we don't want to support it in Chisel (explicitly) and in the testers/interpreters, then it is a defacto blackbox (or a FIRRTL pass that mucks with the clocks of specific components)

Was their some consensus in the meeting as whether or not this is worth supporting or is more data needed on the examples side?

azidar, 2018-01-24:
I believe the consensus of the meeting was to see what the needs of the community are, and let that drive development. I agree with your answer to (1), though I do think it would be useful to directly represent this in FIRRTL. My thoughts behind (2) were some things like latch-based designs are generally not a good idea, so by directly supporting them, we could enable bad design. However, latch-based designs can have huge area/timing improvements, so I was just highlighting this tradeoff. Regarding (3), it might require a refactor of the interpreter, perhaps @chick has some thoughts?

sequencer, 2019-07-16:
So may I ask the question about
> Define edgedness, assumedly as Set(Posedge, Negedge, PosNegEdge)
will this being supported by firrtl, or will firrtl accept PR like this?
I'm work on the `clkgen` which needs latch, and I wanna implement it with pure chisel3
I know this will introduce a horrible impact on the verification system, but how about adding it to firrtl firstly?