chipsalliance/firrtl #698: sbt test failing with no file found error

Sairam-Ganti, 2017-12-15:
Hi, 
So I am new to firrtl and tried to check out the code and get started. `sbt compile` works perfectly, but `sbt test` fails for 27 cases with errors similar to the following.

`Custom Transforms
[info] - should be able to introduce high firrtl *** FAILED ***
[info]   java.io.IOException: Cannot run program "./VCustomTransform" (in direct                                                                                                                ory "test_run_dir\CustomTransform\201712160035366833863110039663112"): CreatePro                                                                                                                cess error=2, The system cannot find the file specified
[info]   at java.lang.ProcessBuilder.start(Unknown Source)
[info]   at scala.sys.process.ProcessBuilderImpl$Simple.run(ProcessBuilderImpl.s                                                                                                                cala:69)
[info]   at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.run(ProcessBuil                                                                                                                derImpl.scala:100)
[info]   at scala.sys.process.ProcessBuilderImpl$AbstractBuilder$$anonfun$runBuf                                                                                                                fered$1.apply(ProcessBuilderImpl.scala:148)
[info]   at scala.sys.process.ProcessBuilderImpl$AbstractBuilder$$anonfun$runBuf                                                                                                                fered$1.apply(ProcessBuilderImpl.scala:148)
[info]   at scala.sys.process.ProcessLogger$$anon$1.buffer(ProcessLogger.scala:9                                                                                                                9)
[info]   at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.runBuffered(Pro                                                                                                                cessBuilderImpl.scala:148)
[info]   at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.$bang(ProcessBu                                                                                                                ilderImpl.scala:114)
[info]   at firrtl.util.BackendCompilationUtilities$class.executeExpectingFailur                                                                                                                e(BackendCompilationUtilities.scala:136)
[info]   at firrtlTests.FirrtlFlatSpec.executeExpectingFailure(FirrtlSpec.scala:                                                                                                                182)
[info]   ...
[info]   Cause: java.io.IOException: CreateProcess error=2, The system cannot fi                                                                                                                nd the file specified
[info]   at java.lang.ProcessImpl.create(Native Method)
[info]   at java.lang.ProcessImpl.<init>(Unknown Source)
[info]   at java.lang.ProcessImpl.start(Unknown Source)
[info]   at java.lang.ProcessBuilder.start(Unknown Source)
[info]   at scala.sys.process.ProcessBuilderImpl$Simple.run(ProcessBuilderImpl.s                                                                                                                cala:69)
[info]   at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.run(ProcessBuil                                                                                                                derImpl.scala:100)
[info]   at scala.sys.process.ProcessBuilderImpl$AbstractBuilder$$anonfun$runBuf                                                                                                                fered$1.apply(ProcessBuilderImpl.scala:148)
[info]   at scala.sys.process.ProcessBuilderImpl$AbstractBuilder$$anonfun$runBuf                                                                                                                fered$1.apply(ProcessBuilderImpl.scala:148)
[info]   at scala.sys.process.ProcessLogger$$anon$1.buffer(ProcessLogger.scala:9                                                                                                                9)
[info]   at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.runBuffered(Pro                                                                                                                cessBuilderImpl.scala:148)
[info]   ...
%Error: D:\Dev:72: syntax error, unexpected INTEGER NUMBER, expecting IDENTIFIER
%Error: D:\Dev:83: syntax error, unexpected INTEGER NUMBER, expecting IDENTIFIER
%Error: Exiting due to 2 error(s) `

I've tried to google for a solution, but I've gotten nowhere so far.
Attaching an error log as well.
[errorLog.txt](https://github.com/freechipsproject/firrtl/files/1563722/errorLog.txt)

azidar, 2017-12-15:
Do you have verilator installed on your machine?

azidar, 2017-12-15:
What's the version of verilator?

Sairam-Ganti, 2017-12-15:
Verilator 3.841 2012-09-03 rev verilator_3_840-22-g39724eb

azidar, 2017-12-15:
Yeah, that's likely your problem. The version I'm using is Verilator 3.886 2016-07-30 rev verilator_3_884-13-gb4a7e46. Try updating it?

ucbjrl, 2017-12-15:
It looks like @Sairam-Ganti is running under cygwin/mingw on Windows and using a native version of sbt which is using DOS paths. I suspect the cygwin/mingw shell is treating the `\` path delimiters as quote characters and altering the strings passed to  verilator:
```bash
verilator --cc D:\Dev Environment\firrtl\firrtl\test_run_dir\NestedSubAccessTester\201712160035327857653271670478765/NestedSubAccessTester.v --assert -Wno-fatal -Wno-WIDTH -Wno-STMTDLY --trace -O1 --top-module NestedSubAccessTester +define+TOP_TYPE=VNestedSubAccessTester +define+PRINTF_COND=!NestedSubAccessTester.reset +define+STOP_COND=!NestedSubAccessTester.reset -CFLAGS -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VNestedSubAccessTester -DVL_USER_FINISH -include VNestedSubAccessTester.h -Mdir D:\Dev Environment\firrtl\firrtl\test_run_dir\NestedSubAccessTester\201712160035327857653271670478765 --exe D:\Dev Environment\firrtl\firrtl\test_run_dir\NestedSubAccessTester\201712160035327857653271670478765\top.cpp
```

Additionally, the spaces in the paths will cause problems if they aren't quoted.

Sairam-Ganti, 2017-12-15:
@azidar  I am trying to update, will let you know if that works
@ucbjrl  Should I update directory names so that they don't have spaces? There is still the problem with `\` right?

ucbjrl, 2017-12-15:
And I take that back. This error indicates the unquoted space is a problem:
```bash
%Error: D:\Dev:25: syntax error, unexpected INTEGER NUMBER, expecting IDENTIFIER
```

I suspect the errorLog.txt file was created by copying the screen. This would explain the crazy white space characters in the log.

Try the following (assuming UNIX shell redirection works):
```bash
$ sbt test >& sbttest.txt
```
and attach that file.

Sairam-Ganti, 2017-12-15:
@ucbjrl  I'll try what you suggested and update. Also, I'm on a windows system, if that wasn't apparent. I'm trying to re-do the entire process on a Linux machine now and see if I run into roadblocks there, Will update soon. 
Thanks for all the help.

Sairam-Ganti, 2017-12-17:
@ucbjrl  Hi, the `sbt test` command ran perfectly on a linux machine. I'm thinking that it was an issue with the misinterpreted `\` character and whitespaces like you said on windows.

azidar, 2018-03-09:
Is there any followup to this? @ucbjrl

ucbjrl, 2018-03-09:
Not without seeing the output of:
```bash
$ sbt test >& sbttest.txt
```

seldridge, 2018-12-18:
Closing this due to inactivity. Feel free to reopen if there's still an issue!

closed 2018-12-18 by seldridge