chipsalliance/firrtl #726: Wiring pass for bundle produces invalid access to non-reference

edcote, 2018-01-09:
@azidar @seldridge 

Kindly consider the unit test at the bottom of this report.

The use case is to wire a bundle from module A to module B through parent module T.

I use the off-the-shelf WiringTransform.  For this example, I reduced to the wiring pass itself.

Is this feature supported?

### Error message

~~~~
firrtl.passes.CheckChirrtl$InvalidAccessException: : [module T] Invalid access to non-reference.
firrtl.passes.CheckChirrtl$InvalidAccessException: : [module T] Invalid access to non-reference.
~~~~

Occurs for expression `a.auto_0`, RHS of connect stmt in module T.

### Annotated output of wiring pass

~~~~
circuit T :
  module A :
    input clock : Clock
    input reset : UInt<1>
    output auto : { [..] }
    output auto_0 : { [..] } // NEWLY created only
    auto_0 <= auto

  module B :
    input clock : Clock
    input reset : UInt<1>
    input pin : { [..] } // PIN added
 
  module T :
    input clock : Clock
    input reset : UInt<1>
    inst a of A
    inst b of B
    wire auto : { [..] }
    auto <= a.auto_0 // <--- **ERROR**
    b.pin <= auto
~~~~

### Unit test

~~~~
 "Wire a bundle from A to B" in {
    val firrtl = """
circuit T :
  module A :
    input clock : Clock
    input reset : UInt<1>
    output auto : { [..] }

  module B :
    input clock : Clock
    input reset : UInt<1>

  module T :
    input clock : Clock
    input reset : UInt<1>
    inst a of A
    inst b of B
      """

    val wi = new WiringInfo(
      source = "A",
      comp = "auto",
      sinks = Set("B"),
      pin = "pin",
      top = "T"
    )

    val passes = Seq( new Wiring(Seq(wi)) )

    val c = passes.foldLeft(Parser.parse(firrtl)) {
      (c: Circuit, p: Pass) => p.run(c)
    }

    //println(c.serialize)

    val compiler = new HighFirrtlCompiler // this will trigger 
    val writer = new java.io.StringWriter
    compiler.compile(CircuitState(c, ChirrtlForm), writer)
  }
~~~~

edcote, 2018-01-09:
Btw, is there a mailing list for firrtl users?

seldridge, 2018-01-09:
I think you may just need to clean the code up following running `Wiring`. This, by itself, will not emit valid MidFirrtl and, for the version that you're looking at, [needs a couple of passes to clean it up](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/wiring/WiringTransform.scala#L67). The pending 1.1.0 refactor needs a few more...

You may try:
```scala
val passes = Seq( 
  new Wiring(Seq(wi)),
  InferTypes,
  ResolveKinds,
  ResolveGenders) )
```

> Btw, is there a mailing list for firrtl users?

Not known to me. Likely the chisel-users Google group would seem to be be the best place, though there's little activity there related to that. There is a [gitter chat](https://gitter.im/freechipsproject/firrtl) which should show up on the README once a PR merges.

seldridge, 2018-01-09:
Looking at this more... the specific problem is that `Wiring` is going to run `ToWorkingIR`. Not a Wiring transform problem, but an issue of trying to pass WIR nodes into `CheckChirrtl`. You should be able to get the same behavior by just running `ToWorkingIR` on your parsed FIRRTL.

This is, I think, expected behavior. Once you're moving to the level of actually running the Compiler, I believe that the assumption is that you're passing in a non-WIR representation. If you then go to that level and run the compiler, the correct way of driving Wiring (or some other pass) would be with annotations. If you're down in the weeds and linking passes together, then it would be appropriate to run a pass directly.

To get your example to work, you can kludgily strip all WIR nodes by serializing and re-parsing, i.e.:
```scala
compiler.compile(CircuitState(Parser.parse(c.serialize), ChirrtlForm), writer)
```

There may be a better way to handle something like this, but I'm not sure...

Let me know if that does it for you, Ed.

azidar, 2018-01-09:
The plan of record is to remove Chirrtl entirely, and its something I'm currently working on. But yes, if you need to convert from WIR to IR, @seldridge's solution works. We should also have a WorkingIRToIR pass, so if somebody wants to PR that, it would readily be accepted.

There is not a mailing list, though there has been some in-person discussions about starting one. We just created a public gitter channel, so for now feel free to ask questions there! It was specifically created to encourage collaboration with non-Berkeley folks.

@edcote , if @seldridge's solution fixes your bug, feel free to close the issue.

edcote, 2018-01-09:
Thanks for replies. I will follow up before EOD.

edcote, 2018-01-10:
Woot! I know about IR vs. WIR now.  I had previously tried the cleanup passes that you suggested.  No change. Output circuit still had WRef statements.

I tried @seldridge hack and ran the circuit through the compiler.

    val compiler = new MiddleFirrtlCompiler
    val writer = new java.io.StringWriter
    val x = compiler.compile(CircuitState(Parser.parse(c.serialize), ChirrtlForm), writer)

It craps out with: "Expression auto is used as a MALE but can only be used as a FEMALE." on the generated-by-wiring-pass statement shown below.

    module A :
        input clock : Clock
        input reset : UInt<1>
        output auto : { [..] }
        output auto_0 : { [..] }
        auto_0 <= auto //  <-- 

Would you say that the input circuit is invalid?

Can anyone think of a workaround for the use case that I am presenting?

edcote, 2018-01-11:
I opened #728 with a different question.

Would like answer to above question before we close this (i.e., is this use case permitted, is there a workaround, is there a bug, or is an enhancement required?).

seldridge, 2018-01-11:
I have a few things to take of today, but I want to take a look this afternoon. I think this is incorrect usage. 

You're trying to connect an output port _inside_ of module A to something. There are ways that the wiring pass could resolve this (e.g., trace connections backwards to figure out what is connected to this)., but it's trying to directly connect to whatever you specify. One of the following approaches should work:
  - Set the sink at the level of Module T via a.auto
  - Set the sink as the driver of Module A's auto

azidar, 2018-03-09:
@edcote, I'm going to assume @seldridge answered the question and close the issue. Please reopen it if I am mistaken.

closed 2018-03-09 by azidar

edcote, 2018-03-09:
Thanks. Sorry for non followup. Was waiting for annotation changes to go in
before trying again