chipsalliance/firrtl #728: Wiring pass doesn't initialize target pin

edcote, 2018-01-11:
This one is related to #726 but I am creating a new issue because the question is subtly different.

Previously, it seems I was trying to connect an output port of bundle type from one module to another.

In this example, I am trying to connect an uninitialized wire of bundle type from one module to another.

The target (destination) module of the emitted circuit is produced with a pin of bundle type.

Lowering to MidForm gives: `Reference pin is not fully initialized. : pin.dmInner_dmInner_tl_in.e.valid <= VOID` for each element of the bundle.

Question is: What pass am I missing that would eliminate the issue?

Input circuit:

~~~~
circuit T :
  module A :

    input clock : Clock
    input reset : UInt<1>

; let's consider a wire bundle <---
    wire bundle : {flip dmInner_dmInner_tl_in : {a : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<6>, address : UInt<12>, mask : UInt<8>, data : UInt<64>}}, flip b : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<6>, address : UInt<12>, mask : UInt<8>, data : UInt<64>}}, c : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<6>, address : UInt<12>, data : UInt<64>, error : UInt<1>}}, flip d : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<6>, sink : UInt<1>, data : UInt<64>, error : UInt<1>}}, e : {flip ready : UInt<1>, valid : UInt<1>, bits : {sink : UInt<1>}}}, dmOuter_anon_out : {sync : UInt<1>[1]}}
    bundle is invalid

  module B :
    input clock : Clock
    input reset : UInt<1>

  module T :
    input clock : Clock
    input reset : UInt<1>

    inst a of A
    inst b of B

~~~~

Output circuit:

~~~~
circuit T :
  module A :
    input clock : Clock
    input reset : UInt<1>
    output bundle_0 : { flip dmInner_dmInner_tl_in : { a : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<6>, address : UInt<12>, mask : UInt<8>, data : UInt<64>}}, flip b : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<6>, address : UInt<12>, mask : UInt<8>, data : UInt<64>}}, c : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<6>, address : UInt<12>, data : UInt<64>, error : UInt<1>}}, flip d : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<6>, sink : UInt<1>, data : UInt<64>, error : UInt<1>}}, e : { flip ready : UInt<1>, valid : UInt<1>, bits : { sink : UInt<1>}}}, dmOuter_anon_out : { sync : UInt<1>[1]}}
;   ^^^ added by wiring pass <---

    wire bundle : { flip dmInner_dmInner_tl_in : { a : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<6>, address : UInt<12>, mask : UInt<8>, data : UInt<64>}}, flip b : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<6>, address : UInt<12>, mask : UInt<8>, data : UInt<64>}}, c : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<6>, address : UInt<12>, data : UInt<64>, error : UInt<1>}}, flip d : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<6>, sink : UInt<1>, data : UInt<64>, error : UInt<1>}}, e : { flip ready : UInt<1>, valid : UInt<1>, bits : { sink : UInt<1>}}}, dmOuter_anon_out : { sync : UInt<1>[1]}}
    bundle is invalid
    bundle_0 <= bundle
;   ^^^ added by wiring pass <---

  module B :
    input clock : Clock
    input reset : UInt<1>
    input pin : { flip dmInner_dmInner_tl_in : { a : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<6>, address : UInt<12>, mask : UInt<8>, data : UInt<64>}}, flip b : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<6>, address : UInt<12>, mask : UInt<8>, data : UInt<64>}}, c : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<6>, address : UInt<12>, data : UInt<64>, error : UInt<1>}}, flip d : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<6>, sink : UInt<1>, data : UInt<64>, error : UInt<1>}}, e : { flip ready : UInt<1>, valid : UInt<1>, bits : { sink : UInt<1>}}}, dmOuter_anon_out : { sync : UInt<1>[1]}}
;   ^^^ added by wiring pass <---

  module T :
    input clock : Clock
    input reset : UInt<1>

    inst a of A
    inst b of B

    wire bundle : { flip dmInner_dmInner_tl_in : { a : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<6>, address : UInt<12>, mask : UInt<8>, data : UInt<64>}}, flip b : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<6>, address : UInt<12>, mask : UInt<8>, data : UInt<64>}}, c : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<6>, address : UInt<12>, data : UInt<64>, error : UInt<1>}}, flip d : { flip ready : UInt<1>, valid : UInt<1>, bits : { opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<6>, sink : UInt<1>, data : UInt<64>, error : UInt<1>}}, e : { flip ready : UInt<1>, valid : UInt<1>, bits : { sink : UInt<1>}}}, dmOuter_anon_out : { sync : UInt<1>[1]}}
    bundle <= a.bundle_0
    b.pin <= bundle
;   ^^^ added by wiring pass <---
~~~~

MiddleFirrtlCompiler complains of the following. The error is on the generated
code.

~~~~
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module B]  Reference pin is not fully initialized.
   : pin.dmInner_dmInner_tl_in.e.valid <= VOID
[..]
~~~~

seldridge, 2018-01-11:
Thanks for reporting this. I should be able to take a look later.

I assume you're using the passes mentioned in #726 (https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/wiring/WiringTransform.scala#L67) and the old wiring transform (not #648)?

There are currently no tests that are using a sink of Aggregate type (only a SubField of an Aggregate). This _may_ be a bug. If it is, I can try and get something in before 1.1.0 closes for the new WiringTransform.

seldridge, 2018-01-11:
Have a fix in place for the new transform and a new test that wires uses an Aggregate source and Module sink. A couple additional passes were needed. 

Note: this is for the _new_ Wiring transform and not the old one. [Full list of required transforms](https://github.com/seldridge/firrtl/blob/1a82218c8a81d5568ce8423ab80adaf540431ece/src/main/scala/firrtl/passes/wiring/WiringTransform.scala#L63).

commit 347cc522 (2018-01-15): WiringTransform Refactor (#648)

Massive refactoring to WiringTransform with the use of a new EulerTour 
class to speed things up via fast least common ancestor (LCA) queries.

Changes include (but are not limited to):

* Use lowest common ancestor when wiring
* Add EulerTour class with naive and Berkman-Vishkin RMQ
* Adds LCA method for Instance Graph
* Enables "Two Sources" using "Top" wiring test as this is now valid
* Remove TopAnnotation from WiringTransform
* Represent WiringTransform sink as `Seq[Named]`
* Remove WiringUtils.countInstances, fix imports
* Support sources under sinks in W [... cut]
closed 2018-01-15 by jackkoenig