chipsalliance/firrtl #729: Firrtl generates verilog with odd variable names (trailing '$')

ben-k, 2018-01-15:
* **Type of issue**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Other enhancement

Firrtl occasionally emits verilog signals with a trailing '$'.  I'm not sure if this is properly a bug, but it's unclear why this occurs, and some commercial verilog parsers don't appreciate the syntax.

I was able to reproduce this on current rocket-chip (8799508).  After building DefaultConfig, you can see examples at line 174566 (`wire [63:0] logic$;` in `AMOALU`) or line 168431 (`wire  default$;` in `PMPChecker`).  I've also observed `reg reg$`, although this doesn't seem to appear in DefaultConfig.

aswaterman, 2018-01-15:
It’s used whenever there’s a name conflict with a Verilog keyword (logic is a SystemVerilog type).

$ is legit, but I’m not surprised that buggy parsers would get this wrong. Maybe we can switch to using _, now that Firrtl supports proper name disambiguation.

jackkoenig, 2018-05-31:
@azidar think we should change the trailing `$` to trailing `_`?

seldridge, 2018-06-21:
Note: `InlineInstances` uses `$` as a delimiter as well and `ClockListTransform` explicitly depends on this being used to encode hierarchy. I'm leaving this as a parameter when constructing `InlineInstances` so that somebody calling `InlineInstances` explicitly can specify the old `$` if needed.

Alternatively, it's very, very useful to have a distinct character that encodes a collapsed hierarchy. If `_` replaces FIRRTL's use of `$` then this overloading gets annoying when looking at FIRRTL or Verilog output. Verilog disallows any other special characters, so short of something quirky (`_o0o_`) no other likely unique combinations are available. With that in mind, there may be some motivation for not acquiescing to buggy parsers... A compromise might be an optional `LowForm => LowForm` transform that will mangle anything with `$` to `_` (with Namespace checking).

@ben-k: do you recall what commercial vendors were buggy?

ben-k, 2018-06-21:
Not certain - I haven't looked at this in a while.  If no one else is running into it then perhaps it was just an internal tool.

If the use of `$` is load-bearing, then an optional Firrtl pass seems like a good solution.

aswaterman, 2018-07-04:
Yeah, many real flows have one-off, hand-written Verilog parsers, sometimes multiple ones per project, each with its own bugs.  It's a design antipattern, to be sure, but we should still aim to be low-friction when (as in this case) it costs us almost nothing.

PR #909 (2018-10-12): Verilog renaming uses "_", works on whole AST
Fixes #729 

This fixes two problems with the current `passes.VerilogRename`:
  1. For #729 , some Verilog parsers (user written or otherwise) may have problems with names containing `$`. The current rename strategy is to blindly append `$` to any keyword that conflicts with a set of Verilog keywords. This changes that to use `_` for name disambiguation.
  2. The existing rename strategy would not rename with respect to namespaces. This only worked before because names with `$` are exceedingly rare.

To address these, this PR generalizes `passes.Ve [... cut]
closed 2018-10-12 by seldridge