chipsalliance/firrtl #737: Deduplication Collisions and Annotations

seldridge, 2018-02-01:
* **Type of issue**
  - [X] Bug report (__maybe...__)
  - [ ] Feature request
  - [ ] Other enhancement

* **If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **If possible, please directly provide the FIRRTL file or a relevant fragment if the file is large**

Consider the following test of `BoringUtils` where `addSource` adds a source and `addSink` adds a sink. Here, I'm trying to change the behavior of `InverterAfterWiring` from a combinational passthrough when given parameter `false` to a combinational inverter when given parameter `true` using the proposed `BoringUtils` annotators of Chisel3. An equivalent test case could be hacked up that is pure Chisel:
```scala
class InverterAfterWiring(wire: Boolean) extends Module with BoringUtils {
  val io = IO(new Bundle {
    val in = Input(Bool())
    val out = Output(Bool())
  })

  val notIn = ~io.in
  io.out := io.in

   if (wire) {
    addSource(notIn, "id")
    addSink(io.out, "id") }
}

class PassthroughAfterWiringTester extends BasicTester {
  val dutOrig = Module(new PassthroughAfterWiring(false))
  val dutWired = Module(new PassthroughAfterWiring(true))

  val (c, done) = Counter(true.B, 3)

  dutOrig.io.in := c(0)
  dutWired.io.in := c(0)
  assert(dutOrig.io.out === dutOrig.io.in)
  assert(dutWired.io.out =/= dutWired.io.in)

  when (done) { stop() }
}

class BoringUtilsSpec extends ChiselFlatSpec with BackendCompilationUtilities {
  it should "connect within a module" in {
    val target = "PassthroughAfterWiringTester"

    val path = createTestDirectory(target)
    val fname = new File(path, target)

    val cppHarness =  new File(path, "top.cpp")
    copyResourceToFile("/chisel3/top.cpp", cppHarness)

    Driver.execute(Array(
      "--target-dir", path.getAbsolutePath),
      () => new PassthroughAfterWiringTester)
    val passed = if ((verilogToCpp(target, path, Seq.empty, cppHarness) #&&
        cppToExe(target, path)).! == 0) {
      executeExpectingSuccess(target, path)
    } else {
      false
    }
    assert(passed)
  }
}
```

  * **What is the current behavior?**

Two instances of this module, instantiated with different parameters will result in deduplication to one of them as deduplication runs before the `WiringTransform`. In this instance, the un-annotated module is deduped to the annotated one resulting in both of them turning into inverters.

  * **What is the expected behavior?**

I would expect that these should not be deduplicated. However, I can see a case where this is user error and this module should be marked manually to not be deduped.

Generally, I would expect that the annotations need to factor into the determination of whether or not two modules are the same. 

  * **Please tell us about your environment:**
    - firrtl: `master`
    - chisel3: https://github.com/freechipsproject/chisel3/pull/718

* **What is the use case for changing the behavior?**

As people start to write annotations that trigger on a parameter, they'll hit this depending on when their transform runs.

* **Impact**
  - [X] no functional change
  - [ ] API addition (no impact on existing code)
  - [ ] API modification
  - [ ] unknown

* **Development Phase**
  - [ ] request
  - [X] proposal

* **Other information** 

One way to fix this would be to include any annotation information in determining dedup collisions. However, this is overly conservative to assume that annotations make two modules different (they may, but the annotation could be an identity transform or something superfluous). However, in almost all cases, any real transform will cause two modules to fail deduplication.

Alternatively, deduplication could be disallowed on annotated modules and run only after these annotations have been successfully deleted.

jackkoenig, 2018-02-02:
It is an interesting question how annotations and deduplication should interact.

On the one hand, it makes sense that things should only be deduplicated if their annotations match in case the annotations are changing behavior. On the other hand, we want to support instance annotations (as opposed to module annotations which is the only kind currently supported) so that we can intentionally dedup things but still have annotations for different instances.

Maybe Module annotations must match for dedup but instance-level don't have to.

This does have implications for how the new annotation API (see https://github.com/freechipsproject/firrtl/pull/721) should work though since the API for Annotation currently does not require mentioning any targets. Perhaps annotations should be required to report what `Seq` or `Set` of `Named` things they target so that we can use that information in deduplication?

seldridge, 2018-12-18:
This should be largely able to be handled by both no dedup annotations and the target work.

closed 2018-12-18 by seldridge