chipsalliance/firrtl #742: DiGraph.BFS doesn't treat the root node as visited

seldridge, 2018-02-13:
* **Type of issue**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Other enhancement

* **If the current behavior is a bug, please provide the steps to reproduce the problem:**

I would expect that `DiGraph.BFS` when run on a cyclic graph should not return its root. The following test case will fail:

```scala
class DiGraphTests extends FirrtlFlatSpec {

  val cyclicGraph = DiGraph(Map(
    "a" -> Set("b","c"),
    "b" -> Set("d"),
    "c" -> Set("d"),
    "d" -> Set("a")))

  "a cyclic graph" should "not report its root node during a BFS" in {
    cyclicGraph.BFS("a").keys should not contain "a"
  }
}
```

The current BFS for this is `Map(b -> a, c -> a, d -> b, a -> d)`.

I'd expect to see `Map(b -> a, c -> a, d -> b)`.

* **What is the use case for changing the behavior?**

The BFS should be reporting visited nodes, not visited edges.

* **Impact**
  - [ ] no functional change
  - [ ] API addition (no impact on existing code)
  - [x] API modification (__technically__)
  - [ ] unknown

* **Development Phase**
  - [ ] request
  - [x] proposal

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

I'll add a PR with a simple fix momentarily.

albert-magyar, 2018-02-13:
I'm not sure that there's a common convention, but there are a couple of things that led to it being the way it is:

- Most pseudocode descriptions in the literature don't initially mark the root
- It's easier to delete the edge that yields a cycle than to add it back

seldridge, 2018-02-13:
Got it. I forget that the implementation decisions here always seem to have a thorough justification...

It sounds like the error is mine: a misunderstanding of how the `BFS` method should behave. I'll close this.

Thanks for the clarification.

closed 2018-02-13 by seldridge