chipsalliance/firrtl #749: In low Firrtl second clock referenced before declared.

chick, 2018-03-02:
This repository's issues are reserved for feature requests and bug reports.

* **Bug Report**
The chisel3 MultiClockSpec ClockDividerTest is converted to low Firrtl, the firrtl is re-arranged so that clock2 is referenced before it is declared.

* **How to reproduce the problem:**

Here is the Chirrtl.
```
;buildInfoPackage: chisel3, version: 3.1-SNAPSHOT, scalaVersion: 2.11.12, sbtVersion: 1.0.4
circuit ClockDividerTest : 
  module ClockDividerTest : 
    input clock : Clock
    input reset : UInt<1>
    output io : {}
    
    reg cDiv : UInt<1>, clock with : (reset => (reset, UInt<1>("h01"))) @[MultiClockSpec.scala 12:21]
    node _T_8 = eq(cDiv, UInt<1>("h00")) @[MultiClockSpec.scala 13:11]
    cDiv <= _T_8 @[MultiClockSpec.scala 13:8]
    node clock2 = asClock(cDiv) @[MultiClockSpec.scala 14:21]
    reg reg1 : UInt<8>, clock with : (reset => (reset, UInt<8>("h00"))) @[MultiClockSpec.scala 16:21]
    node _T_12 = add(reg1, UInt<1>("h01")) @[MultiClockSpec.scala 17:16]
    node _T_13 = tail(_T_12, 1) @[MultiClockSpec.scala 17:16]
    reg1 <= _T_13 @[MultiClockSpec.scala 17:8]
    reg reg2 : UInt<8>, clock2 with : (reset => (reset, UInt<8>("h00"))) @[MultiClockSpec.scala 18:41]
    node _T_17 = add(reg2, UInt<1>("h01")) @[MultiClockSpec.scala 19:16]
    node _T_18 = tail(_T_17, 1) @[MultiClockSpec.scala 19:16]
    reg2 <= _T_18 @[MultiClockSpec.scala 19:8]
    node _T_20 = lt(reg1, UInt<4>("h0a")) @[MultiClockSpec.scala 21:14]
    when _T_20 : @[MultiClockSpec.scala 21:22]
      node _T_22 = div(reg1, UInt<2>("h02")) @[MultiClockSpec.scala 22:26]
      node _T_23 = eq(reg2, _T_22) @[MultiClockSpec.scala 22:17]
      node _T_24 = bits(reset, 0, 0) @[MultiClockSpec.scala 22:11]
      node _T_25 = or(_T_23, _T_24) @[MultiClockSpec.scala 22:11]
      node _T_27 = eq(_T_25, UInt<1>("h00")) @[MultiClockSpec.scala 22:11]
      when _T_27 : @[MultiClockSpec.scala 22:11]
        printf(clock, UInt<1>(1), "Assertion failed\n    at MultiClockSpec.scala:22 assert(reg2 === reg1 / 2.U) // 1:2 clock relationship\n") @[MultiClockSpec.scala 22:11]
        stop(clock, UInt<1>(1), 1) @[MultiClockSpec.scala 22:11]
        skip @[MultiClockSpec.scala 22:11]
      skip @[MultiClockSpec.scala 21:22]
    node _T_29 = eq(reg1, UInt<4>("h0a")) @[MultiClockSpec.scala 25:14]
    when _T_29 : @[MultiClockSpec.scala 25:24]
      node _T_30 = bits(reset, 0, 0) @[MultiClockSpec.scala 26:9]
      node _T_32 = eq(_T_30, UInt<1>("h00")) @[MultiClockSpec.scala 26:9]
      when _T_32 : @[MultiClockSpec.scala 26:9]
        stop(clock, UInt<1>(1), 0) @[MultiClockSpec.scala 26:9]
        skip @[MultiClockSpec.scala 26:9]
      skip @[MultiClockSpec.scala 25:24] 
```
After transforming to low Firrtl
```
circuit ClockDividerTest :
  module ClockDividerTest :
    input clock : Clock
    input reset : UInt<1>
  
    reg cDiv : UInt<1>, clock with :
      reset => (UInt<1>("h0"), cDiv) @[MultiClockSpec.scala 12:21]
    reg reg1 : UInt<8>, clock with :
      reset => (UInt<1>("h0"), reg1) @[MultiClockSpec.scala 16:21]
    reg reg2 : UInt<8>, clock2 with :
      reset => (UInt<1>("h0"), reg2) @[MultiClockSpec.scala 18:41]
    node _T_8 = eq(cDiv, UInt<1>("h0")) @[MultiClockSpec.scala 13:11]
    node clock2 = asClock(cDiv) @[MultiClockSpec.scala 14:21]
    node _T_12 = add(reg1, UInt<1>("h1")) @[MultiClockSpec.scala 17:16]
    node _T_13 = tail(_T_12, 1) @[MultiClockSpec.scala 17:16]
    node _T_17 = add(reg2, UInt<1>("h1")) @[MultiClockSpec.scala 19:16]
    node _T_18 = tail(_T_17, 1) @[MultiClockSpec.scala 19:16]
    node _T_20 = lt(reg1, UInt<4>("ha")) @[MultiClockSpec.scala 21:14]
    node _T_22 = div(reg1, UInt<2>("h2")) @[MultiClockSpec.scala 22:26]
    node _T_23 = eq(reg2, _T_22) @[MultiClockSpec.scala 22:17]
    node _T_24 = bits(reset, 0, 0) @[MultiClockSpec.scala 22:11]
    node _T_25 = or(_T_23, _T_24) @[MultiClockSpec.scala 22:11]
    node _T_27 = eq(_T_25, UInt<1>("h0")) @[MultiClockSpec.scala 22:11]
    node _T_29 = eq(reg1, UInt<4>("ha")) @[MultiClockSpec.scala 25:14]
    node _T_30 = bits(reset, 0, 0) @[MultiClockSpec.scala 26:9]
    node _T_32 = eq(_T_30, UInt<1>("h0")) @[MultiClockSpec.scala 26:9]
    cDiv <= mux(reset, UInt<1>("h1"), _T_8)
    reg1 <= mux(reset, UInt<8>("h0"), _T_13)
    reg2 <= mux(reset, UInt<8>("h0"), _T_18)
    printf(clock, and(and(and(UInt<1>("h1"), _T_20), _T_27), UInt<1>("h1")), "Assertion failed\n    at MultiClockSpec.scala:22 assert(reg2 === reg1 / 2.U) // 1:2 clock relationship\n") @[MultiClockSpec.scala 22:11]
    stop(clock, and(and(and(UInt<1>("h1"), _T_20), _T_27), UInt<1>("h1")), 1) @[MultiClockSpec.scala 22:11]
    stop(clock, and(and(and(UInt<1>("h1"), _T_29), _T_32), UInt<1>("h1")), 0) @[MultiClockSpec.scala 26:9]
```
Clock2 is referenced in the reg2 declaration, this is before it is declared two statements later.

* **What is the use case for changing the behavior?**

This causes problems for the treadle firrtl compiler. If it is acceptable, a work-around in the compiler might be found but it seems to be the only case in which this ordering problems occurs.

* **Impact**
There should be no functional changes other than a test that doesn't run will run. This
will have no impact on the Treadle API

seldridge, 2018-03-02:
This does violate static single assignment for low FIRRTL which should be a guarantee of middle FIRRTL and lower.

Quick drive-by analysis... this appears to be an issue with the `RemoveWires` transform in that declarations are simply collected and placed before all logic (in this case, all nodes and connects). See: https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/transforms/RemoveWires.scala#L81

This likely motivates the eventual `CheckMidForm` and `CheckLowForm` passes with SSA being one of the checks.

jackkoenig, 2018-03-02:
Whoops. Well it looks like we can't replace all wires with nodes. Bummer.

seldridge, 2018-03-02:
@jackkoenig: Is it that bad or does it just mean that register declarations _may_ need to be intermixed with nodes in low FIRRTL? Maybe I'm missing something? It would seem like a properly linearized netlist is still possible, just that all `DefReg` cannot be grouped before `DefNode` if there exist multiple used clocks in a module?

`CheckChirrtl` does trip on this, FWIW. I haven't verified, but I'd assume that `CheckHighForm` will, too. In lieu of actually adding checks for mid and low form, it may be worth getting some coverage here by running high form checks before all lowerings.

jackkoenig, 2018-03-02:
I think the issue is that you could technically have a loop through a register back to it's own clock, eg.

```
wire clockWire : Clock
reg r : UInt<1>, clockWire
clockWire <= asClock(r)
```
I don't think this is common (or something anyone should be doing), but it's technically legal FIRRTL.

It's long overdue that we insert additional checks, at least HighFormCheck, between all transforms during testing, and probably more checks during normal compilation.

seldridge, 2018-03-02:
After thinking about this some more, I think that we may want to make this type of connection disallowed in the same way that combinational loops are disallowed.

I'm racking my brain for some situation where you'd want to do this. I think that you wind up two major classes of circuits if this is allowed:
- In the situation of a register with a combinational path from its output to its own clock, this would seem to describe some type of sticky logic that will always converge to a constant value eventually.
- In the situation of a register with a combinational path from its output to its own clock and its own input, this would seem to be enabling the building of astable circuit.

If you are using these approaches, you're either describing some logic which should be described differently or trying to describe some odd circuit that's likely not suitable for synthesis. I fully admit that I may be overlooking something here. 

I don't think that existing tools will complain about this like they will with combinational loops, e.g., Yosys is fine with it. I haven't tried playing with this in FPGA/ASIC tools. I expect that allowing this is more due to the freedom of Verilog/VHDL more than it is due to sound logic design. Caveat that I'm overlooking something still applies.

One solution here would be to add a pass, running after the combinational loop check, that prevents these types of connections. Assuming that that's the only situation that would disallow SSA after wire removal, then SSA with mixed nodes and registers would be possible.

jackkoenig, 2018-03-02:
Yeah I think disallowing this type of connection makes sense. I believe we could just do it in the combinational loop check (by adding a "combinational" dependency from a registers clock to it's output) although distinguishing for the purposes of error messages might be non-trivial.

chick, 2018-06-12:
@jackkoenig @seldridge 
Ping!
Any chance of getting a  fix to this cobbled together (such as the graph dependency approach).
This has now bitten @ducky64 in working on multi-clock in testers2.
Or maybe there is some other workaround or combination of passes that might get us past this.

seldridge, 2018-06-13:
I think I have a quick solution here. A quick extension to `getOrderedNodes` seems to be working. I'm writing up some testing infrastructure and should then have something for you to look at.

jackkoenig, 2018-06-22:
@chick can you check if this is fixed?

jackkoenig, 2018-06-22:
Fixed by #823

closed 2018-06-22 by jackkoenig