chipsalliance/firrtl #753: Performing a bitwise NOT on a SInt doesn't return the correct output

lfiolhais, 2018-03-06:
This repository's issues are reserved for feature requests and bug reports.

* **Type of issue**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Other enhancement

* **If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **If possible, please directly provide the FIRRTL file or a relevant fragment if the file is large**

A simple Chisel module which receives two inputs, one unsigned and one signed, and performs a bitwise NOT on both of them. (BTW, I couldn't get Chisel to output the hexadecimal representation of `b` without casting to UInt...)
```scala
import chisel3._

class InvBug() extends Module {
	val io = IO(new Bundle {
		val a = Input(UInt(32.W))
                val b = Input(SInt(32.W))

                val a_out = Output(UInt(32.W))
                val b_out = Output(SInt(32.W))
	})
  io.a_out := ~io.a
  io.b_out := ~io.b
  printf("a %d | a (hex) %x | a_out %d | a_out (hex) %x\n", io.a, io.a, io.a_out, io.a_out)
  printf("b %d | b (hex) %x | b_out %d | b_out (hex) %x\n", io.b, io.b.asUInt, io.b_out, io.b_out)
}
```

Generated FIRRTL
```
circuit InvBug : @[:@2.0]
  module InvBug : @[:@3.2]
    input clock : Clock @[:@4.4]
    input reset : UInt<1> @[:@5.4]
    input io_a : UInt<32> @[:@6.4]
    input io_b : SInt<32> @[:@6.4]
    output io_a_out : UInt<32> @[:@6.4]
    output io_b_out : SInt<32> @[:@6.4]
  
    node _T_6 = not(io_a) @[inv.scala 13:15:@8.4]
    node _T_7 = not(io_b) @[inv.scala 14:15:@10.4]
    node _T_8 = asSInt(_T_7) @[inv.scala 14:15:@11.4]
    node _T_9 = bits(reset, 0, 0) @[inv.scala 15:9:@13.4]
    node _T_11 = eq(_T_9, UInt<1>("h0")) @[inv.scala 15:9:@14.4]
    node _T_12 = asUInt(io_b) @[inv.scala 16:72:@18.4]
    node _T_13 = bits(reset, 0, 0) @[inv.scala 16:9:@19.4]
    node _T_15 = eq(_T_13, UInt<1>("h0")) @[inv.scala 16:9:@20.4]
    io_a_out <= _T_6
    io_b_out <= _T_8
    printf(clock, and(and(UInt<1>("h1"), _T_11), UInt<1>("h1")), "a %d | a (hex) %x | a_out %d | a_out (hex) %x\n", io_a, io_a, io_a_out, io_a_out) @[inv.scala 15:9:@16.6]
    printf(clock, and(and(UInt<1>("h1"), _T_15), UInt<1>("h1")), "b %d | b (hex) %x | b_out %d | b_out (hex) %x\n", io_b, _T_12, io_b_out, io_b_out) @[inv.scala 16:9:@22.6]
```

The test bench used is
```scala
import chisel3.iotesters
import chisel3.iotesters.{ChiselFlatSpec, Driver, PeekPokeTester}

import math.BigInt

class InvBugUnitTester(c: InvBug) extends PeekPokeTester(c) {
  poke(c.io.a, BigInt("ffffffff", 16))
  poke(c.io.b, -1)

  step(1)

  expect(c.io.a_out, 0)
  expect(c.io.b_out, 0)
}
```
  * **What is the current behavior?**

Using FIRRTL's backend, the prints of the test bench are
```
a 4294967295 | a (hex) ffffffff | a_out 0 | a_out (hex) 0
b -1 | b (hex) ffffffff | b_out 2147483646 | b_out (hex) 7ffffffe
```
and the test fails when expecting `b_out`. Scala computes 0 and FIRRTL 2147483646. The tests successfully pass using the verilator backend.
  * **What is the expected behavior?**

The tests to pass with the following output:
```
a 4294967295 | a (hex) ffffffff | a_out 0 | a_out (hex) 0
b -1 | b (hex) ffffffff | b_out 0 | b_out (hex) 0
```
  * **Please tell us about your environment:**
    - version: `3.0-SNAPSHOT`
    - OS: `Linux TIE-Fighter 4.9.0-6-amd64 #1 SMP Debian 4.9.82-1+deb9u3 (2018-03-02) x86_64 GNU/Linux`
    - verilator version: `Verilator 3.900 2017-01-15 rev verilator_3_890-15-ge6d7e7e`

* **What is the use case for changing the behavior?**

Using an adder for addition and subtraction depending on a selection signal by performing the two's complement on the second operand.

* **Impact**
  - [ ] no functional change
  - [ ] API addition (no impact on existing code)
  - [x] API modification
  - [ ] unknown

* **Development Phase**
  - [x] request
  - [ ] proposal

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

This is my first bug report here, hopefully I provided all the necessary information. I also searched for other similar issues and couldn't find any.

It seems that FIRRTL is interpreting the signed input as a sign-magnitude number and performing the bitwise complement using that representation. Chisel's [wiki](https://github.com/freechipsproject/chisel3/wiki/Datatypes-in-Chisel) states that signed integers are represented using two's complement format.

I don't mind putting a PR together to fix this.

lfiolhais, 2018-03-06:
This test bench does _not_ fail in verilator, I get the correct output:
```
[info] [0.006] a 4294967295 | a (hex) ffffffff | a_out          0 | a_out (hex) 00000000
[info] [0.006] b 4294967295 | b (hex) ffffffff | b_out          0 | b_out (hex) 00000000
```

seldridge, 2018-03-06:
Thanks for the clarification. I guess you mentioned that in the original PR...

lfiolhais, 2018-03-06:
No worries @seldridge =D! Thanks for the help. I'll close this in favor of https://github.com/freechipsproject/firrtl-interpreter/issues/122. You were correct, the problem lies in the interpreter.

closed 2018-03-06 by lfiolhais