chipsalliance/firrtl #754: Accessing SyncReadMem causes FIRRTL Internal Error.

chick, 2018-03-06:
This repository's issues are reserved for feature requests and bug reports.

* **Bug Report**

* **The following test generates an Internal Error**
```scala
// See LICENSE for license details.

package puzzles

import chisel3._
import chisel3.iotesters.PeekPokeTester
import org.scalatest.{FreeSpec, Matchers}

class MemMask extends Module {
  val io = IO(new Bundle {
    val addr      = Input(UInt(8.W))
    val write     = Input(Bool())

    val writeData = Input(Vec(4, UInt(8.W)))
    val readData  = Output(Vec(4, UInt(8.W)))

    val writeMask = Input(Vec(4, Bool()))
    val readMask  = Input(Vec(4, UInt(8.W)))
  })

  val mem = SyncReadMem(1024, Vec(4, UInt(8.W)))

  val writeData = Vec
  when(io.write) {
    mem.write(io.addr, io.writeData, io.writeMask)
  }
  .otherwise {
    val bits = mem.read(io.addr, true.B).zip(io.readMask).map { case (a, b) => a & b }

    io.readData := bits
  }
}

class MemMaskTester(c: MemMask) extends PeekPokeTester(c) {
  poke(c.io.addr, 2)
  poke(c.io.write, 1)
  poke(c.io.writeMask(0), 1)
  poke(c.io.writeMask(1), 1)
  poke(c.io.writeMask(2), 1)
  poke(c.io.writeMask(3), 1)

  poke(c.io.writeData(0), 1)
  poke(c.io.writeData(0), 2)
  poke(c.io.writeData(0), 3)
  poke(c.io.writeData(0), 4)

  step(1)

  for(j <- 0 until 4) {
    poke(c.io.write, 0)
    poke(c.io.readMask(0), 0)
    poke(c.io.readMask(1), 0)
    poke(c.io.readMask(2), 0)
    poke(c.io.readMask(3), 0)

    poke(c.io.readMask(j), 1)

    step(1)

    println(s"j = $j mem(2) ${(0 until 4).map { i => peek(c.io.readData(i)) }.mkString(", ")}")
  }
}

class MemMaskSpec extends FreeSpec with Matchers {
  "should show things properly" in {
    iotesters.Driver.execute(Array(), () => new MemMask) { c =>
      new MemMaskTester(c)
    } should be (true)
  }
}
```
  * **Error**
```
Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
	at firrtl.Utils$.error(Utils.scala:399)
	at firrtl.Utils$.throwInternalError(Utils.scala:180)
	at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:214)
	at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:163)
	at logger.Logger$$anonfun$makeScope$1.apply(Logger.scala:129)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at logger.Logger$.makeScope(Logger.scala:127)
	at firrtl.Driver$.execute(Driver.scala:163)
	at chisel3.Driver$.execute(Driver.scala:182)
	at chisel3.iotesters.setupFirrtlTerpBackend$.apply(FirrtlTerpBackend.scala:126)
	at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply$mcZ$sp(Driver.scala:51)
	at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply(Driver.scala:38)
	at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply(Driver.scala:38)
	at logger.Logger$$anonfun$makeScope$1.apply(Logger.scala:129)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at logger.Logger$.makeScope(Logger.scala:127)
	at chisel3.iotesters.Driver$$anonfun$execute$1.apply$mcZ$sp(Driver.scala:38)
	at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:38)
	at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:38)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.iotesters.Driver$.execute(Driver.scala:37)
	at chisel3.iotesters.Driver$.execute(Driver.scala:94)
	at puzzles.MemMaskSpec$$anonfun$6.apply(MemMask.scala:66)
	at puzzles.MemMaskSpec$$anonfun$6.apply(MemMask.scala:68)
  * **What is the current behavior?**
  * **What is the expected behavior?**
  * **Please tell us about your environment:**
    (examples)
    - version: `3.0-SNAPSHOT`
    - OS: `Linux knight 4.4.0-92-generic #115-Ubuntu SMP Thu Aug 10 09:04:33 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`
    - verilator version (if relevant)
```
* **What is the use case for changing the behavior?**
Not sure.  This was found while trying to build a working example of
[stackoverflow question about memory read masks](https://stackoverflow.com/questions/49126542/chisel-memories-support-read-masks-directly)

* **Impact Unknown**

chick, 2018-03-16:
The problem is an unintitialized output. The question is why is this blowing up the way that it does.

albert-magyar, 2020-02-03:
SyncReadMems have been totally refactored, so I'l close this unless it's still an issue.

closed 2020-02-03 by albert-magyar