chipsalliance/firrtl #759: Error converting fir to verilog

nnamenoname, 2018-03-12:
Hello:
I have write hardware module with Chisel, and converted it to verilog code.
However, it seems odd behavior when convert fir to verilog.
As below:
.fir

```
module TLBridgeDMA_bridgedma : 
input clock : Clock
input reset : UInt<1>
...................
reg sdma_state : UInt<3>, clock with : (reset => (reset, UInt<3>("h00"))) @[TLDMA.scala 92:42]
.................
```
.v
```
module TLBridgeDMA_bridgedma(
input         clock, // @[:BridgeDMA.TestHarness.HasBridgeDMAConfig.fir@163502.4]
input         reset, 
.............

```
I cannot find something as "reg sdma_state" in verilog code and it disappears.
I am not sure why such odd behavior occurs.
Is it bug or my Chisel implementation code violate some coding style rule?

edwardcwang, 2018-03-12:
Is it possible that your register might have gotten `ConstProp`ed away?

nnamenoname, 2018-03-12:
Sorry about that I am not familiar with firrtl end.
Could you describe ConstProp detailed?

nnamenoname, 2018-03-12:
In my Chisel implementation, there is another state register ddma_state: `val ddma_state              = RegInit(ddma_idle)`. I also write two state List in single module.
```scala
val sdma_idle :: .... :: Nil 
val ddma_idle :: .... :: Nil 
```

I can find ddma_state register in output verilog code, but sdma_state is missing.
```
reg [2:0] ddma_state; // @[TLBridgeDMA.scala 101:42:BridgeDMA.TestHarness.HasBridgeDMAConfig.fir@163563.4]

```
Can I write multiple state List as above in single module?
Or it cannot. Therefor I get such error.

nnamenoname, 2018-03-12:
Sorry for being nagging again.
I have traced out the fir file and verilog file.
Some register as "sdma_state" and fsm operation blocks exist certainly in fir fie.
However, they disappear in out verilog file.
I still cannot figure out the reasons behind.

seldridge, 2018-03-12:
As @edwardcwang gets at, this is likely either due to one of two transforms optimizing away parts of your code through either:
  - Dead Code Eliminataion
  - Constant Propagation

Consider the following FIRRTL code. This has three modules: `A`, `B`, `C`. All three have an internal 1-bit register called `sdam_state`. Dead code will remove this register in `A` (because `sdma_state` is never used), constant propagation will remove `B` (and the entirety of module `B` because `sdam_state` is set to zero and never changes), and `C` will be not modified by either of these two passes.

```
circuit Top:
  module A:
    input clock: Clock
    input reset: UInt<1>
    input in : UInt<1>
    output out: UInt<1>

    reg sdma_idle : UInt<1>, clock with : (reset => (reset, UInt<1>("h00")))

    out <= in

  module B:
    input clock: Clock
    input reset: UInt<1>
    input in : UInt<1>
    output out: UInt<1>

    reg sdma_idle : UInt<1>, clock with : (reset => (reset, UInt<1>("h00")))

    out <= sdma_idle

  module C:
    input clock: Clock
    input reset: UInt<1>
    input in : UInt<1>
    input switch : UInt<1>
    output out: UInt<1>

    reg sdma_idle : UInt<1>, clock with : (reset => (reset, UInt<1>("h00")))

    when switch :
      sdma_idle <= xor(UInt<1>("h01"), sdma_idle)

    out <= or(in, sdma_idle)

  module Top:
    input clock: Clock
    input reset: UInt<1>
    input in : UInt<1>
    output out_a : UInt<1>
    output out_b : UInt<1>
    output out_c : UInt<1>

    inst a of A
    inst b of B
    inst c of C

    a.clock <= clock
    a.reset <= reset
    a.in <= in
    out_a <= a.out

    b.clock <= clock
    b.reset <= reset
    b.in <= in
    out_b <= b.out

    c.clock <= clock
    c.reset <= reset
    c.in <= in
    c.switch <= asUInt(clock)
    out_c <= c.out
```
I then run this through the FIRRTL compiler with:
```
firrtl/utils/bin/firrtl -ll Debug -i test.fir -o test.v
```

While all this is flying by, you see the following which provides information about what dead code is being eliminated.
```
======== Starting Transform DeadCodeElimination ========
[DCE] port A.clock
[DCE] port A.reset
[DCE] reg A.sdma_idle
[DCE] port B.clock
[DCE] port B.reset
[DCE] port B.in
[DCE] port B.out
[DCE] reg B.sdma_idle
[DCE] module B
[DCE] inst Top.b
--------------------------------------------------------

Time: 11.1 ms
Form: UnknownForm
Annotations:
targetDir
.
EmitCircuitAnnotation(class firrtl.VerilogEmitter)
TargetDirAnnotation(.)
======== Finished Transform DeadCodeElimination ========
```
I then wind up with the following Verilog (where I've manually removed the randomization stuff to make it less cluttered):
```verilog
module A( // @[:test.fir@2.2]
  input   in, // @[:test.fir@5.4]
  output  out // @[:test.fir@6.4]
);
  assign out = in;
endmodule
module C( // @[:test.fir@22.2]
  input   clock, // @[:test.fir@23.4]
  input   reset, // @[:test.fir@24.4]
  input   in, // @[:test.fir@25.4]
  input   switch, // @[:test.fir@26.4]
  output  out // @[:test.fir@27.4]
);
  reg  sdma_idle; // @[:test.fir@29.4]
  reg [31:0] _RAND_0;
  wire  _GEN_0; // @[:test.fir@31.4]
  assign _GEN_0 = switch ? 1'h1 ^ sdma_idle : sdma_idle; // @[:test.fir@31.4]
  assign out = in | sdma_idle;
  always @(posedge clock) begin
    if (reset) begin
      sdma_idle <= 1'h0;
    end else begin
      if (switch) begin
        sdma_idle <= 1'h1 ^ sdma_idle;
      end
    end
  end
endmodule
module Top( // @[:test.fir@36.2]
  input   clock, // @[:test.fir@37.4]
  input   reset, // @[:test.fir@38.4]
  input   in, // @[:test.fir@39.4]
  output  out_a, // @[:test.fir@40.4]
  output  out_b, // @[:test.fir@41.4]
  output  out_c // @[:test.fir@42.4]
);
  wire  a_in; // @[:test.fir@44.4]
  wire  a_out; // @[:test.fir@44.4]
  wire  c_clock; // @[:test.fir@46.4]
  wire  c_reset; // @[:test.fir@46.4]
  wire  c_in; // @[:test.fir@46.4]
  wire  c_switch; // @[:test.fir@46.4]
  wire  c_out; // @[:test.fir@46.4]
  A a ( // @[:test.fir@44.4]
    .in(a_in),
    .out(a_out)
  );
  C c ( // @[:test.fir@46.4]
    .clock(c_clock),
    .reset(c_reset),
    .in(c_in),
    .switch(c_switch),
    .out(c_out)
  );
  assign out_a = a_out;
  assign out_b = 1'h0;
  assign out_c = c_out;
  assign a_in = in;
  assign c_clock = clock;
  assign c_reset = reset;
  assign c_in = in;
  assign c_switch = $unsigned(clock);
endmodule
```

Usually, DCE/ConstProp are doing the right things and they likely are in your case. If you want to mark specific signals as not getting modified by these types of passes, you can use a `dontTouch` annotation in your Chisel code: https://chisel.eecs.berkeley.edu/api/#chisel3.core.dontTouch$

nnamenoname, 2018-03-14:
Thanks for your tips. I just fix out partial problem.
However, there are still some problems and condition regarded  as  Dead Code Eliminataion from FIRRTL compiler.
Code:
```scala
class FrontendModule(c: FrontendModuleParams)(implicit p: Parameters) extends TLRegisterRouter(
    c.address, "FrontendConfig", beatBytes = c.beatBytes)(
      new TLRegBundle(c, _) with DeviceCtrlBundle)(
      new TLRegModule(c, _, _) with FrontendTLModule)

trait DeviceCtrlBundle extends Bundle {
  implicit val p: Parameters

  val front = new ControllerIOBundle  //Output
  val back  = new ResponseIO             //Input
}
```
Top module:
```scala
class TLTOP (address: BigInt, beatBytes: Int)(implicit p: Parameters) extends LazyModule {
  
  val frontend = LazyModule(new FrontendModule(FrontendModuleParams(address, beatBytes)))
 
  lazy val module = new LazyModuleImp(this) {

   val source_address  = RegInit(0.U(xLen.W))
   source_address   := frontend.module.io.front.source_address
   ......

  }
}

```
```
[DCE] port FrontendModule.io_front_source_address
[DCE] port TLTop_tltop.io_front_source_address
```
It always return as above message with using FIRRTL in spite of 
applying donotTouch API or modifying coding style.

jackkoenig, 2018-04-20:
Dead Code Elimination doesn't remove things because they aren't used, it removes them because they do not affect the output. In the partial snippet given, source_address only updates itself, it must affect the output for DCE to keep it. You can mark it `dontTouch` (eg. `dontTouch(source_address)`) to ensure it is kept.

seldridge, 2018-12-18:
If this is still an issue, feel free to reopen.

closed 2018-12-18 by seldridge