chipsalliance/firrtl #764: [RFC] Annotations subsume ExecutionOptions?

seldridge, 2018-03-19:
After thinking more about @azidar's comments related to command line options as "that's just an annotation", I think I have an idea on how the options and annotations can be better unified. (Note: this may be me either not realizing that I'm parroting what was being said or that this may have been discussed before.)

Basically, command line options are annotations and with the maturity of the annotations refactor it makes sense to have a discussion about whether `ExecutionOptionsManager` is needed in its current capacity.

Currently, `ExecutionsOptionsManager` is serving two purposes:
1. Maintaining Driver-level information, e.g., the input file name (`inputFileNameOverride` for `FirrltExecutionOptions`)
2. Providing lightweight shims to generate annotations either indirectly with the help of the Driver (e.g., `noDCE`) or directly in the options manager (e.g., `inline`)

However, there are two oddities here that make things somewhat incongruent now that the annotations refactor is in place and lacking in extensibility once new, custom annotations/transforms are added:

__Incongruency of annotations vs. Options__: All options are just annotations. These either target nothing (`NoTargetAnnotation`) or some named component (`SingleTargetAnnotation`) or, assumedly, some to-be-defined multiple components (a not-yet-implemented `MultiTargetAnnotation`). The ability to define both "top level options" (e.g., `CommonOptions`) as well as `NoTargetAnnotations` seems particularly kludgy. Unifying on one (assumedly annotations) then means that everything that needs this information is passed along with the only thing that matters, the Circuit state.

__Difficulties in adding command line options for external transforms__: If I go and build an external library that adds annotations and transforms, it would be great if I can hack these directly into the command line option parsing providing by the current options manager. The current approach to doing this would seem to be to define my own options manager. While not insurmountable, this gets complicated if I"m combining several FIRRTL libraries together and want to run all their transforms. Here, there's nothing that precludes forcing annotations to expose what options they take or using a trait that gets mixed into annotations (e.g., `HasCommandLineOptions`). The job of the options manager is then to use reflection to find all annotations that exist or all annotations that have the CLI options trait and running their methods to generate help text and parse everything. You then get some nice magic in that all the external libraries that may be used are then automatically included in command line parsing. Note: the ability of reflection to adequately do this has not been verified on my part, yet.

This seems intuitive to me and perhaps the direction that @jackkoenig was thinking with the annotations refactor (or this may have been brought up and discussed before).

The major downside of this is that this may not be doable in a way that avoids a simultaneous refactor of Chisel's options manager. Furthermore, there are odd situations that may arise if certain annotations incorrectly modify (or delete) top-level, Driver-like annotations (like directories or output filenames). Technically this should all be allowable, though...

I can think about this some more and/or try to get a small PR that migrates an existing annotation into using this approach.

* **Type of issue**
  - Other enhancement

* **What is the use case for changing the behavior?**

This enables writers of custom passes to expose their command line options all the way to the top.

* **Impact**
  - API addition (no impact on existing code)

* **Development Phase**
  - proposal

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

azidar, 2018-03-20:
Good thoughts @seldridge. I've talked with @jackkoenig about using reflection to find classes, and apparently its pretty non-trivial (though it is possible, as scalatest does it). If we could figure that out, it would also make scheduling transforms much easier as well, as you could also use reflection to find the custom transforms specified by the commandline options.

chick, 2018-03-20:
I just talked to @ucbjrl about this. Sounds like the BuildInfo system has some very relevant code. Jim will add a comment here on the subject

ucbjrl, 2018-03-20:
Maybe I just dreamt this (I can't locate the example code at the moment), but I was sure that at one point I was collecting all the `BuildInfo` objects using reflection to produce suitable version strings for tagging generated output.

azidar, 2018-03-20:
I just started playing around with http://software.clapper.org/classutil/, and got something working pretty quickly inside firrtl.

```scala
package firrtl
import org.clapper.classutil.ClassFinder

object ExampleReflection {
  def main(args: Array[String]): Unit = {
    val finder = ClassFinder()
    val classes = finder.getClasses // classes is an Iterator[ClassInfo]
    val annotationClasses = ClassFinder.concreteSubclasses("firrtl.annotations.Annotation", classes)
    annotationClasses.foreach(println)
  }
}
```

It prints out:
```
firrtl.passes.InlineAnnotation
firrtl.passes.clocklist.ClockListAnnotation
firrtl.passes.memlib.ReplSeqMemAnnotation
firrtl.transforms.NoDedupAnnotation
firrtl.EmitAllModulesAnnotation
firrtl.passes.memlib.PinAnnotation
firrtl.passes.wiring.SinkAnnotation
firrtl.transforms.BlackBoxInlineAnno
firrtl.passes.memlib.InferReadWriteAnnotation$
firrtl.transforms.OptimizableExtModuleAnnotation
firrtl.transforms.FlattenAnnotation
firrtl.transforms.DontTouchAnnotation
firrtl.passes.memlib.NoDedupMemAnnotation
firrtl.transforms.DontCheckCombLoopsAnnotation$
firrtl.EmittedFirrtlModuleAnnotation
firrtl.EmitCircuitAnnotation
firrtl.transforms.BlackBoxResourceAnno
firrtl.EmittedVerilogModuleAnnotation
firrtl.EmittedFirrtlCircuitAnnotation
firrtl.TargetDirAnnotation
firrtl.annotations.DeletedAnnotation
firrtl.EmittedVerilogCircuitAnnotation
firrtl.transforms.BlackBoxTargetDirAnno
firrtl.transforms.NoDCEAnnotation$
firrtl.annotations.LegacyAnnotation
firrtl.passes.wiring.SourceAnnotation
```

chick, 2018-03-20:
Sweet!

jackkoenig, 2018-03-20:
I had a lot of issues with ClassFinder when I was originally playing around with refactoring annotations, I'm doing some archeology to see if I can reproduce the problems

seldridge, 2018-03-20:
@azidar: Both of those points (me failing with complicated reflection and the heaviness of a mutable scopt refactor) derailed what I thought would be a one hour example PR exercise. Anyway, I'm in agreement that those two points would likely make this tractable. That `classutil` looks better than what I was playing (fighting) with: https://github.com/freechipsproject/firrtl/pull/765/files#diff-330f66cb371a46100e1ae1b462cb8aa4R409. I was able to find anything that was indicating (via a trait) that it had annotations. 

Something like this does look to be exactly what we need.

@chick: Thanks for the pointer. I'll need to dig into that more, but that may help if the projects are disjoint or being pulled in as actual "project in directory" things. The one annoyance here is if you're constructing a project that uses FIRRTL as a fat jar executable, but is pulling annotations that get compiled into Rocket (e.g., from a `ROCKETCHIP_ADDONS`). @grebe's black box refactor indirectly helps this by using a broader class path when running the FIRRTL executable, but there may be some overall directory structure that can be used with BuildInfo here. I'll have to think more.

__all__: Don't let me block you with this. If somebody wants to take a more concrete stab at it, go for it. I just want to make sure that the approach seems sound (as this looks to be a more intrusive refactor than I originally thought).

azidar, 2018-03-20:
This library lets you provide a path to a jar for it to look into. A fallback could be that a Firrtl (non-annotation) commandline option is to pass in a jar for it to inspect?

jackkoenig, 2018-03-20:
Here's a stripped down version of what won't work for me. I'm probably just doing something stupid but I abandoned ClassFinder when I couldn't get this to work (to be fair, using ClassFinder would be merely a performance improvement and code cleanup for the current annotations implementation, whereas for providing command-line options it's more of a requirement).

```scala
package experiment

import org.clapper.classutil.ClassFinder

abstract class MyAbstractClass

case class Foo(name: String) extends MyAbstractClass
case class Bar(xs: List[Int]) extends MyAbstractClass

object Test extends App {

  val finder = ClassFinder()
  val classes = finder.getClasses

  println(s"There are ${classes.size} total classes")

  val subclasses1 = ClassFinder.concreteSubclasses(classOf[experiment.MyAbstractClass], classes).toList
  val subclasses2 = ClassFinder.concreteSubclasses("experiment.MyAbstractClass", classes).toList

  println(s"Subclasses1: $subclasses1")

  println(s"Subclasses2: $subclasses2")
}
```

Running this emits the following:
```
[info] Running experiment.Test
There are 1116 total classes
Subclasses1: List()
Subclasses2: List()
```

chick, 2018-03-20:
Weird, what is the difference between @azidar and @jackkoenig examples. Abstract class vs trait? I don't get it

jackkoenig, 2018-03-20:
I have no idea what the issue is, I've tried doing a trait, I've tried moving the class definitions to a different file, I've tried different versions of sbt and scala. I hope it's just something dumb in my code but I worry it's an environment or platform thing.

azidar, 2018-03-20:
Lol. Running that on my machine:

```scala
package experimental

import org.clapper.classutil.ClassFinder

abstract class MyAbstractClass

case class Foo(name: String) extends MyAbstractClass
case class Bar(xs: List[Int]) extends MyAbstractClass

object Test extends App {

  val finder = ClassFinder()
  val classes = finder.getClasses

  println(s"There are ${classes.size} total classes")

  val subclasses1 = ClassFinder.concreteSubclasses(classOf[experimental.MyAbstractClass], classes).toList
  val subclasses2 = ClassFinder.concreteSubclasses("experimental.MyAbstractClass", classes).toList

  println(s"Subclasses1: $subclasses1")

  println(s"Subclasses2: $subclasses2")
}
```

prints out

```
There are 52096 total classes
Subclasses1: List(experimental.Foo, experimental.Bar)
Subclasses2: List(experimental.Foo, experimental.Bar)
```

azidar, 2018-03-20:
I reverted my sbt version to 0.13, I wonder if that causes this to work....

jackkoenig, 2018-03-20:
I've tried it with 0.13.16 and 0.13.17, this is so bizarre. I'm going to try running it on another machine

azidar, 2018-03-20:
What's your directory structure?

jackkoenig, 2018-03-20:
I'm just doing a regular sbt command-line project with the files in `src/main/scala/experiment`. Considering the difference in number of classes, I'm guessing there's something up with the way ClassFinder accesses the classpath. Are you using IntelliJ?

jackkoenig, 2018-03-20:
Same issue running on a different machine

seldridge, 2018-03-20:
To concretize this: https://github.com/seldridge/firrtl-issue-764-test. I'm getting the same issue as @jackkoenig :cry:

```bash
git clone https://github.com/seldridge/firrtl-issue-764-test wtf
cd wtf
make
```

Output:
```
> make
sbt run
[info] Loading settings from plugins.sbt ...
[info] Loading global plugins from /home/se/.sbt/1.0/plugins
[info] Loading project definition from /home/se/tmp/wtf/project
[info] Loading settings from build.sbt ...
[info] Set current project to wtf (in build file:/home/se/tmp/wtf/)
[info] Packaging /home/se/tmp/wtf/target/scala-2.12/wtf_2.12-0.1.0-SNAPSHOT.jar ...
[info] Done packaging.
[info] Running experimental.Test 
There are 1116 total classes
Subclasses1: List()
Subclasses2: List()
[success] Total time: 1 s, completed Mar 20, 2018 3:09:37 PM
```

jackkoenig, 2018-03-20:
Doing a little [RTFC of the ClassFinder source code](https://github.com/bmc/classutil/blob/366f7cc2f4254ca00094731f099fc932c78b197e/src/main/scala/org/clapper/classutil/ClassFinder.scala#L301) and checking the classpath it uses via:
```
println(finder.classpath)
```

I get

```
List(/usr/local/Cellar/sbt/1.1.1/libexec/bin/sbt-launch.jar)
```

It looks to me like it's not including the actual project (or any of the Scala libraries) on the classpath. Doing some digging to see if I can figure out a way to get a classpath with the project itself on it.

jackkoenig, 2018-03-20:
Alright I found something that seems to work. You can give `ClassFinder` a `Seq[File]` instead of having it figure that out on it's own. The following works:

```scala
package experiment

import org.clapper.classutil.ClassFinder
import java.net.URLClassLoader
import java.io.File

object Test extends App {

  val files = getClass.getClassLoader.asInstanceOf[URLClassLoader].getURLs.map(x => new File(x.getFile))
  val finder = ClassFinder(files)
  println(finder.classpath)
  val classes = finder.getClasses

  println(s"There are ${classes.size} total classes")

  val subclasses1 = ClassFinder.concreteSubclasses(classOf[experiment.MyAbstractClass], classes).toList
  println(s"There are ${classes.size} total classes")
  val subclasses2 = ClassFinder.concreteSubclasses("experiment.MyAbstractClass", classes).toList

  println(s"Subclasses1: $subclasses2")
  println(s"Subclasses2: $subclasses2")
}
```

There's probably a better way to express this but at least it's something.

chick, 2018-03-20:
To get the above to work in IntelliJ for me, I had to change the val files to be
```scala

  val files = getClass.getClassLoader.asInstanceOf[URLClassLoader].getURLs.flatMap { x =>
    try {
      val file = new File(x.getFile)
      if(file.exists()) {
        Some(file)
      }
      else {
        None
      }
    }
    catch {
      case e: Throwable =>
        None
    }
  }
```

seldridge, 2018-03-24:
I now realize that @jackkoenig was thinking about this from a similar direction: #558

seldridge, 2018-03-28:
I have an initial solution in #765 that should help concretize the discussion. 

This does two major things:
  - `ExecutionOptionsManager` now parses immutably where all command line options map to one or more annotations and/or transforms
  - Transforms that extend `ProvidesOptions` can inject options

I've left `commonOptions` and `firrtlOptions` in place as I view these as the data structure that the compiler is building up. However, these can no longer be modified mutably. This makes some of the API for writing tests a bit klunkier, but I think the immutability here makes it worthwhile.

For the second point, I find concrete subclasses of `ProvidesOptions` using @jackkoenig's `ClassFinder` approach from above. This works fine, however, `ClassFinder` is notoriously slow if you have to inspect everything. I'm currently, kludgily ignoring anything in the ".ivy2" cache to speed things up. However, that's an incorrect solution as external libraries should be able to define transforms that expose options. This may warrant some more discussion, but can be kicked down the road until FIRRTL libraries actually exist.

I've moved away from Annotations exposing options and rather having Transforms expose options. This was the original suggesting of #558 and I think it makes sense. Relatedly, I was not able to figure out a way to have a trait that gets mixed into an Annotation and then construct an instance of that annotation. It works fine if it's a singleton (NoTargetAnnotation), but anything else is a bit odd. It may be worthwhile to have a discussion if there should be an Annotation command line interface as opposed to what Transform writers want to expose.

seldridge, 2018-05-10:
I've thought about a couple of alternatives, inspired by some extrapolations from Chick last week, would be to more strictly delineate between an options manager, a driver, and the view of the options manager by the driver. These are outline below, from most complex to least:

### Type Classes

1. An `ExecutionOptionsManager` storing all annotations as an `AnnotationSeq`
2. An abstract `MoreOptions` trait that will provide additional options
3. An `OptionsView[T]` type class that allows for conversion of an `AnnotationSeq` to `Option[T]`

A tool that wants to take options (FIRRTL compiler, Chisel Compiler, Treadle) then defines:
1. A concrete subclass of `MoreOptions` defining it's command line options
2. A case class `X` defining tool options
3. A concrete `OptionsView[X]`

The tool driver (`firrtl.Driver`, `chisel.Driver`) then internally creates:
1. An options manager with its options, i.e., `ExecutionOptionsManager with MoreToolOptions`
2. A view of its options, i.e., `val myOptions = view[X]`

This has the effect of decoupling the subclassed options manager from it's tool options (case class `X`).

In Scala, this would look something like the following (possible errors follow, I'm just sketching this out). At the `ExecutionOptionsManager` level:

```scala
class ExecutionOptionsManager(args: Array[String], annotations: AnnotationSEq) {
  final val parser = new scopt.OptionParser[AnnotationSeq]
  lazy final val options: AnnotationSeq = parser.parse(args, annotations)
  def copy(args: Array[String], annos: AnnotationSeq) = ExecutionOptionsManager(args, annos)
}

/** Define additional scopt options only */
trait MoreOptions { this: ExecutionOptionsManager =>

  /** Additional parser options */
  def newOptions(p: scopt.OptionParser[AnnotationSeq]): Unit
  newOptions(parser)
}

/** Trait defining a conversion of AnnotationSeq to T */
trait OptionsView[T] {
  /** A conversion of [[AnnotationSeq]] to Option[T] */
  def view(implicit options: AnnotationSeq): Option[T]
}
```

Each tool, e.g., FIRRTL, would then provide concrete implementations:
```scala
/** Defines additional options */
trait HasFirrtlOptions extends MoreOptions { this: ExecutionOptionsManager => 
  def newOptions(p: scopt.OptionParser[AnnotationSeq]): Unit = {
    parser.opt[String]("top-name")
    // ...
  }
}

/** The output of a view */
case class FirrtlOptions(
  topName: String,
  // ... 
)

/** The implementation of a view */
implicit object FirrtlOptionsView extends OptionsView[FirrtlExecutionOptions] {
  def view(implicit options: AnnotationSeq): Option[FirrtlOptions] = {
    // ...
  }
}
```

The Driver, e.g., `firrtl.Driver` or `chisel3.Driver`, would then create a view of the annotations that only exists inside the driver and not in the options manager.
```scala
object ChiselDriver {
  implicit val options = new ExecutionOptionsManager(args) with FirrtlOptions with ChiselOptions
  val chiselOptions = view[ChiselOptions]
  val firrtlOptions = view[FirrtlOptions]
}
```

### Implicit Conversions

This would work very much like above, except an implicit conversion would be defined for `AnnotationSeq` to `X` instead of a type class.

### Complete Driver Freedom

The observation is that the processing of `AnnotationSeq` into usable options already happens in every `Transform`. Why should we overlay a complicated approach to extracting information from `Annotation`s for a tool? 

Here, the `ExecutionOptionsManager` just stores an `AnnotationSeq`. All logic related to getting information out of this related inside a `Driver` is ignored. We may opt for type classes in FIRRTL, Chisel, or in certain Transforms, but this is not the suggested way of doing things. This just migrates code from existing `ExecutionOptionsManager` subclasses into each tool's respective driver.

jackkoenig, 2018-05-23:
This isn't directly related to the proposal above, but one thing I've been thinking about with this pattern is that people will commonly want to find and remove an annotation. In the case that a Driver or particularly enthusiastic Transform is using the TypeClass and view approach, I suspect they will provide a mixin for their annotations and can just do a `filterNot(_.isInstanceOf[MyMixin])`, but another pattern I see a lot is:
```scala
val myAnno = annotations.collectFirst { case a: MyAnnotation => a }.head
```
Which in some cases should be consuming the annotation and removing it. Perhaps we could add a method to AnnotationSeq to allow something to get and remove:
```scala
val (myAnno, annosx) = annotations.extractFirst { case a: MyAnnotation => a }
```
I suspect this might result in stuff like:
```scala
val (anno1, annosx1) = annotations.extractFirst { case a: MyAnno1 => a }
val (anno2, annosx2) = annosx1.extractFirst { case a: MyAnno2 => a }
val (anno3, annosx3) = annosx2.extractFirst { case a: MyAnno3 => a }
```
Which isn't great. 

With a value class per Tuple we could maybe enable this monstrosity:
```scala
val (anno1, (anno2, (anno3, annosx))) = annotations.extractFirst { case a: MyAnno1 => a }
                                                   .extractFirst { case a: MyAnno2 => a }
                                                   .extractFirst { case a: MyAnno3 => a }
```
Any ideas on a better pattern?

seldridge, 2018-06-08:
Consensus up to this point is that the TypeClass/view pattern is the way to go. I will work on a PR (or update the existing PR) demonstrating this pattern for FIRRTL and update the PR for Chisel.

@jackkoenig: No current thoughts on a better pattern for this... I was thinking about doing this with repeated application of `partition`, but I don't think anything came of it. I'll think about this more.

seldridge, 2018-06-12:
I have a prototype of the Typeclass/View pattern in #765 that's up-to-date with master.

Through deprecated interfaces this continues to work with Chisel3 (and assumedly Treadle, excepting it's REPL). I'll update the Chisel3 PR accordingly.

[The Scala code in the comment above](https://github.com/freechipsproject/firrtl/issues/764#issuecomment-388183187) now reflects what is going on to enable this pattern and what this pattern looks like.

ducky64, 2018-06-15:
Feedback notes from today's live mega-PR review, discussion including @seldridge @jackkoenig @ucbjrl @chick
- Augmented options parser seems useful and fine. Perhaps name it something more descriptive.
- HasChiselExecutionOptions trait: you really only need it if building an ExecutionOptionsManager. But can define a new class that pre-mixes things in for you.
- To add a new option, need to know the existing ones. You can have conflicts, can consider style guidelines for external libraries (like having a prefix).
- If you're building your own ExecutionOptionsManager, you need to mix in all the right traits, but a failure should be a static (compile-time) failure. Can we have these propagate more automatically? This has been a problem.
- What about a more drastic refactoring that gets rid of the execution options manager in favor of annotations? Just want some way to parse arguments?
  - Things that query options should query annotations instead? Then the whole managers discussion may be moot?
- People are currently using ExecutionOptionsManager, especially where that is invoked programmatically. Like in some Berkeley chip tapeouts. Tried to preserve as much of the existing semantics as possible, though not 100%.
- Why chiselOptions can't be a var (making it a val breaks compatibility)? We could, but probably shouldn't?
- All the Chisel Driver stuff works, only stuff using ExecutionOptionsManager explicitly might break.
- FirrtlOption and ChiselOption are just type tags. Perhaps the functionality of the option should be encoded in the object itself, instead of in the viewer?
  - Can we seal some options? Some options exist in a different file, but can refactor to seal things.
- Still multiple sources of options: annotations file (which itself is an option), command line options. Confusing?
- Validation happens in both scopt and on annotations sequence
- Must bump Chisel as well, Chisel tests pass with a Chisel PR that @seldridge has. Treadle, firrtl-interpreter, and testers need to be bumped as well.
- Is there a more elegant way to define the top name (which is needed for the target-dir, which now depends on the circuit compile)? Is there a better future path for top name? Can we start deprecating stuff? Larger issue with Circuit and scopes?
- Where can common options shared between libraries be put? (example: between firrtl and treadle, they might be 90% the same)
- Is there a cleaner way to specify "add all the transforms"?

edwardcwang, 2018-06-15:
Could someone summarize the effects/results/changes for FIRRTL/Chisel users, particularly those using the FIRRTL jar as an executable as in rocket-chip's build system and most of the Berkeley chip tapeouts?

seldridge, 2018-06-15:
@ducky64: Thanks for taking all these notes! I'll try to work up some responses so that the wider community can see what's going on with this.

@edwardcwang: There are two levels of impact, depending on what you're doing:
1. __Using FIRRTL fat jar as an executable__: No impact. I.e., Rocket Chip should be unaffected.
2. __Explicitly Constructing an `ExecutionOptionsManager`__: You need to modify your code to not mutate options since `var (chiselOptions, commonOptions, firrtlOptions)` either no longer exist are immutable. This results in you being highly advised to pass all options directly to the options manager. Most of the required modifications are pretty basic and make the construction of an options manager use the same API as the FIRRTL fat jar. Here's one example: https://github.com/freechipsproject/firrtl/pull/765/files#diff-5663f8e864f20e181ba77401a196b733L522.

Note that for _2_ you can always build a complete `ExecutionOptionsManager` (that will not error when you view it) from an incomplete `ExecutionOptionsManager` (that will error when you view it) with additional options that make in complete. The problem only comes if you want to "view" the annotations that an options manager collects before it has enough options to do this (it is an "incomplete" options manager). One example of doing this is how one of the old Chisel3 `Driver.execute` methods had to be modified as the FIRRTL source is not initially available: https://github.com/freechipsproject/chisel3/pull/804/files#diff-f7a19e5d7b232a5be3545b8ed7de50a5R168

seldridge, 2018-06-15:
> FirrtlOption and ChiselOption are just type tags. Perhaps the functionality of the option should be encoded in the object itself, instead of in the viewer?
Can we seal some options? Some options exist in a different file, but can refactor to seal things.

I think that making `FirrtlOption` more meaningful is the way to go. I have an example of moving option parsing into the annotation (only for `TopNameAnnotation`) [in this commit](https://github.com/freechipsproject/firrtl/pull/765/commits/a79c999693bbfcacbded298da913620656bcb3e0). This has the huge benefit of moving all the information about a specific annotation/option into one place. @jackkoenig: what do you think?

seldridge, 2018-06-19:
> - Augmented options parser seems useful and fine. Perhaps name it something more descriptive.

Refactored into two traits, `TerminateOnExit[T]` and `DuplicateHandling[T]`

> - HasChiselExecutionOptions trait: you really only need it if building an ExecutionOptionsManager. But can define a new class that pre-mixes things in for you.

This was more critical before as the old equivalent of an options view, `var chiselOptions` or `var firrtlOptions`, was contained within the trait. However, with the Typeclass/view pattern, the options manager really is just a (command line) options manager. All the interpretation of the options as some type is handled by the Typeclass/view pattern. If we allow for fully breaking changes without deprecation, then the type that you're viewing provides the methods that you would formerly exist in a trait that you're mixing into an options manager.

> - To add a new option, need to know the existing ones. You can have conflicts, can consider style guidelines for external libraries (like having a prefix).

To handle this automatically is painful, but I have some ideas on how to do it.
Roughly, we would need to override the `def opt` methods of `OptionParser` to first check to see if a long and/or short option existed.
If it did exist, we use reflection to figure out what package it's in and mangle the option accordingly.
In the short term, I think a style guide is the way to go.

> - If you're building your own ExecutionOptionsManager, you need to mix in all the right traits, but a failure should be a static (compile-time) failure. Can we have these propagate more automatically? This has been a problem.

This should be the same issue as the second bullet. The Typeclass/view pattern avoids the needs for mixing in a bunch of traits to define a parameter's type. However, that should largely mean that any API that passes an options manager in as a trait (with the exception of `Driver.execute`) should break.

> - What about a more drastic refactoring that gets rid of the execution options manager in favor of annotations? Just want some way to parse arguments?

I am totally in favor of this, however, that would break the API even more than the companion PR already does. This is targeted for 1.2.0, though...

Thoughts?

> - Things that query options should query annotations instead? Then the whole managers discussion may be moot?

Correct. They may either query annotations directly (just like most `Transform`s do) or they can use the `Typeclass/view` pattern. This pattern canonicalizes the approach for converting a lot of annotations to some type that encapsulates all information that a driver or transform needs.

> - People are currently using ExecutionOptionsManager, especially where that is invoked programmatically. Like in some Berkeley chip tapeouts. Tried to preserve as much of the existing semantics as possible, though not 100%.

Similar to what @edwardcwang was getting at. The only user set that should see an impact is the set of people manually manipulating `commonOptions`, `firrtlOptions`, `chiselOptions`, etc. I anticipate that this set is minimal. If you're only calling the `Driver` API, you should be fine.

> - Why chiselOptions can't be a var (making it a val breaks compatibility)? We could, but probably shouldn't?

This needs to be a deprecated `lazy val` or a `def` that is backed by a `view[ChiselExecutionOptions]`. The `view` can fail (currently with an exception). I could catch this exception and then emit the default `ChiselExecutionOptions`, but this would also break the API. 

Unfortunately, I think there's no good way around this.

> - All the Chisel Driver stuff works, only stuff using ExecutionOptionsManager explicitly might break.

Correct.

> - FirrtlOption and ChiselOption are just type tags. Perhaps the functionality of the option should be encoded in the object itself, instead of in the viewer?

I've updated this to move option parsing logic inside of the Annotation. The `FirrtlOption` and `ChiselOption` traits now are `sealed` and extend `HasScoptOption` (which has the abstract method `addOptions`). This has the benefit of making the trait do something and, more critically, moving the parsing logic directly next to the annotation that it will be emitting.

> - Can we seal some options? Some options exist in a different file, but can refactor to seal things.

Now fixed.

> - Still multiple sources of options: annotations file (which itself is an option), command line options. Confusing?

This is possibly confusing. Perhaps this can be remedied with sufficient documentation in the wiki?

Annotations are the underlying information. Command line option parsing is a function with signature `Array[String] => AnnotationSeq`. As long as people grok that functional relationship, I think this is relatively straightforward.

> - Validation happens in both scopt and on annotations sequence

Validation _should_ only happen when constructing a `view`. However, I have done no kludgery to prevent users trying to do validation directly on the command line options. This is totally reasonable in some situations, but should generally be avoided. As a consequence, FIRRTL avoids all validation of options and leaves that until a `view` is constructed.

> - Must bump Chisel as well, Chisel tests pass with a Chisel PR that @seldridge has. Treadle, firrtl-interpreter, and testers need to be bumped as well.

[... cut ...]
This would be accomplished with a new `view` that shares these options. You could re-use the option parsing `HasFirrtlExecutionOptions` and `HasTreadleExecutionOptions` or not.

> - Is there a cleaner way to specify "add all the transforms"?

I assume this is about adding the options of transforms? If so, then yes. That is complicated as you need to find transforms that provide options. Reflection looking for sealed traits would work nicely, but it's not possible to do then mix in the trait in multiple files. This would require putting all annotations for a tool (e.g., FIRRTL) in one file if they were going to add options.

seldridge, 2018-07-30:
@chick has been working on a Treadle refactor to align with this in the [freechips/treadle:options-as-annos](https://github.com/freechipsproject/treadle/tree/options-as-annos) branch.

Feedback on this is available here: https://gist.github.com/seldridge/9ae79b247d4b389a066f71dc9aa6777d

chick, 2018-09-18:
@seldridge I am back to working on testers and I have a chicken and egg problem with Logger. The logger needs to get fired up before the dut annotation has been processed, but I can't view the Firrtl options without firrtl source.  Seems like either we have to ease the requirement for firrtl source or probably better but a side trip would be to break out the Logger options from firrtl (They really should be separate anyway, Logger is a separate package that just lives in Firrtl because it's the bottom of the stack.  Do you have another idea on how to proceed here?

chick, 2018-09-18:
While I am at it I should note that my backward compatibility strategy for testers is to continue to support the creation of old style options manager stuff (with deprecations of course) duplicating the old options classes and manually adding annotation generators from those. I've got things compiling with that strategy but not running yet

seldridge, 2018-09-18:
@chick: 
  - Logger Chicken and Egg: Breaking out logger makes sense. I think the (Options are Annotations) canonical way to do this is to add a logger options view. Then you can safely inspect the command line options at any time without having to go through the FIRRTL viewer's strict requirements on what the annotations are supposed to look like.
  - Backward compatibility: Options as Annotations is so damn different that this compatibility layer is nearly impossible to write without replicating the old way and deprecating all of it. I'm worried about making sure that this behaves as before and that it behaves like the new stuff (possibly not as problematic). One thing that we could do is to have a special viewer that does options views with any validation checking disabled (but not make this public). I think that may make the compatibility layer easier to write. That way, you can add annotations to an incomplete view.

closed 2019-04-26 by jackkoenig