chipsalliance/firrtl #802: Delete icache or shrink it to 1 line and bypass icache in rochet chip

xiwangwl, 2018-05-02:
I want to delete the icache to shrink size. And I got the information that icache cannot be deleted.
I start to config it to be 1 line. What I change is the file src/main/scala/subsystem/Configs.scala:line 91
icache = Some(ICacheParams(
rowBits = site(SystemBusKey).beatBits,
nSets = 1,
nWays = 1,
nTLBEntries = 4,
blockBytes = site(CacheBlockBytes)))))

And it report errors:
[info] [55.847] Done elaborating.
[success] Total time: 138 s, completed May 2, 2018 1:28:27 PM
mkdir -p /root/riscv/freedom/rocket-chip/vsim/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /root/riscv/freedom/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /root/riscv/freedom/rocket-chip/vsim/generated-src/freechips.rocketchip.system.TinyConfig.fir -o /root/riscv/freedom/rocket-chip/vsim/generated-src/freechips.rocketchip.system.TinyConfig.v -X verilog --infer-rw TestHarness --repl-seq-mem -c:TestHarness:-o:/root/riscv/freedom/rocket-chip/vsim/generated-src/freechips.rocketchip.system.TinyConfig.conf -faf /root/riscv/freedom/rocket-chip/vsim/generated-src/freechips.rocketchip.system.TinyConfig.anno
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
Exception in thread "main" firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
at firrtl.Utils$.error(Utils.scala:399)
at firrtl.Utils$.throwInternalError(Utils.scala:180)
at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:245)
at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:191)
at logger.Logger$$anonfun$makeScope$1.apply(Logger.scala:129)
at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
at logger.Logger$.makeScope(Logger.scala:127)
at firrtl.Driver$.execute(Driver.scala:191)
at firrtl.Driver$.execute(Driver.scala:299)
at firrtl.Driver$.main(Driver.scala:315)
at firrtl.Driver.main(Driver.scala)
Caused by: firrtl.passes.PassExceptions:
firrtl.passes.CheckWidths$BitsWidthException: @[ICache.scala 132:40:freechips.rocketchip.system.TinyConfig.fir@102436.4]: [module ICache] High bit 6 in bits operator is larger than input width 6 in bits(s1s3_slaveAddr, 6, 6).
firrtl.passes.CheckWidths$BitsWidthException: @[ICache.scala 228:31:freechips.rocketchip.system.TinyConfig.fir@102488.4]: [module ICache] High bit 6 in bits operator is larger than input width 6 in bits(s1s3_slaveAddr, 6, 2).
firrtl.passes.CheckWidths$BitsWidthException: @[ICache.scala 132:40:freechips.rocketchip.system.TinyConfig.fir@102550.4]: [module ICache] High bit 6 in bits operator is larger than input width 6 in bits(s1s3_slaveAddr, 6, 6).
firrtl.passes.PassException: 3 errors detected!
at firrtl.passes.Errors.trigger(Passes.scala:45)
at firrtl.passes.CheckWidths$.run(CheckWidths.scala:112)
at firrtl.passes.Pass$class.execute(Passes.scala:24)
at firrtl.passes.CheckWidths$.execute(CheckWidths.scala:11)
at firrtl.Transform$$anonfun$4.apply(Compiler.scala:242)
at firrtl.Transform$$anonfun$4.apply(Compiler.scala:242)
at firrtl.Utils$.time(Utils.scala:185)
at firrtl.Transform.runTransform(Compiler.scala:242)
at firrtl.SeqTransformBased$$anonfun$runTransforms$1.apply(Compiler.scala:294)
at firrtl.SeqTransformBased$$anonfun$runTransforms$1.apply(Compiler.scala:294)
at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
at scala.collection.immutable.List.foldLeft(List.scala:84)
at firrtl.SeqTransformBased$class.runTransforms(Compiler.scala:294)
at firrtl.SeqTransform.runTransforms(Compiler.scala:298)
at firrtl.SeqTransform.execute(Compiler.scala:304)
at firrtl.Transform$$anonfun$4.apply(Compiler.scala:242)
at firrtl.Transform$$anonfun$4.apply(Compiler.scala:242)
at firrtl.Utils$.time(Utils.scala:185)
at firrtl.Transform.runTransform(Compiler.scala:242)
at firrtl.Compiler$$anonfun$13$$anonfun$apply$10.apply(Compiler.scala:461)
at firrtl.Compiler$$anonfun$13$$anonfun$apply$10.apply(Compiler.scala:461)
at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
at scala.collection.immutable.List.foldLeft(List.scala:84)
at firrtl.Compiler$$anonfun$13.apply(Compiler.scala:461)
at firrtl.Compiler$$anonfun$13.apply(Compiler.scala:461)
at firrtl.Utils$.time(Utils.scala:185)
at firrtl.Compiler$class.compile(Compiler.scala:460)
at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:139)
at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:231)
... 8 more
make: *** [/root/riscv/freedom/rocket-chip/vsim/generated-src/freechips.rocketchip.system.TinyConfig.v] Error 1

aswaterman, 2018-05-02:
You probably need to make it two lines (so that it can be indexed with a non-null index).

seldridge, 2018-12-18:
See @aswaterman's comment. Likely the cache needs to be > one line. Feel free to reopen if this is still an issue.

closed 2018-12-18 by seldridge