chipsalliance/firrtl #827: Support multi-argument cat() in the IR?

edwardcwang, 2018-06-19:
Can we support (at some level of firrtl) `cat()` with multiple arguments?

Would be nice for a few reasons (that I can think of at the top of my head) - this is by no means limited to these cases:

1. Easier to emit from tests/front-ends - would make writing bit concatenations in tests less complicated (e.g. no need to write verbose statements like https://github.com/freechipsproject/firrtl/blob/535d8025412a64471d8cc9c315505a8e2cbddbe0/src/test/scala/firrtlTests/UnitTests.scala#L43)

2. Cleaner emitter - e.g. in emitting to Verilog, etc we can emit `{b3, b2, b1, b0}` and avoid the creation of a whole bunch of intermediate nodes

3. Slightly easier to read/debug: e.g. `cat(cat(cat(cat(cat(cat(cat(b7, b6), b5), b4), b3), b2), b1), b0)` vs `cat(b7, b6, b5, b4, b3, b2, b1, b0)`

jackkoenig, 2018-06-30:
Related to https://github.com/freechipsproject/firrtl/issues/624

The issues of how primops are emitted is separate from the internal data structure. There's no reason we can't combine nested cats for Verilog emission. 

Currently, all primops are required to have 2 arguments which makes handling them consistent. If we were to allow more, we would have to handle it all over the place. I feel less strongly against this than I used to, but we should be careful not to conflate the readability of the Verilog with the IR datastructure.

albert-magyar, 2020-03-02:
The emission concerns were resolved in #851, so I'm closing this for now, as that resolves the immediate incentive to increase arity in primops.

closed 2020-03-02 by albert-magyar