chipsalliance/firrtl #836: Combinational Loop not detected for self-connections

albertchen-sifive, 2018-06-27:
**Type of issue**: bug report

 * **What is the current behavior?**
firrtl input:
```
circuit loop :
  module loop :
    output y : UInt<8>

    wire w : UInt<8>
    w <= w
    y <= w
```
emitted verilog:
```
module loop( // @[:combLoop.fir@2.2]
  output [7:0] y // @[:combLoop.fir@3.4]
);
  wire [7:0] w; // @[:combLoop.fir@5.4]
  assign y = w; // @[:combLoop.fir@7.4]
  assign w = w; // @[:combLoop.fir@6.4]
endmodule
```

  * **What is the expected behavior?**
throw `CombLoopException`

**Impact**: no functional change

**Development Phase**: request

albert-magyar, 2018-06-27:
I never thought about this case :) Do existing Verilog consumers like synthesis or fancy linters flag this?

This is a trivial fix, I can do this today.

albertchen-sifive, 2018-06-27:
I'm not sure about what the tools do, but I'm trying use combinational loop detection to optimize vecs of bools and ran into this issue.

edwardcwang, 2018-06-27:
One synthesis tool just optimizes connecting a wire to itself away:

```
module loop(y);
  output [7:0] y;
  wire [7:0] y;
endmodule
```

albert-magyar, 2018-06-28:
I addressed this in #837 and added your example code as the test case, @albertchen-sifive

PR #837 (2018-06-29): Make CheckCombLoops find combinational nodes with self-edges
Fixes #836
closed 2018-06-29 by jackkoenig