chipsalliance/firrtl #840: Issue with RemoveChirrtl smem

mehnadnerd, 2018-07-02:
**Type of issue**: bug report

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **If possible, please directly provide the FIRRTL file or a relevant fragment if the file is large**
These two Chisel statements:
`outBundle:= mem.read(currentItem)`
`outBundle := mem.read(Mux(currentItem =/= RegNext(currentItem), currentItem, currentItem))`

And corresponding CHIRRTL:
`read mport _T_428 = mem[currentItem], clock`
`node _T_408 = neq(currentItem, _T_407)
node _T_409 = mux(_T_408, currentItem, currentItem)
read mport _T_410 = mem[_T_409], clock`

Do not result in the same emitted Verilog. In fact, the first leads to an issue where the memory read is enabled only when currentItem changes. However, since there is a one-cycle delay for this mem, this means that the read will only update after the next address is given, not one cycle later!

By adding in the unnecessary Mux, the correct behaviour emerges, which leads me to suspect that it might have to do with optimisation of the read, but looking at the RemoveChirrtl transform, I am unable to find the cause.

  * **Please tell us about your environment:**
    a6

**What is the use case for changing the behavior?**
Circuits working like the documentation suggests

**Impact**: API modification

**Development Phase**: request

**Other information**
#702 #727 both include similar things, although 702 appears to be dealing with the clock rather than the enable. This would hopefully be fixed by 727, although it is important to ensure this bug does not remain in the transition.

I can provide the full chisel/firrtl/verilog to people within BAR if need be, as well as the full trace of the firrtl run. It is quite large (392 MB), but the important part is before and after the RemoveChirrtl transform.

mehnadnerd, 2018-07-03:
Issue appears to be with using a register as the address, adding in a temporary wire fixes it as well.

mehnadnerd, 2018-07-03:
Adam's plan: Insert a temporary wire between register and address

albert-magyar, 2019-10-29:
Closed and absorbed into #727

closed 2019-10-29 by albert-magyar