chipsalliance/firrtl #850: FIRRTL Exception, cause unclear (with test case)

hutch31, 2018-07-20:
**Type of issue**: bug report 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **If possible, please directly provide the FIRRTL file or a relevant fragment if the file is large**

Files attached as whbug.zip, extracts to whbug directory.

  * **What is the current behavior?**
[error] (run-main-0) firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error] firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error] 	at firrtl.Utils$.error(Utils.scala:396)
[error] 	at firrtl.Utils$.throwInternalError(Utils.scala:177)
[error] 	at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:236)
[error] 	at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:209)
[error] 	at logger.Logger$$anonfun$makeScope$1.apply(Logger.scala:129)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at logger.Logger$.makeScope(Logger.scala:127)
[error] 	at firrtl.Driver$.execute(Driver.scala:209)
[error] 	at chisel3.Driver$.execute(Driver.scala:182)
 (long stack trace)

  * **What is the expected behavior?**

There is likely a bug in the Chisel code but difficult to track down with the provided error info.  Expected behavior would be error report with line info.

  * **Please tell us about your environment:**
    (examples)
    - version: `3.1`
    - OS: `MacOS 10.13, using IntelliJ, Java 1.8`

**What is the use case for changing the behavior?**

**Impact**: unknown

**Development Phase**: request | proposal

**Other information** 

Full code attached.  To reproduce:

unzip whbug.zip
cd whbug
make

[whbug.zip](https://github.com/freechipsproject/firrtl/files/2215725/whbug.zip)

hutch31, 2018-07-21:
Narrowed down the issue to the McPacketBuffer block, reduced environment below has only essential files.

[whbug2.zip](https://github.com/freechipsproject/firrtl/files/2215900/whbug2.zip)

jackkoenig, 2018-07-21:
With that workaround applied, you're going to run into a combinational loop exception in:
```scala
  num_valid := 0.U
  next_port := 0.U
  for (i <- 0 until clients) {
    when (port_vec(i)) {
      num_valid := num_valid + 1.U
      next_port := i.U
    }
  }
```
The issue here is that `num_valid := num_valid + 1.U` is a combinational loop. In Chisel, last connect semantics make later connections overrule prior connections and there's no notion of the "current" value connected to a wire. For example:
```scala
val x = Wire(UInt())
x := 1.U
io.out := x // last connect wins, x is 2.U
x := 2.U
```
Last connect semantics make it as if the connection to 1.U never occurred.

You can accomplish what it looks like you're trying to do by using a `var` as kind of an elaboration-time mutable "pointer" to some hardware. You can thus update this pointer each iteration of the for loop:
```scala
  var num_valid_var = 0.U
  next_port := 0.U
  for (i <- 0 until clients) {
    num_valid_var = Mux(port_vec(i), num_valid_var + 1.U, num_valid_var)
    when (port_vec(i)) {
      next_port := i.U
    }
  }
  num_valid := num_valid_var
```
 However, there are more Scala-y ways to accomplish this operation of counting the number of valid bits. The best way is the build-in `PopCount` utility:
```scala
num_valid := chisel3.util.PopCount(port_vec)
```

hutch31, 2018-07-22:
@jackkoenig thanks for the pointer on PopCount, and more importantly the heads up on reassignment semantics.  The late assignment was actually a cut-and-paste error from refactoring my code.

After moving the assignments up above the FSM block and changing the for loop to use PopCount and PriorityEncoder, the block compiles clean.

jackkoenig, 2018-07-22:
No problem at all, thanks for your help in tracking down this issue in providing a test case! Glad you were able to workaround the issue. I’ve filed an issue (#852) to track the bug you ran into here so I’m going to close this one.

closed 2018-07-22 by jackkoenig