chipsalliance/firrtl #856: Why is ToMemIR restricted to at most 1 read or RW port?

edwardcwang, 2018-07-30:
**Type of issue**: bug report | feature request

From [ToMemIR.scala](https://github.com/freechipsproject/firrtl/blob/1dcf9907eaa2f5fd2bc1e5a7dafeb1ae4b8e1434/src/main/scala/firrtl/passes/memlib/ToMemIR.scala#L9):

```scala
/** Annotates sequential memories that are candidates for macro replacement.
  * Requirements for macro replacement:
  *   - read latency and write latency of one
  *   - only one readwrite port or write port
  *   - zero or one read port
  */
```

ReplSeqMem seemed to be silently doing nothing/failing on sequential memories with >1 read port, so upon further investigation it turns out that `ToMemIR` has an artificial restriction (e.g. `m.readers.length == 1`) on the number of ports it supports.

Just as a local experiment, I lifted the port count restriction and nothing bad immediately jumped out at me. Was there a particular reason that the pass is restricted to only 1 read or 1 write/rw port for a given memory?

The restriction is inherited from the original PR (#265) that introduced this functionality and a quick search through the history of this code + the original PR didn't yield any obvious explanations.

**What is the use case for changing the behavior?** I wanted to use ReplSeqMem to blackbox a memory that turned out to have >1 read port but the pass silently ignored it until I dug a bit into the memory-replacement passes' code in FIRRTL.

**Impact**: API addition (no impact on existing code)

**Development Phase**: request | proposal

shunshou, 2018-07-30:
> Assigned #856 <https://github.com/freechipsproject/firrtl/issues/856> to
> @shunshou <https://github.com/shunshou>.

All of the ASIC memory compilers we've worked with only generate memories
with 1 read port (it's been awhile since I've looked at FPGAs, so I'm not
sure if this is true...). The intent of the original PR was to get
something usable ASAP for tapeout. I suppose generalizing this would be
useful, but that involves adding new transform steps (or post processing)
that are not agnostic to the memory compiler used. Also, it might be best
practice to have designers try to only use 1 read port if possible, or have
them explicitly describe how they can utilize multiple 1 read port memory
for multi-read purposes (see
https://www.eetimes.com/author.asp?section_id=36&doc_id=1320067). At the
very least, it should loudly fail...

edwardcwang, 2018-07-30:
I see. Perhaps it would be possible to have a flag for the mode - singleport (<= 1 r/w/rw port, would be the default) and multiport (allow >1 read/write/readwrite port)? (perhaps with annotations per module) That way if I'm expecting a memory to be mapped to multiple ports, we can set a flag to allow that in our design.

I think the changes that involve converting/double-pumping dual-port memories into multi-port memories could be done in separate passes (I think Albert did a double-pumping pass and MacroCompiler or sibling passes could do some postprocessing of the type described), so it seems like these changes would work together - downstream passes/tools can do more adaptation.

shunshou, 2018-07-30:
I don't disagree. It's expanded functionality above what currently exists. If it serves to satisfy your use case, then definitely adding downstream passes/tools would be the way to go. Since I don't expect that too many people intentionally want this at least right now (I think there's some notion when designing memory interfaces that you typically have one read, and people will have this in mind when building out infrastructure), it's probably worth it to spend some time to think about integrating it as a FIRRTL pass rather than a totally separate tool. If there are more immediate use cases, then I presume designing a one-off tool would be faster...

edwardcwang, 2018-07-30:
Yeah, concretely in the immediate/short-term, I think expanding `ToMemIR` to support >1 read and >1 write/readwrite would be a good step, perhaps enabled via an extra flag an annotation or something.

If this sounds good I can look into doing a PR at some point.

shunshou, 2018-07-30:
Sounds fine as long as the user is aware of what's happening under the hood. People who work on super high-level code (which I guess is the vision of Chisel) might not be aware of fundamental hardware limitations, and until a solid alternative solution is put into place, it's better to shock them earlier in their design process instead of forcing them to scramble to find a last minute multi-port solution. Even then, if the alternative (tool or compiler pass) naively doubles the area over a single-ported solution, it might be good to present some kind of warning message. In such cases, more often than not, I think they could've re-architected their block not to use multi-port memories in the first place.

seldridge, 2018-12-18:
There's a two-part fix here:
  - Throw an exception if the user violates the requirements, don't just pass along statements (1.2.0)
  - Refactor ReplSeqMem to support this (1.x)

juliusbaxter, 2019-04-01:
Did this result in any updates?

edwardcwang, 2019-04-02:
I don't believe there have been further updates on this issue yet.

kammoh, 2019-06-20:
Is this issue on the radar at all? What is the recommended workaround to use dual-port memories?

sequencer, 2019-12-15:
@kammoh 
maybe only BlackBox can workaround :(
just come across to this issue again.

kenzhang82, 2020-09-02:
Any update on this issue yet?