chipsalliance/firrtl #860: IR improvement : Seq[Statement] instead of Statement

doofin, 2018-08-01:
**Type of issue**: other enhancement

The current IR of the  module is : `case class Module(info: Info, name: String, ports: Seq[Port], body: Statement) ` ,How about `case class Module(info: Info, name: String, ports: Seq[Port], body: Seq[Statement]) ` ? This would be much easier to work with and prevent some unexpected runtime errors

edwardcwang, 2018-08-01:
Usually in IRs they have a Block statement like https://github.com/freechipsproject/firrtl/blob/f0beb8e9efb57dc4aef08103426bd28cf8445147/src/main/scala/firrtl/ir/IR.scala#L291 to handle the case where a statement is a sequence of other statements. Perhaps one of the FIRRTL designers can chime in if they have any specific insights.

azidar, 2018-08-01:
@edwardcwang is correct. If you want to replace a single statement with multiple statements, its easiest to just add a Block node, rather than figure out how to append/insert into the parent Seq. In addition, we have the Conditionally node, which enables nesting of Statements, so if you are relying on only iterating through the module's seq of Statements, its possible to miss some statements anyways.

doofin, 2018-08-01:
What i want is to insert some stmts into existing stmts,this currently does not work
```scala
  val in2 = """
circuit tag_array_ext :
  module tag_array_ext :
    input RW0_clk : Clock
    input RW0_addr : UInt<6>
    input RW0_wdata : UInt<80>
    output RW0_rdata : UInt<80>
    input RW0_en : UInt<1>
    input RW0_wmode : UInt<1>
    input RW0_wmask : UInt<4>
             
    mem_0_0.clk <= RW0_clk
    mem_0_0.addr <= RW0_addr
"""

  val mytransform = new Transform {
    override def inputForm: CircuitForm = ChirrtlForm

    override def outputForm: CircuitForm = ChirrtlForm

    override protected def execute(state: CircuitState): CircuitState = {
      println("mytransform")
      val circInsr = firrtl.Parser.parse(in2)
      val mod = circInsr.modules.head.asInstanceOf[Module]

      val portsOri             = state.circuit.modules.head.asInstanceOf[Module]
      val portsAdd1: Seq[Port] = mod.ports
      val blocked  = Block(Seq(mod.body))
//      val portsAdd = 1 to 10 zip portsAdd1 map { case (i, p) => p.copy(name = "nm_" + i) }
      val cirstmts = state.circuit.copy(modules = Seq(mod.copy(body = blocked)))
      state.copy(circuit = cirstmts)
    }
  }
```

edwardcwang, 2018-08-01:
Would it be possible for you to explain what your transform is intended to accomplish in plain language?

seldridge, 2018-12-18:
The correct way to do this is to operate on the `Block`. Feel free to re-open if this is still an issue.

closed 2018-12-18 by seldridge