chipsalliance/firrtl #866: Do we ever need `Dshlw` and `Shlw`?

donggyukim, 2018-08-03:
**Type of issue**:  other enhancement

**What is the use case for changing the behavior?**
Seems like `Dshlw` and `Shlw` are not necessary anymore while these two are introduced by the `PadWidths` pass:
https://github.com/freechipsproject/firrtl/blob/67eb4e2de6166b8f1eb5190215640117b82e8c48/src/main/scala/firrtl/passes/PadWidths.scala#L44

The problem is if we apply `PadWidths`, generate high firrtl code, and run high firrtl passes again, we run into match errors like:
```
[error] (run-main-2d) scala.MatchError: shlw (of class firrtl.Shlw$)
[error] scala.MatchError: shlw (of class firrtl.Shlw$)
[error]         at firrtl.passes.CheckHighForm$.checkHighFormPrimop$1(Checks.scala:75)
[error]         at firrtl.passes.CheckHighForm$.firrtl$passes$CheckHighForm$$checkHighFormE$1(Checks.scala:152)
```

However, there's no point of having these two operators anymore in the current firrtl. Just search them  in github, and they'll appear only three times(in their definition, `PadWidths`, and `Emitter`).

**Impact**: no functional change

**Development Phase**: proposal

**Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

jackkoenig, 2018-08-06:
Yeah this is a good question. I thought they had some differing semantics from `Dshl` and `Shl` but they appear to be emitted the exact same way, @azidar are these still needed?

donggyukim, 2018-08-06:
BTW, I recently noticed that `PadWidths` is not supposed to be applied before other high-level passes, which is frustrating when I need to recover all verilog signals for my transforms. Is it a little bit relevant?

seldridge, 2018-12-18:
A fix to #270 will take care of this.

albert-magyar, 2020-03-02:
Subsumed by #1410

closed 2020-03-02 by albert-magyar