chipsalliance/firrtl #871: Better Error Message for Uninferred Width

jackkoenig, 2018-08-20:
**Type of issue**: feature request

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **If possible, please directly provide the FIRRTL file or a relevant fragment if the file is large**
  * **What is the current behavior?**
The error message just gives the module name, it doesn't name what component didn't have it's width inferred. Source locators help here, but currently Chisel doesn't grab source locators for ports and if you're using ProtoBuf, there is no FIRRTL source locator.

  * **What is the expected behavior?**
It would be nice if FIRRTL could report what component didn't have it's width inferred. This error usually would also be caught by CheckInitialization because it usually means the component was left unconnected. 

**What is the use case for changing the behavior?**

Better error reporting

**Impact**: no functional change

**Development Phase**: request

**Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

ccelio, 2018-09-07:
I would like to second this issue. In particular, I would like to highlight the scenario of unspecified widths of I/O elements.

Error message:
```
Exception in thread “main” firrtl.passes.PassExceptions:
firrtl.passes.CheckWidths$UninferredWidth:  @[:myAwesomeDesign.fir@22382.4] : [module MyAwesomeModule]  Uninferred width
```

That line points to the IO declaration:
```
output io : {... myUnspecifiedPort : UInt}}
```

**Request:** It would be nice if the "uninferredWidth" error message did the following:

* Note which signal has an unspecified/uninferrable width.
* Tell the user that the signal is inside of the IO bundle of a module.
* Tell the user that IO bundles *must* have specified widths.

jackkoenig, 2018-09-07:
I should note that IO Bundles do not necessarily have to have specified widths, but if you don't connect anything to a given field in the Bundle, it won't be able to infer the width and you get this error

edwardcwang, 2018-09-07:
Will do it now.

edwardcwang, 2018-09-07:
https://github.com/freechipsproject/chisel3/wiki/Width-Inference

jackkoenig, 2018-09-07:
@edwardcwang ProtoBuf is a binary format, there is no firrtl source locator when using protobuf. It *does* store the Chisel source locator, but unfortunately ports currently do not get source locators

seldridge, 2018-12-18:
Note that this is fixed for `CheckWidths` via #932. A similar approach that passes `Target` around would work for other `Check*` transforms.

seldridge, 2018-12-18:
This would be far easier to do if we had a visitor pattern that would communicate Target as it walked.

seldridge, 2018-12-18:
Closed via #932.

closed 2018-12-18 by seldridge