chipsalliance/firrtl #873: FIR crashes on Bool type resolution

tampler, 2018-08-22:
Hello 
A snippet below crashes when I try to instantiate a Counter module in my design. **Bool type resolution** being the victim. If I use a function to implement a counter in my design, all works seamlessly:
**System**: Ubuntu 16.04.5, Scala version: 2.11.12, Chisel version: 3.1.0
**Code**
package  counter_pkg 
import    chisel3._
import    chisel3.util._

class Counter (max:UInt, ena:Bool ) extends Module {
  val width = max.getWidth
    val io = IO (new Bundle {
    val out  = Output(UInt(width.W))
  })
    val x = RegInit(0.asUInt(width.W))
  x := Mux(x === max, 0.U, Mux (ena, x + 1.U, x))
  io.out := x
}

class CounterWrap extends Module {
  val io = IO (new Bundle {
    val out  = Output (UInt(16.W))
    })
   
  val idle :: run :: Nil = Enum (2)
  val state = RegInit(idle)
    val ena  = RegInit (false.B)
  ena := state === run
  val cnt  = Module (new Counter(16.U, **ena.toBool**)) // This won't compile if I remove .toBool
   io.out := 0.U
}

**Execution Result Log**

[info] [0.003] Elaborating design...
[info] [0.237] Done elaborating.
Total FIRRTL Compile Time: 327.7 ms
Total FIRRTL Compile Time: 35.0 ms
End of dependency graph
Circuit state created
Exception during evaluation: mux(WRef(cnt._T_12,UIntType(IntWidth(6)),NodeKind,MALE)) must be (0|1).U<1> was 18.PU<6>  @[Counter.scala 16:32:@12.4]  @[Counter.scala 16:32:@12.4]
Expression Evaluation stack
   0 cnt.x:Reg -> mux(cnt.reset, UInt<5>("h0"), cnt._T_15)
   1 cnt._T_15:Node -> mux(cnt._T_9, UInt<1>("h0"), cnt._T_14)
   2 cnt._T_14:Node -> mux(cnt._T_12, cnt._T_13, cnt.x)
Error: InterpreterExecption(mux(WRef(cnt._T_12,UIntType(IntWidth(6)),NodeKind,MALE)) must be (0|1).U<1> was 18.PU<6>  @[Counter.scala 16:32:@12.4] during warmup evaluation
Exception during evaluation: mux(WRef(cnt._T_12,UIntType(IntWidth(6)),NodeKind,MALE)) must be (0|1).U<1> was 18.PU<6>  @[Counter.scala 16:32:@12.4]  @[Counter.scala 16:32:@12.4]
Expression Evaluation stack
   0 cnt.x:Reg -> mux(cnt.reset, UInt<5>("h0"), cnt._T_15)
   1 cnt._T_15:Node -> mux(cnt._T_9, UInt<1>("h0"), cnt._T_14)
   2 cnt._T_14:Node -> mux(cnt._T_12, cnt._T_13, cnt.x)
[error] (run-main-2) firrtl_interpreter.InterpreterException: mux(WRef(cnt._T_12,UIntType(IntWidth(6)),NodeKind,MALE)) must be (0|1).U<1> was 18.PU<6>  @[Counter.scala 16:32:@12.4]
[error] firrtl_interpreter.InterpreterException: mux(WRef(cnt._T_12,UIntType(IntWidth(6)),NodeKind,MALE)) must be (0|1).U<1> was 18.PU<6>  @[Counter.scala 16:32:@12.4]
[error] 	at firrtl_interpreter.InterpreterException$.apply(InterpreterException.scala:10)
[error] 	at firrtl_interpreter.LoFirrtlExpressionEvaluator.evaluate(LoFirrtlExpressionEvaluator.scala:302)
[error] 	at firrtl_interpreter.LoFirrtlExpressionEvaluator$$anonfun$5.apply(LoFirrtlExpressionEvaluator.scala:453)

seldridge, 2018-08-22:
You're trying to pass a hardware type in as a parameter when you should be making a connection to a port. I think that the error message _without_ `toBool` may have provided better information. Unfortunately, this got out of Chisel and barfed in FIRRTL with an unintelligible error message... The real question/issue is why it got that far.

Can you add some additional IO to your `Counter` and make a connection to an enable signal?

Something like:
```scala
class Counter (max:UInt) extends Module {
  val width = max.getWidth
  val io = IO (new Bundle {
                 val out = Output(UInt(width.W))
                 val en = Input(Bool())          // Add a Chisel Bool, `en`
               })
  val x = RegInit(0.asUInt(width.W))
  x := Mux(x === max, 0.U, Mux (en, x + 1.U, x)) // Use the input `en`
  io.out := x
}

class CounterWrap extends Module {
  val io = IO (new Bundle {
                 val out = Output (UInt(16.W))
               })

  val idle :: run :: Nil = Enum (2)
  val state = RegInit(idle)
  val ena = RegInit (false.B)
  ena := state === run

  val cnt = Module(new Counter(16.U))
  cnt.io.en := ena                               // Make a connection to the `en` port
  io.out := 0.U
}
```

As a rough rule, the things that you pass as parameters to a `Module` are usually Scala types that you are using to _parameterize_ your module. Things like an `Int` for the width of some input port or a `String` to provide better `printf` information. This isn't at all true (you can pass a `Module` as a call-by-name parameter to be constructed inside another `Module`), but it's a more advanced pattern that you won't come across for normal HDL actions like making hardware connections.

tampler, 2018-08-22:
Excellent, that solves my issue

Thank you!

closed 2018-08-22 by tampler