chipsalliance/firrtl #886: Module-level DontTouchAnnotation

donggyukim, 2018-09-10:
**Type of issue**: feature request

**What is the use case for changing the behavior?**

**Impact**: API addition (no impact on existing code)

**Development Phase**: proposal

**Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

I don't think it'll be hard to implement, but it will be extremely useful when we write custom transforms.

jackkoenig, 2018-09-12:
I think this makes sense. The reason I haven't yet is I wasn't sure exactly what the semantics should be so I punted on it 😇

Should marking a Module as "don't touch" mean:
1. Treat all outputs as circuit sinks
1. Treat all ports as circuit sinks
1. Perform no optimizations* at all on the Module

*What counts as an optimization?
I think DCE, CSE, and ConstProp are obviously optimizations, but what about ReplaceTruncatingArithmetic (probably) and FlattenRegUpdate (probably not)?

donggyukim, 2018-09-13:
What's the current donttouch scope? I just assumed DCE, CSE, and ConstProp.

jackkoenig, 2018-09-14:
Currently, "don't touch" means treat a given signal as a sink of the circuit, thus logic and state elements that feed into the signal will also not be removed, but it provides no guarantee on anything driven by the "don't touched" signal.

If we were only to treat ports of a don't touched module as circuit sinks, that wouldn't prevent internal signals from being optimized. I'm thinking the most natural behavior would to just not optimize anything within the module.

This does leave a follow on question though, what happens if you have a module marked dont touch (so that no optimizations will be run), but it's also not actually used in the overall circuit. Should we still be able to delete the instantiation of the module? This relates to BlackBoxes as well where the current behavior is that the BlackBox instantiation is *not* deleted even though it is unused. For a BlackBox (and it's instantiation) to be removable by DCE, it has to be marked as optimizable.

donggyukim, 2018-09-15:
My expectation is:
1)  Do not optimize/delete any signals in the modules
2) Do not optimize/delete any instances and the port connections to the module, whether or nor it is used. I believe, in general, we mark a module "donttouch" as we expect it will be used in the circuit.

I specifically want module-level donttouch due to 2. Otherwise, it'll be very tedious to do 2.