chipsalliance/firrtl #893: FIR crashes when incorrect Verilog resource is provided for a BlackBox

tampler, 2018-09-19:
FIR violently crashes with the msg below when an invalid resource Verilog file is provided in a Black Box instantiation:
[error] (run-main-15) firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues

```scala
class VxuIO () extends Bundle {
  // stuff
}

class Vxu extends Module {
  val io  = IO(new VxuIO)
  val vxu = Module (new VxuImpl)
  vxu.io <> io  
}

class VxuImpl extends BlackBox with HasBlackBoxResource {
  val io  = IO(new VxuIO)
  setResource ("/VxuImpl0.v")
}

```
Of course, VxuImp.v is present in /src/main/resources and this code works with correct Verilog name

seldridge, 2018-10-24:
Yeah, this isn't an internal error, but a user error. For completeness, here's a longer stack trace:
```
[info] - should provide a useful error message *** FAILED ***
[info]   firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[info]   at firrtl.Utils$.error(Utils.scala:396)
[info]   at firrtl.Utils$.throwInternalError(Utils.scala:178)
[info]   at firrtl.Driver$.$anonfun$execute$1(Driver.scala:251)
[info]   at logger.Logger$.$anonfun$makeScope$2(Logger.scala:138)
[info]   at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[info]   at logger.Logger$.makeScope(Logger.scala:136)
[info]   at firrtl.Driver$.execute(Driver.scala:223)
[info]   at chisel3.Driver$.execute(Driver.scala:228)
[info]   at chisel3.Driver$.execute(Driver.scala:248)
[info]   at chiselTests.Sandbox.$anonfun$new$1(Sandbox.scala:25)
[info]   ...
[info]   Cause: java.io.FileNotFoundException: Resource '/VxuImpl0.v'
[info]   at firrtl.transforms.BlackBoxSourceHelper$.copyResourceToFile(BlackBoxSourceHelper.scala:129)
[info]   at firrtl.transforms.BlackBoxSourceHelper$.writeResourceToDirectory(BlackBoxSourceHelper.scala:117)
[info]   at firrtl.transforms.BlackBoxSourceHelper$$anonfun$1.applyOrElse(BlackBoxSourceHelper.scala:79)
[info]   at firrtl.transforms.BlackBoxSourceHelper$$anonfun$1.applyOrElse(BlackBoxSourceHelper.scala:77)
[info]   at scala.PartialFunction.$anonfun$runWith$1$adapted(PartialFunction.scala:141)
[info]   at scala.collection.Iterator.foreach(Iterator.scala:944)
[info]   at scala.collection.Iterator.foreach$(Iterator.scala:944)
[info]   at scala.collection.AbstractIterator.foreach(Iterator.scala:1432)
[info]   at scala.collection.IterableLike.foreach(IterableLike.scala:71)
[info]   at scala.collection.IterableLike.foreach$(IterableLike.scala:70)
[info]   ...
```

This is currently throwing a `FileNotFoundException`, which is reasonable, but because it's not a `FIRRTLException` FIRRTL thinks that it's off the rails. If we change this to a `FIRRTLException` with a better error message, I think that will fix this.

I'm proposing to change this to:
```
[info] - should provide a useful error message *** FAILED ***
[info]   firrtl.FIRRTLException: Verilog BlackBox '/VxuImpl0.v' not found. Did you misspell it? Is it in src/{main,test}/resources?
[info]   at firrtl.transforms.BlackBoxSourceHelper$.copyResourceToFile(BlackBoxSourceHelper.scala:129)
[info]   at firrtl.transforms.BlackBoxSourceHelper$.writeResourceToDirectory(BlackBoxSourceHelper.scala:117)
[info]   at firrtl.transforms.BlackBoxSourceHelper$$anonfun$1.applyOrElse(BlackBoxSourceHelper.scala:79)
[info]   at firrtl.transforms.BlackBoxSourceHelper$$anonfun$1.applyOrElse(BlackBoxSourceHelper.scala:77)
[info]   at scala.PartialFunction.$anonfun$runWith$1$adapted(PartialFunction.scala:141)
[info]   at scala.collection.Iterator.foreach(Iterator.scala:944)
[info]   at scala.collection.Iterator.foreach$(Iterator.scala:944)
[info]   at scala.collection.AbstractIterator.foreach(Iterator.scala:1432)
[info]   at scala.collection.IterableLike.foreach(IterableLike.scala:71)
[info]   at scala.collection.IterableLike.foreach$(IterableLike.scala:70)
[info]   ...

```

PR inbound...

jackkoenig, 2018-10-24:
@seldridge that looks good.

I did a small amount of work toward getting rid of stack traces on user error (that stack trace isn't helpful), but then lit it sit. I'll see if I can get something PRed soon.

PR #921 (2018-10-24): Better error message on missing BlackBox resource
Fixes #893 

If you try to access a non-existent BlackBox resource, this should generate a `FIRRTLException` child and not a raw `FileNotFoundException`. The latter will cause the "internal FIRRTL error, file an issue" message while the former does not. 

This fixes this for both `BlackBoxPathAnno` and `BlackBoxResourceAnno` by wrapping calls that interact with the (possibly non-existent) resource file in a method that will rethrow `FileNotFoundException` as a new `BlackBoxNotFoundException extends FIRRTLException`. 

Small changes:
  - The man [... cut]
closed 2018-10-24 by seldridge