chipsalliance/firrtl #899: HasBlackBoxResource doesn't work with absolute paths

mwachs5, 2018-09-26:
**Type of issue**: bug report | feature request 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Do something like:
```
class MyBlackBox extends BlackBox with HasBlackBoxResource {
 ...
 setResource("/absoluate/path/to/resource.v")

}
```
You'll get 
```
Caused by: java.io.FileNotFoundException: Resource '/absolute/path.to/resource.v'
```
even though the file exists.

  * **If possible, please directly provide the FIRRTL file or a relevant fragment if the file is large**
  * **What is the current behavior?**
java.io.FileNotFoundException
  * **What is the expected behavior?**

The file should get copied in as if it were a relative path.

  * **Please tell us about your environment:**

**What is the use case for changing the behavior?**

Makes HasBlackBoxResource more useful

**Impact**: API addition (no impact on existing code)

**Development Phase**: request

**Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

I'm working on a fix

jackkoenig, 2018-09-26:
Related, but support multiple files would be useful as well. Often Blackboxes involve multiple Verilog files

mwachs5, 2018-09-26:
Oh, is that not already supported? I thought I could call `setResource` as much as I wanted.

mwachs5, 2018-09-26:
Also, I am no longer working on a fix, because I realized the concept of `resource` is a Scala one that already assumes your files are in a particular location