chipsalliance/firrtl #907: firrtl build crashes

luudee, 2018-10-04:
Any ideas/suggestions ?

sbt -v compile
[process_args] java_version = '8'
# Executing command line:
java
-Xms1024m
-Xmx1024m
-XX:ReservedCodeCacheSize=128m
-XX:MaxMetaspaceSize=256m
-jar
/usr/share/sbt/bin/sbt-launch.jar
compile

[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /share/risc-v/rocket-chip/firrtl-1/project
[info] Loading settings from build.sbt ...
[info] Set current project to firrtl (in build file:/share/risc-v/rocket-chip/firrtl-1/)
[info] Executing in batch mode. For better performance use sbt's shell
[info] Compiling 92 Scala sources and 5 Java sources to /share/risc-v/rocket-chip/firrtl-1/target/scala-2.12/classes ...
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/Emitter.scala:544:9: match may not be exhaustive.
[warn] It would fail on the following input: (_, _)
[warn]         (dir, tpe) match {
[warn]         ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/passes/InferWidths.scala:226:60: match may not be exhaustive.
[warn] It would fail on the following inputs: (AnalogType(_), BundleType(_)), (AnalogType(_), ClockType), (AnalogType(_), FixedType(_, _)), (AnalogType(_), SIntType(_)), (AnalogType(_), Type()), (AnalogType(_), UIntType(_)), (AnalogType(_), VectorType(_, _)), (AnalogType(_), _), (BundleType(_), AnalogType(_)), (BundleType(_), ClockType), (BundleType(_), FixedType(_, _)), (BundleType(_), SIntType(_)), (BundleType(_), Type()), (BundleType(_), UIntType(_)), (BundleType(_), VectorType(_, _)), (BundleType(_), _), (ClockType, BundleType(_)), (ClockType, FixedType(_, _)), (ClockType, Type()), (ClockType, UIntType(_)), (ClockType, VectorType(_, _)), (ClockType, _), (FixedType(_, _), AnalogType(_)), (FixedType(_, _), BundleType(_)), (FixedType(_, _), ClockType), (FixedType(_, _), SIntType(_)), (FixedType(_, _), Type()), (FixedType(_, _), UIntType(_)), (FixedType(_, _), VectorType(_, _)), (FixedType(_, _), _), (SIntType(_), BundleType(_)), (SIntType(_), FixedType(_, _)), (SIntType(_), Type()), (SIntType(_), UIntType(_)), (SIntType(_), VectorType(_, _)), (SIntType(_), _), (Type(), AnalogType(_)), (Type(), BundleType(_)), (Type(), ClockType), (Type(), FixedType(_, _)), (Type(), SIntType(_)), (Type(), Type()), (Type(), UIntType(_)), (Type(), VectorType(_, _)), (Type(), _), (UIntType(_), BundleType(_)), (UIntType(_), FixedType(_, _)), (UIntType(_), Type()), (UIntType(_), VectorType(_, _)), (UIntType(_), _), (VectorType(_, _), BundleType(_)), (VectorType(_, _), ClockType), (VectorType(_, _), FixedType(_, _)), (VectorType(_, _), Type()), (VectorType(_, _), UIntType(_)), (VectorType(_, _), _), (_, AnalogType(_)), (_, BundleType(_)), (_, ClockType), (_, FixedType(_, _)), (_, SIntType(_)), (_, Type()), (_, UIntType(_)), (_, VectorType(_, _)), (_, _)
[warn]     def get_constraints_t(t1: Type, t2: Type): Seq[WGeq] = (t1,t2) match {
[warn]                                                            ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/passes/Resolves.scala:87:42: match may not be exhaustive.
[warn] It would fail on the following inputs: (MInfer, _), (MPortDir(), _), (MRead, _), (MReadWrite, _), (MWrite, _), (_, MInfer), (_, MPortDir()), (_, MRead), (_, MReadWrite), (_, MWrite), (_, _)
[warn]         case Some(p) => mports(e.name) = (p, dir) match {
[warn]                                          ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/proto/FromProto.scala:142:76: match may not be exhaustive.
[warn] It would fail on the following inputs: MEMORY_PORT_DIRECTION_UNKNOWN, UNRECOGNIZED
[warn]   def convert(mportdir: Firrtl.Statement.MemoryPort.Direction): MPortDir = mportdir match {
[warn]                                                                            ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/proto/FromProto.scala:251:5: match may not be exhaustive.
[warn] It would fail on the following inputs: PORT_DIRECTION_UNKNOWN, UNRECOGNIZED
[warn]     dir match {
[warn]     ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/transforms/ConstantPropagation.scala:67:42: match may not be exhaustive.
[warn] It would fail on the following inputs: (Literal(), _), (SIntLiteral(_, _), UIntLiteral(_, _)), (SIntLiteral(_, _), _), (UIntLiteral(_, _), SIntLiteral(_, _)), (UIntLiteral(_, _), _), (_, Literal()), (_, SIntLiteral(_, _)), (_, UIntLiteral(_, _)), (_, _)
[warn]     def fold(c1: Literal, c2: Literal) = (c1, c2) match {
[warn]                                          ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/transforms/DeadCodeElimination.scala:156:27: match may not be exhaustive.
[warn] It would fail on the following input: Port(_, _, (x: firrtl.ir.Direction forSome x not in (Input, Output)), (x: firrtl.ir.Type forSome x not in firrtl.ir.AnalogType))
[warn]         ext.ports.foreach {
[warn]                           ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/Driver.scala:93:40: method getAnnotationFileName in class FirrtlExecutionOptions is deprecated (since 1.1): Use FirrtlOptions.annotationFileNames instead
[warn]     val oldAnnoFileName = firrtlConfig.getAnnotationFileName(optionsManager)
[warn]                                        ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/graph/DiGraph.scala:194:29: class Stack in package mutable is deprecated (since 2.12.0): Stack is an inelegant and potentially poorly-performing wrapper around List. Use a List assigned to a var instead.
[warn]     val stack = new mutable.Stack[T]
[warn]                             ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/graph/DiGraph.scala:210:33: class Stack in package mutable is deprecated (since 2.12.0): Stack is an inelegant and potentially poorly-performing wrapper around List. Use a List assigned to a var instead.
[warn]     val callStack = new mutable.Stack[StrongConnectFrame[T]]
[warn]                                 ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/passes/clocklist/ClockList.scala:25:23: method getChildrenMap in object WiringUtils is deprecated (since 1.1.1): Use DiGraph/InstanceGraph
[warn]     val childrenMap = getChildrenMap(c)
[warn]                       ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/passes/clocklist/ClockList.scala:27:20: method getLineage in object WiringUtils is deprecated (since 1.1.1): Use DiGraph/InstanceGraph
[warn]     val lineages = getLineage(childrenMap, top)
[warn]                    ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/passes/clocklist/ClockListUtils.scala:21:61: class Lineage in package wiring is deprecated (since 1.1.1): Use DiGraph/InstanceGraph
[warn]   def getSourceList(moduleMap: Map[String, DefModule])(lin: Lineage): Seq[String] = {
[warn]                                                             ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/passes/clocklist/ClockListUtils.scala:39:90: class Lineage in package wiring is deprecated (since 1.1.1): Use DiGraph/InstanceGraph
[warn]   def getOrigins(connects: Connects, me: String, moduleMap: Map[String, DefModule])(lin: Lineage): Map[String, String] = {
[warn]                                                                                          ^
[warn] /share/risc-v/rocket-chip/firrtl-1/src/main/scala/firrtl/passes/memlib/ReplaceMemMacros.scala:217:68: method getMyAnnotations in class Transform is deprecated (since 1.1): Just collect the actual Annotation types the transform wants
[... 162 lines cut ...]
[error]         at sbt.internal.inc.IncrementalCompile$.apply(Compile.scala:61)
[error]         at sbt.internal.inc.IncrementalCompilerImpl.compileInternal(IncrementalCompilerImpl.scala:309)
[error]         at sbt.internal.inc.IncrementalCompilerImpl.$anonfun$compileIncrementally$1(IncrementalCompilerImpl.scala:267)
[error]         at sbt.internal.inc.IncrementalCompilerImpl.handleCompilationError(IncrementalCompilerImpl.scala:158)
[error]         at sbt.internal.inc.IncrementalCompilerImpl.compileIncrementally(IncrementalCompilerImpl.scala:237)
[error]         at sbt.internal.inc.IncrementalCompilerImpl.compile(IncrementalCompilerImpl.scala:68)
[error]         at sbt.Defaults$.compileIncrementalTaskImpl(Defaults.scala:1429)
[error]         at sbt.Defaults$.$anonfun$compileIncrementalTask$1(Defaults.scala:1403)
[error]         at scala.Function1.$anonfun$compose$1(Function1.scala:44)
[... cut ...]
[error]         at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error]         at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error]         at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error]         at java.lang.Thread.run(Thread.java:748)
[error] (Compile / compileIncremental) java.io.IOException: Cannot run program "javac" (in directory "/share/risc-v/rocket-chip/firrtl-1"): error=2, No such file or directory
[error] Total time: 37 s, completed Oct 4, 2018 11:40:36 PM

edwardcwang, 2018-10-04:
Is the JDK installed and functional by any chance?

```
 [error] Cannot run program "javac" (in directory "/share/risc-v/rocket-chip/firrtl-1"): error=2, No such file or directory
```

luudee, 2018-10-04:
Ahh, the JDK was installed, but the JDK-development was not.

Thank you for pointing that out Edward !

closed 2018-10-04 by luudee