chipsalliance/firrtl #908: Wrong JDK ???

luudee, 2018-10-04:
What version JDK  isrequired for the build ?

[warn] use -help for a list of possible options.
[error] javac: invalid source release: 1.8
[warn] javac exited with exit code 2
[error] (Compile / compileIncremental) javac returned non-zero exit code
[error] Total time: 55 s, completed Oct 5, 2018 12:02:36 AM

Thanks !

jackkoenig, 2018-10-04:
Did you figure this out and that’s why you closed the issue?

In any case Java 8 is the correct JDK.

jackkoenig, 2018-10-04:
Oh I see, I’m guessing the issues was resolved in #907

luudee, 2018-10-04:
Yeah, I had the correct JDK, but installed an older jdk-development kit by accient ...

Sorry about this ;)

jackkoenig, 2018-10-04:
No worries! Glad you were able to solve the issue!