chipsalliance/firrtl #910: (possible?) Windows compile issue

ducky64, 2018-10-05:
I get this error while trying to compile firrtl in sbt under Windows on two different machines. However, firrtl compiles fine under Linux, even using WSL on the same machine. I tried blowing away the entire firrtl repository as well as the entire .ivy2 cache and versions of firrtl in the .ivy2 local.

Anyone else have issues building firrtl under Windows?
```
[error] C:\GitHub\firrtl\target\scala-2.12\src_managed\main\firrtl\antlr\FIRRTLBaseVisitor.java:13:14: FIRRTLBaseVisitor is already defined as object FIRRTLBaseVisitor
[error] public class FIRRTLBaseVisitor<T> extends AbstractParseTreeVisitor<T> implements FIRRTLVisitor<T> {
[error]              ^
[error] C:\GitHub\firrtl\target\scala-2.12\src_managed\main\firrtl\antlr\FIRRTLLexer.java:16:14: FIRRTLLexer is already defined as object FIRRTLLexer
[error] public class FIRRTLLexer extends Lexer {
[error]              ^
[error] C:\GitHub\firrtl\target\scala-2.12\src_managed\main\firrtl\antlr\FIRRTLParser.java:13:14: FIRRTLParser is already defined as object FIRRTLParser
[error] public class FIRRTLParser extends Parser {
[error]              ^
[error] C:\GitHub\firrtl\target\scala-2.12\src_managed\main\firrtl\antlr\FIRRTLVisitor.java:12:18: FIRRTLVisitor is already defined as object FIRRTLVisitor
[error] public interface FIRRTLVisitor<T> extends ParseTreeVisitor<T> {
[error]                  ^
[error] four errors found
[error] C:\GitHub\firrtl\target\scala-2.12\src_managed\main\firrtl\antlr\FIRRTLBaseVisitor.java:13:14: FIRRTLBaseVisitor is already defined as object FIRRTLBaseVisitor
[error] public class FIRRTLBaseVisitor<T> extends AbstractParseTreeVisitor<T> implements FIRRTLVisitor<T> {
[error]              ^
[error] C:\GitHub\firrtl\target\scala-2.12\src_managed\main\firrtl\antlr\FIRRTLLexer.java:16:14: FIRRTLLexer is already defined as object FIRRTLLexer
[error] public class FIRRTLLexer extends Lexer {
[error]              ^
[error] C:\GitHub\firrtl\target\scala-2.12\src_managed\main\firrtl\antlr\FIRRTLParser.java:13:14: FIRRTLParser is already defined as object FIRRTLParser
[error] public class FIRRTLParser extends Parser {
[error]              ^
[error] C:\GitHub\firrtl\target\scala-2.12\src_managed\main\firrtl\antlr\FIRRTLVisitor.java:12:18: FIRRTLVisitor is already defined as object FIRRTLVisitor
[error] public interface FIRRTLVisitor<T> extends ParseTreeVisitor<T> {
[error]                  ^
[info] No documentation generated with unsuccessful compiler run
[warn] two warnings found
[error] four errors found
[error] (Compile / compileIncremental) Compilation failed
[error] (Scalaunidoc / doc) Scaladoc generation failed
[error] Total time: 30 s, completed Oct 5, 2018 12:43:30 PM
```

**Type of issue**: bug report

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **If possible, please directly provide the FIRRTL file or a relevant fragment if the file is large**
  * **What is the current behavior?** It don't build
  * **What is the expected behavior?** It should build
  * **Please tell us about your environment:**
    (examples)
    - version: master
    - OS: Windows 10

**What is the use case for changing the behavior?**

**Impact**: unknown

**Development Phase**: something is broken and I don't know what

**Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)
See above for stack trace.

jackkoenig, 2018-10-05:
Are you using an IDE? We've previously had similar issues on MacOS using IntelliJ

ducky64, 2018-10-05:
No, this is with just sbt.

borancar, 2018-10-08:
I can reproduce it on my machine. All it takes is a fresh clone followed by a `sbt assembly`.

fire3, 2018-10-31:
I have the same problem on my Windows 10 machine with fresh installed Oracle JDK8 and SBT tools.

qiner199, 2018-11-09:
I have the same problem ,anyone know how to solve it

ducky64, 2018-11-09:
I haven't found a good solution, but here are some workarounds:
- Use [WSL](https://docs.microsoft.com/en-us/windows/wsl/install-win10), which compiles FIRRTL fine. It's still a Linux system in Windows, but with better integration compared to a virtual machine.
- Use a published jarfile from Maven, by using managed dependencies. Note that we've recently started publishing snapshots, so this is viable if you're developing something that depends on FIRRTL (for example, if you only need FIRRTL as a dependency to run the latest Chisel), but not if you're developing FIRRTL.
- You can also take a jarfile compiled in a non-Windows environment (like WSL) and drop it in the .ivy2 directory where publishLocal normally drops them.

Trudddddy, 2018-11-14:
I have the same problem on win7 and fedora29.  and get a temporary solution from a kind-hearted guy, and I also share it with you.
When the error comes up compiling firrtl, remove other files except FirrtlProtos.java in firrtl\target\streams\protobuf\protobufGenerate\$global\streams\protobuf_2.12\out-cache, then re-compile, it will pass and you will notice the java files to be compiled are reduced from 9 to 5.
Hope anyone can find the root cause for this problem.

InterestedTom, 2018-11-21:
> I have the same problem on win7 and fedora29. and get a temporary solution from a kind-hearted guy, and I also share it with you.
> When the error comes up compiling firrtl, remove other files except [...]

It may compile but it doesn't pass the tests.

banditos13, 2018-12-24:
sbt -mem 2048 ++2.11.12 publishLocal

zengjianjiang, 2020-01-31:
The problem can be solved by using AdoptOpenJDK prebuild windows binaries. I got the same problem when I use Oracle JAVA 1.8.0_231 in my win10 x64 machine. After installing AdoptOpenJDK 1.8.0_242 instead, the build was successful.