chipsalliance/firrtl #916: Add command line options to Verilator

stevenmburns, 2018-10-16:
**Type of issue**: feature request

**What is the use case for changing the behavior?**
Need to enable system verilog code in Verilator

**Impact**: API addition (no impact on existing code)

**Development Phase**: request

We have a use case where we want the command line option "-sv" added to the verilator invocation (from @desmonddak).
There is a general facility for doing this with VCS. We would like that replicated for verilator.

Here is the patch we have been using to hard code this option:

````
diff --git a/src/main/scala/firrtl/util/BackendCompilationUtilities.scala b/src/main/scala/firrtl/util/BackendCompilationUtilities.scala
index bd5d2a9a..6b3eee61 100644
--- a/src/main/scala/firrtl/util/BackendCompilationUtilities.scala
+++ b/src/main/scala/firrtl/util/BackendCompilationUtilities.scala
@@ -139,6 +139,7 @@ trait BackendCompilationUtilities {
       { if(suppressVcd) { Seq.empty } else { Seq("--trace")} } ++
       Seq(
         "-O1",
+        "-sv",
         "--top-module", topModule,
         "+define+TOP_TYPE=V" + dutFile,
         s"+define+PRINTF_COND=!$topModule.reset",
````

ucbjrl, 2018-10-22:
I think the currently _cheapest_ way to do this (in terms of code changes), is to duplicate the existing vcs code in chisel-testers. To that end, I've created freechipsproject/chisel-testers#214 for discussion. If we agree that is a workable solution, I'll close this issue.
Otherwise, we should probably consider migrating the relevant chisel-testers code here.

desmonddak, 2018-10-22:
Seems reasonable to get the same capability for Verilator as VCS.  A wrinkle could be command ordering, so we can test this (our issue was adding flags to support SystemVerilog).

seldridge, 2018-10-24:
https://github.com/freechipsproject/chisel-testers/pull/214 makes sense, generally. The user should be able to add options. However, I think there's motivation for `verilogToCpp` or `verilogToVerilator` to choose specific options based on the value of `firrtlOptions.compilerName`. I.e., if the user has requested `-X sverilog` it would be reasonable to include `-sv` when calling Verilator automatically.

Note: once the options/annotations stuff lands, the clean way to do this is to examine the contents of the `EmitterAnnotation`.

ucbjrl, 2018-10-29:
I'm going to close this issue and open a new chisel-testers issue based on @seldridge's comment.

closed 2018-10-29 by ucbjrl