chipsalliance/firrtl #917: Don't include verilog header files in "FileList" for VCS/Verilator

stevenmburns, 2018-10-16:
**Type of issue**: feature request

**What is the use case for changing the behavior?** Can't compile some existing verilog code which use header files.

**Impact**: API addition (no impact on existing code)

**Development Phase**: request

Also from @desmonddak.
We need a way to distinguish between verilog header files and verilog source files.
Currently both are being added to the "FileList". Our solution below is to exclude files from the "FileList" based on name, but there might be a more general solution, or a different workaround.

````
diff --git a/src/main/scala/firrtl/transforms/BlackBoxSourceHelper.scala b/src/main/scala/firrtl/transforms/BlackBoxSourceHelper.scala
index bf419840..f48e8e22 100644
--- a/src/main/scala/firrtl/transforms/BlackBoxSourceHelper.scala
+++ b/src/main/scala/firrtl/transforms/BlackBoxSourceHelper.scala
@@ -97,8 +97,9 @@ class BlackBoxSourceHelper extends firrtl.Transform {
       BlackBoxSourceHelper.writeTextToFile(text, file)
       file
     }
+    val trueResourceFiles = resourceFiles.filterNot( _.getName().endsWith(".vh"))

-    BlackBoxSourceHelper.writeFileList(resourceFiles ++ inlineFiles, targetDir)
+    BlackBoxSourceHelper.writeFileList(trueResourceFiles ++ inlineFiles, targetDir)

     state
   }
````

stevenmburns, 2018-10-19:
@chick You might be the right one to address this and also #916

desmonddak, 2018-10-22:
Yes.  The main issue is not compiling headers and other files.

PR #918 (2018-10-31): Don't include verilog header files in "FileList" for VCS/Verilator.
When constructing the black box helper file list (firrtl_black_box_resource_files.f), filter out Verilog header files (*.vh) - Fixes #917
commit 297fbda1 (2018-10-31): Don't include verilog header files in "FileList" for VCS/Verilator. (#918)

When constructing the black box helper file list (firrtl_black_box_resource_files.f), filter out Verilog header files (*.vh) - Fixes #917
commit 70772951 (2018-12-13): Don't include verilog header files in "FileList" for VCS/Verilator. (#918)

When constructing the black box helper file list (firrtl_black_box_resource_files.f), filter out Verilog header files (*.vh) - Fixes #917

(cherry picked from commit 297fbda180584cc3456145faecdc40418babeef1)
closed 2018-10-31 by ucbjrl