chipsalliance/firrtl #922: Instance Annotations seem to have broken Chisel StrongEnum and LoadMemory

seldridge, 2018-10-25:
**Type of issue**: bug report

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

I'm seeing strong enum and load memory failures in FIRRTL with FIRRTL and Chisel master.

To reproduce: run Chisel's tests with FIRRTL 7e2f787 published locally.

Apparently Chisel tests may have not run on this?

  * **If possible, please directly provide the FIRRTL file or a relevant fragment if the file is large**

I'm not sure where the problem is, but this seems to be Target related from Instance Annotations.

  * **What is the current behavior?**

StrongEnumSpec and LoadMemoryFromFileSpec are barfing.

  * **What is the expected behavior?**

These should pass...

  * **Please tell us about your environment:**

FIRRTL: 7e2f787
Chisel: 6004052

**What is the use case for changing the behavior?**

It's broken.

**Impact**: unknown

**Development Phase**: request

(I don't know what's going on here, so it's a request...)

**Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

StrongEnum failure:
```
[info] StrongEnum FSM
[info] - should work *** FAILED ***
[info]   firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[info]   at firrtl.Utils$.error(Utils.scala:396)
[info]   at firrtl.Utils$.throwInternalError(Utils.scala:178)
[info]   at firrtl.Driver$.$anonfun$execute$1(Driver.scala:251)
[info]   at logger.Logger$.$anonfun$makeScope$2(Logger.scala:138)
[info]   at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[info]   at logger.Logger$.makeScope(Logger.scala:136)
[info]   at firrtl.Driver$.execute(Driver.scala:223)
[info]   at chisel3.testers.TesterDriver$.execute(TesterDriver.scala:54)
[info]   at chiselTests.ChiselRunners.runTester(ChiselSpec.scala:24)
[info]   at chiselTests.ChiselRunners.runTester$(ChiselSpec.scala:23)
[info]   ...
[info]   Cause: java.lang.NumberFormatException: For input string: "value"
[info]   at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
[info]   at java.lang.Integer.parseInt(Integer.java:580)
[info]   at java.lang.Integer.parseInt(Integer.java:615)
[info]   at scala.collection.immutable.StringLike.toInt(StringLike.scala:301)
[info]   at scala.collection.immutable.StringLike.toInt$(StringLike.scala:301)
[info]   at scala.collection.immutable.StringOps.toInt(StringOps.scala:29)
[info]   at firrtl.annotations.Target$.$anonfun$toTargetTokens$1(Target.scala:108)
[info]   at scala.collection.immutable.List.foreach(List.scala:389)
[info]   at firrtl.annotations.Target$.toTargetTokens(Target.scala:106)
[info]   at firrtl.annotations.ComponentName.toTarget(Target.scala:647)
[info]   ...
```
LoadMemory failure:
```
[info] LoadMemoryFromFileSpec:
[info] - Users can specify a source file to load memory from *** FAILED ***
[info]   firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[info]   at firrtl.Utils$.error(Utils.scala:396)
[info]   at firrtl.Utils$.throwInternalError(Utils.scala:178)
[info]   at firrtl.Driver$.$anonfun$execute$1(Driver.scala:251)
[info]   at logger.Logger$.$anonfun$makeScope$2(Logger.scala:138)
[info]   at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[info]   at logger.Logger$.makeScope(Logger.scala:136)
[info]   at firrtl.Driver$.execute(Driver.scala:223)
[info]   at chisel3.Driver$.execute(Driver.scala:228)
[info]   at chisel3.Driver$.execute(Driver.scala:248)
[info]   at chiselTests.LoadMemoryFromFileSpec.$anonfun$new$1(LoadMemoryFromFileSpec.scala:130)
[info]   ...
[info]   Cause: java.lang.Exception: Multiple (2 found for memory UsesMem.UsesMemLow.memory one LoadMemoryAnnotation is allowed per memory
[info]   at chisel3.util.experimental.LoadMemoryTransform.$anonfun$run$2(LoadMemoryTransform.scala:144)
[info]   at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:234)
[info]   at scala.collection.immutable.Map$Map2.foreach(Map.scala:156)
[info]   at scala.collection.TraversableLike.map(TraversableLike.scala:234)
[info]   at scala.collection.TraversableLike.map$(TraversableLike.scala:227)
[info]   at scala.collection.AbstractTraversable.map(Traversable.scala:104)
[info]   at chisel3.util.experimental.LoadMemoryTransform.run(LoadMemoryTransform.scala:141)
[info]   at chisel3.util.experimental.LoadMemoryTransform.execute(LoadMemoryTransform.scala:227)
[info]   at firrtl.Transform.$anonfun$runTransform$2(Compiler.scala:199)
[info]   at firrtl.Utils$.time(Utils.scala:183)
[info]   ...
```

seldridge, 2018-10-25:
All good... Just glad it's debuggable and not Jenkins off the rails... I have not been looking at this... another fire :fire::fire::fire:. I may be able to loop back later tonight on this... You and/or @azidar may have the request knowledge cached from fixing the Inline Annotation bug.

Yeah, what the hell is `$TRAVIS_COMMIT_RANGE` doing?

jackkoenig, 2018-10-25:
I thought it was getting just the commits in the PR but I think I'm wrong about that

seldridge, 2018-10-25:
Yeah, it's parsing the whole history: https://travis-ci.org/freechipsproject/firrtl/jobs/445954084#L1538

seldridge, 2018-10-29:
#914 sorts this out for `LoadMemoryTransform`. Looking at `StrongEnum`, but haven't figured it out yet. The strong enum FSM test is producing the following illegal annotation that references the `value` index:
```json
  {
    "class":"chisel3.core.EnumAnnotations$EnumComponentAnnotation",
    "target":"StrongEnumFSMTester.StrongEnumFSMTester.expected_state[value]",
    "enumTypeName":"chiselTests.StrongEnumFSM$State"
  }
```

hngenc, 2018-10-29:
The issue with `StrongEnum` is that instance names aren't supposed to refer to indexed elements in `Vec`s. It turns out this wasn't working perfectly with the old annotations either, but no exceptions were being thrown. I've been thinking about this issue all weekend, but it seems tricky to fix.

Just for reference, this is the code generating the error:
```scala
val expected_state = VecInit(sNone, sNone, sOne1, sNone, sOne1, sTwo1s, sTwo1s, sNone, sOne1, sTwo1s)
// ...
assert(dut.io.state === expected_state(cycle))
```
In this example, `cycle` is equal to `Counter(...).value`, so its instance name is `value`. Therefore, `expected_state(cycle)`'s instance name is `expected_state(value)`. However, since `expected_state(value)` can't be determined statically at compile-time, it can't be annotated.

hngenc, 2018-10-29:
@seldridge Sorry, I'm not sure I understand. Do you mean we should only annotate a wire/reg/etc. if it is the first hardware to connect to an enum? So in this example:
```scala
val x = Reg(eType())
val y = Wire(eType())

x := e
y := e
```
Should only `x` be annotated in the above example?

seldridge, 2018-10-29:
No, sorry... I wasn't very clear there. You want to annotate all of the declarations (which is what I think you're attempting to do now?). You'd want one annotation for `x` and one annotation for `y` indicating that they are enumerated types. You don't want to annotate anything else, though.

The current problem is that `bind` is called by both of the following statements and generates an annotation for both:
```scala
val x = VecInit(eType()) // first bind, generate an EnumComponentAnnotation for x[0]
val y = x(0) // this will call bind, but shouldn't generate an EnumComponentAnnotation for x[0]
```

The first statement is all good, but the latter results in a generated annotation pointing at a target that may be incorrect FIRRTL syntax (e.g., `x[value]`) and does not exist. Elaborating on the underlying problem here... All annotations should have a valid target (subclass `SingleTargetAnnotation`) or no target (subclass `NoTargetAnnotation`). You can cram whatever information you want into the annotation, but the `target` needs to make sense.

What you really want, which I think you're trying to do, is to annotate the declaration. Locking down what is the declaration seemed hard when I was trying to do it. Kludgily overriding `connect` seems to work, but that smells very bad... There should be a better way to do this.

Note: As this seems hard to do... I'd be fine with removing the annotation capability of `StrongEnum` to un-block instance annotations and give you more time to work on a nice, clean solution.

hngenc, 2018-10-29:
Yeah, I think turning off annotations will be the best short-term solution.

For the long-term, there are four possible solutions that I can think of:
1) Annotating `Vec`s of enums, rather than annotating each element of a `Vec` one-by-one. However, the annotations can get long when you have `Vec`s of `Vec`s, or `Vec`s of bundles which may have enums as their elements.
2) Turning enums into a first-class FIRRTL construct. There won't be any annotation hacking necessary then.
3) Overriding `connect`, as you suggested. This seems to work even for two-dimensional vectors. I'll look into how well it works with bundles.
4) Overriding `setRef` may also be an option. I still need to see how well that option works with multi-dimensional vectors and bundles.

seldridge, 2018-10-29:
Fixed via:

- [freechipsproject/chisel#914](https://github.com/freechipsproject/chisel3/pull/914)
- [freechipsproject/chisel#916](https://github.com/freechipsproject/chisel3/pull/916)

This should clear the way for #926.

closed 2018-10-29 by seldridge