chipsalliance/firrtl #923: Chisel tests not running

seldridge, 2018-10-26:
**Type of issue**: bug report

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Look at the Travis output for a PR. See that Chisel tests don't run because it's incorrectly parsing the full commit log. See: https://travis-ci.org/freechipsproject/firrtl/jobs/445954084#L1538

@jackkoenig and I know about this, but I'm adding an issue to track it.

 **What is the use case for changing the behavior?**

**Impact**: no functional change

**Development Phase**: request

**Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

seldridge, 2018-10-30:
Alright, so there's a further problem here. Travis is doing a shallow clone. This consequently breaks any type of `$MY_BRANCH..master` or `$MY_BRANCH...master` type `git log` parsing for a PR with a sufficiently large number of commits. The only way to guarantee that we get around this is to do a full clone (I think).

Alternatively, we can do a `git log --format=%B --no-merges -n1`. This _should_ examine the most recent non-merge commit to see if Chisel tests have to be run. I'm testing this now...

Does this make sense?

seldridge, 2018-10-30:
More concretely, the problem is in checking out the PR. Once you do that, you wind up with a mostly full history. You have a full history going from your HEAD all the way to the initial commit, but only a partial history for the PR base. You can see the explosion of commits below:

```bash
$ git clone --depth=10 https://github.com/freechipsproject/firrtl.git
$ cd firrtl
$ git fetch origin +refs/pull/926/merge:
$ git log --pretty=oneline | wc -l

16

$ git checkout -qf FETCH_HEAD
$ git log --pretty=oneline | wc -l

1414
```

jackkoenig, 2018-10-30:
The `git log` command to search for `[skip chisel tests]` is *supposed* to only search the commits in the PR, ie. `TRAVIS_COMMIT_RANGE` should be just the range of commits in the PR. Clearly something is wrong with that but ideally the command should work no matter how much history is included in the repo.

seldridge, 2018-10-30:
The thing is, `TRAVIS_COMMIT_RANGE` isn't `$FIRST_COMMIT..$LAST_COMMIT_IN_THIS_PR`, it's `$BRANCH_IM_MERGING_INTO..$LAST_COMMIT_IN_THIS_PR`. So, this is really asking, "What can I reach from the last PR commit in this series that I can't reach from master?

In this way, a shallow clone does matter, because I wind up with the weird situation where the ancestry of master never converges back to the ancestry of the PR. Hence, commit ranges just don't work and you wind up with the full history. You can see the broken ancestry in the image above. The highlighted commit is the last one that we have from master.

If you have a full ancestry, then this will be only the commits in the PR, but if you have partial ancestry on master, then you wind up with every commit.

Does that make sense? (It took me a while to convince myself of this...)

jackkoenig, 2018-10-31:
Yeah, I think I understand now. I don't understand why `TRAVIS_COMMIT_RANGE` is `$BRANCH_IM_MERGING_INTO..$LAST_COMMIT_IN_THIS_PR` which seems unhelpful but they probably know more about git than I do :)

seldridge, 2018-10-31:
We could increase the clone depth or always use a full clone. However, the former will eventually fail for some big PR and the latter is inefficient. What I've settled on in #927 is to examine `$TRAVIS_COMMIT_RANGE` and see if the oldest commit contained in the first commit in the repo. If this is the case, then the repo is unshallowed before proceeding.

You can see this getting hit in Travis as #927 is based off of the protobuf PR commit: https://travis-ci.org/freechipsproject/firrtl/jobs/448632803#L471

PR #927 (2018-10-31): Fix $TRAVIS_COMMIT_RANGE
Fixes #923 

This adds a `before_script` block to the Travis configuration that fixes problems with use of `$TRAVIS_COMMIT_RANGE`. This looks at the oldest commit contained in this range. If this is the oldest commit in the repository then this indicates that a shallow clone was not able to fully capture where the base branch and the PR merge. This will then unshallow the repo before running any scripts.

Requested merge strategy: squash.

And note that this is cherry-picked onto the protobuf commit to exercise this code path in Travis.

Old, cherry-picked build: https [... cut]
closed 2018-10-31 by seldridge