chipsalliance/firrtl #933: Make CheckWidths aware of WiringTransform annotations

albertchen-sifive, 2018-11-06:
**Type of issue**: feature request

**What is the use case for changing the behavior?**
I would like to make a Chisel API that connects wires across modules using `WiringTransform`. I'm doing this by declaring a wire of the same type using `chiselTypeOf` and creating sink and source wiring annotations. However, this leads to `UninferredWidth` errors, because sink widths are not set until after `WiringTransform` which runs after `CheckWidths`.

**Impact**: unknown

**Development Phase**: request

jackkoenig, 2018-11-06:
Possible approaches:
1. Make WiringTransform run before InferWidths
1. Make InferWidths aware of Wiring annotations (could also be spun out into a special case pre-width inference transform)
1. Add WidthInference ignore annotation (for specific signals) and then rerun InferWidths after WiringTransform

I think (3) has high potential for brittleness/unintended consequences, (1) or (2) seem more reasonable to me

seldridge, 2018-11-06:
The real problem here is not `CheckWidths`, but `InferWidths` doesn't know anything about `WiringTransform` annotations. The `UninferredWidth` error is a pretty dumb check -- this just looks to see if everything has a width. If it doesn't, then it asserts.

I agree with your summary, @jackkoenig. (1) I don't think is possible, because `CheckWidths` runs as soon as you read in Chirrtl. Doing a `WiringTransform` on High FIRRTL is totally doable, but it would have to run almost before anything else (which may actually make sense. (2) is likely tractable because `WiringTransform` soft connects are just ports. When building constraints these could just be treated as ports and everything should work as it does now. (3) would be a fallback if (1) or (2) proved intractable for whatever reason. The problem with (3) is that it means that use of `WiringTransform` annotations would result in breaking guarantees on High/Mid/Low FIRRTL. Custom Transforms that are relying on this would barf.

closed 2019-03-01 by azidar