chipsalliance/firrtl #934: Runtime increases

juliusbaxter, 2018-11-07:
**Type of issue**: bug report

I've got a design based on the SiFive Freedom E310. We're doing some customization of the platform and at some point we started noticing our FIRRTL runtimes were becoming much longer, and are now the longest pole in our simulation compile iterations. Chisel/sbt runs are still at about 40s but we're seeing FIRRTL runtime spike from 40s to 6-8minutes.

I've done some digging and it appears it's related to a bus master module we've written to expose TileLink master ports at the top of the resulting Verilog netlist. (Obviously for use with other things outside the chip which are mastering the SBus.) I initially thought it might be related to the updates we took to FIRRTL and rocket-chip, but varying the revisions of rocket-chip and FIRRTL appear to make no difference.

This is the Chisel for the module:

```case class MyBusMasterParams(name : String = "bus_master" )
case object PeripheryMyBusMasterKey extends Field[Seq[MyBusMasterParams]]
trait HasMyBusMasterBundleContents extends Bundle {
  def params: MyBusMasterParams
}

trait HasPeripheryMyBusMaster { this: BaseSubsystem =>
  val busmasters = p(PeripheryMyBusMasterKey).map { params =>
    val busmaster = LazyModule(new MyBusMaster(params))
    // This was taken from rocket-chip/.../subsystem/Ports.scala
    sbus.fromMaster(Some(params.name), buffer = BufferParams.default) {
      TLSourceShrinker(1 << 14) := TLWidthWidget(busmaster.beatBytes)
    } := busmaster.node
    busmaster
  }
}

class MyBusMaster(params: MyBusMasterParams)(implicit p: Parameters)  extends LazyModule
{
  val beatBytes = 4
  val node = TLClientNode(Seq(TLClientPortParameters(Seq(TLClientParameters(name = params.name)))))
  lazy val module = new LazyModuleImp(this) {
    val (tl_out, edge) = node.out(0)
    val sinksourceBits = 20 //log2Ceil(edge.manager.endSinkId)
    val addressBits = log2Ceil(edge.manager.maxAddress)
    // Expose the address and data tilelink channels to the outside world:
    val busparams = TLBundleParameters(addressBits=addressBits,dataBits=32,sourceBits=sinksourceBits,sinkBits=sinksourceBits,sizeBits=2)
    val busmaster_a_io = IO(Decoupled(new TLBundleA(busparams)).flip)
    val busmaster_d_io = IO(Decoupled(new TLBundleD(busparams)))
    tl_out.a <> busmaster_a_io
    busmaster_d_io <> tl_out.d
    // // Tie off unused channels
    tl_out.b.valid := Bool(false)
    tl_out.c.ready := Bool(true)
    tl_out.e.ready := Bool(true)
  }
}

trait HasPeripheryMyBusMasterModuleImp extends LazyModuleImp {
  val outer: HasPeripheryMyBusMaster
}
```

Further on up the stack (in Platform.scala and System.scala) we expose the busmaster_a/d_io bundles to the outside world.

The FIRRTL module generated from one of these modules (we have a couple) looks very basic:

```
  module MyBusMaster :
    input clock : Clock
    input reset : UInt<1>
    output auto : {out : {a : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<3>, size : UInt<3>, source : UInt<1>, address : UInt<32>, mask : UInt<4>, data : UInt<32>, corrupt : UInt<1>}}, flip b : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<2>, size : UInt<3>, source : UInt<1>, address : UInt<32>, mask : UInt<4>, data : UInt<32>, corrupt : UInt<1>}}, c : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<3>, size : UInt<3>, source : UInt<1>, address : UInt<32>, data : UInt<32>, corrupt : UInt<1>}}, flip d : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<2>, size : UInt<3>, source : UInt<1>, sink : UInt<1>, denied : UInt<1>, data : UInt<32>, corrupt : UInt<1>}}, e : {flip ready : UInt<1>, valid : UInt<1>, bits : {sink : UInt<1>}}}}
    input busmaster_a_io : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<20>, address : UInt<32>, mask : UInt<4>, data : UInt<32>, corrupt : UInt<1>}}
    output busmaster_d_io : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<20>, sink : UInt<20>, denied : UInt<1>, data : UInt<32>, corrupt : UInt<1>}}

    clock is invalid
    reset is invalid
    auto is invalid
    busmaster_a_io is invalid
    busmaster_d_io is invalid
    wire tl_out : {a : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<3>, size : UInt<3>, source : UInt<1>, address : UInt<32>, mask : UInt<4>, data : UInt<32>, corrupt : UInt<1>}}, flip b : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<2>, size : UInt<3>, source : UInt<1>, address : UInt<32>, mask : UInt<4>, data : UInt<32>, corrupt : UInt<1>}}, c : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<3>, size : UInt<3>, source : UInt<1>, address : UInt<32>, data : UInt<32>, corrupt : UInt<1>}}, flip d : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<2>, size : UInt<3>, source : UInt<1>, sink : UInt<1>, denied : UInt<1>, data : UInt<32>, corrupt : UInt<1>}}, e : {flip ready : UInt<1>, valid : UInt<1>, bits : {sink : UInt<1>}}} @[Nodes.scala 332:76]
    tl_out is invalid @[Nodes.scala 332:76]
[... 42 lines cut ...]
   3  0.82%  2.68%      63 304549 firrtl.Utils$.$anonfun$get_flip$2
   4  0.47%  3.15%      36 304547 scala.collection.TraversableOnce.nonEmpty
   5  0.39%  3.54%      30 304195 scala.collection.TraversableOnce.nonEmpty
   6  0.27%  3.82%      21 302501 firrtl.passes.InferWidths$.$anonfun$run$9
   7  0.26%  4.08%      20 308043 java.io.FileOutputStream.close0
   8  0.25%  4.32%      19 307061 firrtl.transforms.ConstantPropagation.constPropPrim
   9  0.23%  4.56%      18 305884 scala.collection.TraversableOnce.nonEmpty
  10  0.23%  4.79%      18 305894 scala.collection.AbstractTraversable.<init>
  11  0.23%  5.03%      18 307977 java.security.AccessController.doPrivileged
  12  0.22%  5.25%      17 306785 firrtl.transforms.RemoveWires.onStmt$1
  13  0.20%  5.44%      15 302507 firrtl.passes.InferWidths$.get_constraints_s$1
  14  0.20%  5.64%      15 304193 scala.collection.TraversableOnce.nonEmpty
  15  0.20%  5.83%      15 304225 firrtl.Utils$.$anonfun$get_flip$2
  16  0.20%  6.03%      15 307048 scala.collection.mutable.HashTable.resize
  17  0.20%  6.22%      15 307072 firrtl.Mappers$ExprMagnet$$anon$6.map
  18  0.20%  6.42%      15 307080 scala.collection.mutable.HashTable.findOrAddEntry
  19  0.18%  6.60%      14 304551 scala.collection.TraversableOnce.nonEmpty
  20  0.18%  6.78%      14 305890 scala.collection.AbstractTraversable.<init>
```

If I'm reading this correctly, it indicates that the problematic module causes a dramatic increase in the number of calls to a handful of functions, dwarfing the amount of CPU spent on any other single thing in a normal run.

At the very least I thought this might be of interest to the FIRRT devs.

I'm going to try rewriting the master module to avoid these runtime problems, and if anyone could suggest a different way (which hopefully doesn't trip the same runtime blowout) that'd be appreciated.

azidar, 2018-11-07:
Thanks for the breakdown - actually, we merged a pull request (#928 ) which I believe should have fixed your issue?

juliusbaxter, 2018-11-07:
Thanks @azidar it does reduce the runtime to about 2mm40s now, which is a good improvement, 
but is still double than without the module, which took 1m20s on my run just then.

The profiling output looks fairly different for the same input FIRRTL (after updating master, of course):

With module (2m40s):

> CPU SAMPLES BEGIN (total = 18730) Wed Nov  7 05:56:23 2018
> rank   self  accum   count trace method
>    1  1.43%  1.43%     268 311123 scala.collection.IndexedSeqOptimized.prefixLengthImpl
>    2  1.32%  2.75%     247 311144 java.util.regex.Pattern$5.isSatisfiedBy
>    3  1.25%  4.00%     235 304509 scala.collection.TraversableOnce.nonEmpty
>    4  1.20%  5.21%     225 311125 java.util.regex.Pattern$5.isSatisfiedBy
>    5  1.20%  6.40%     224 311124 scala.collection.IndexedSeqOptimized.prefixLengthImpl
>    6  1.16%  7.57%     218 311120 java.util.regex.Pattern$5.isSatisfiedBy
>    7  1.11%  8.68%     208 304495 scala.collection.TraversableOnce.nonEmpty
>    8  1.11%  9.79%     208 304520 scala.collection.TraversableOnce.nonEmpty
>    9  1.09% 10.88%     204 311116 java.util.regex.Pattern$5.isSatisfiedBy
>   10  1.02% 11.90%     191 304533 scala.collection.TraversableOnce.nonEmpty
>   11  0.74% 12.64%     139 304532 scala.collection.AbstractTraversable.<init>
>   12  0.68% 13.32%     127 304512 scala.collection.AbstractTraversable.<init>
>   13  0.68% 13.99%     127 311582 java.security.AccessController.doPrivileged
>   14  0.60% 14.59%     112 304494 scala.collection.AbstractTraversable.<init>
>   15  0.59% 15.18%     111 304521 scala.collection.AbstractTraversable.<init>
>   16  0.53% 15.72%     100 311592 java.security.AccessController.doPrivileged
>   17  0.34% 16.06%      64 311145 java.util.regex.Pattern$CharProperty.match
>   18  0.34% 16.40%      63 311131 java.util.regex.Pattern$CharProperty.match
>   19  0.28% 16.67%      52 311132 java.util.regex.Pattern$CharProperty.match
>   20  0.25% 16.92%      47 311146 scala.collection.IndexedSeqOptimized.segmentLength

Profile without module (1m20s):

> CPU SAMPLES BEGIN (total = 10168) Wed Nov  7 06:04:51 2018
> rank   self  accum   count trace method
>    1  2.24%  2.24%     228 308361 java.util.regex.Pattern$5.isSatisfiedBy
>    2  2.15%  4.40%     219 308367 java.util.regex.Pattern$5.isSatisfiedBy
>    3  1.97%  6.36%     200 308370 java.util.regex.Pattern$5.isSatisfiedBy
>    4  1.93%  8.29%     196 308359 java.util.regex.Pattern$5.isSatisfiedBy
>    5  1.66%  9.95%     169 308382 scala.collection.IndexedSeqOptimized.prefixLengthImpl
>    6  1.47% 11.42%     149 308399 scala.collection.IndexedSeqOptimized.prefixLengthImpl
>    7  0.91% 12.33%      93 308387 scala.collection.IndexedSeqOptimized.segmentLength
>    8  0.79% 13.12%      80 308362 scala.collection.IndexedSeqOptimized.segmentLength
>    9  0.62% 13.74%      63 308799 java.security.AccessController.doPrivileged
>   10  0.55% 14.29%      56 308791 java.security.AccessController.doPrivileged
>   11  0.43% 14.72%      44 308388 java.util.regex.Pattern$CharProperty.match
>   12  0.34% 15.07%      35 308384 java.util.regex.Pattern$CharProperty.match
>   13  0.33% 15.40%      34 307464 firrtl.transforms.ConstantPropagation.constPropPrim
>   14  0.33% 15.74%      34 308402 java.util.regex.Pattern$CharProperty.match
>   15  0.29% 16.02%      29 308404 java.util.regex.Pattern$5.isSatisfiedBy
>   16  0.27% 16.29%      27 308391 java.util.regex.Pattern$CharProperty.match
>   17  0.25% 16.53%      25 308357 scala.collection.generic.Growable.$anonfun$$plus$plus$eq$1
>   18  0.24% 16.77%      24 308397 java.util.regex.Pattern$5.isSatisfiedBy
>   19  0.21% 16.97%      21 308386 scala.collection.generic.Growable.$anonfun$$plus$plus$eq$1
>   20  0.20% 17.17%      20 308433 java.util.regex.Pattern$5.isSatisfiedBy

Given this was zero effort, I think I'll take it and not worry about rewriting my module.

Thanks again!

closed 2018-11-07 by azidar

jackkoenig, 2018-11-07:
The slowdown is still unfortunate. We are experiencing fairly long runtimes for large designs (~10min) so I can truly say I feel your pain @juliusbaxter. I hope to do some more performance improving in the nearish future, stay tuned!

jackkoenig, 2018-11-26:
@juliusbaxter per the improved profiling results, I believe the regex runtime should have been drastically improved by https://github.com/freechipsproject/firrtl/pull/942, have you bumped FIRRTL in the last 10 days?

juliusbaxter, 2018-11-27:
I just pulled the latest and re-ran a couple of times, it looks like the runtime is about the same as before. Profile before getting the latest Chisel (HEAD at 00ce759):

```
   1  1.08%  1.08%     211 311461 scala.collection.IndexedSeqOptimized.prefixLengthImpl
   2  1.04%  2.12%     202 311468 java.util.regex.Pattern$5.isSatisfiedBy
   3  1.00%  3.12%     196 311478 java.util.regex.Pattern$5.isSatisfiedBy
   4  0.99%  4.11%     193 304740 scala.collection.TraversableOnce.nonEmpty
   5  0.98%  5.09%     191 304728 scala.collection.TraversableOnce.nonEmpty
   6  0.94%  6.03%     184 311469 java.util.regex.Pattern$5.isSatisfiedBy
   7  0.94%  6.97%     183 304714 scala.collection.TraversableOnce.nonEmpty
   8  0.93%  7.90%     182 311462 scala.collection.IndexedSeqOptimized.prefixLengthImpl
   9  0.92%  8.82%     180 304721 scala.collection.TraversableOnce.nonEmpty
  10  0.90%  9.73%     176 311471 java.util.regex.Pattern$5.isSatisfiedBy
  11  0.55% 10.27%     107 304713 scala.collection.AbstractIterable.<init>
  12  0.54% 10.81%     105 311927 java.security.AccessController.doPrivileged
  13  0.48% 11.29%      94 311920 java.security.AccessController.doPrivileged
  14  0.48% 11.77%      93 304722 scala.collection.AbstractIterable.<init>
  15  0.43% 12.20%      83 304739 scala.collection.AbstractIterable.<init>
  16  0.42% 12.62%      82 304730 scala.collection.AbstractIterable.<init>
  17  0.33% 12.94%      64 310171 scala.collection.mutable.HashTable.resize
  18  0.29% 13.23%      56 311483 java.util.regex.Pattern$CharProperty.match
  19  0.25% 13.48%      49 311475 java.util.regex.Pattern$CharProperty.match
  20  0.24% 13.72%      46 311464 java.util.regex.Pattern$CharProperty.match
```

and after pulling latest (HEAD at ab95104):

```
   1  1.95%  1.95%     328 304472 scala.collection.generic.Growable.loop$1
   2  1.95%  3.90%     327 304479 scala.collection.generic.Growable.loop$1
   3  1.94%  5.84%     325 304483 scala.collection.generic.Growable.loop$1
   4  1.86%  7.70%     313 304488 scala.collection.generic.Growable.loop$1
   5  1.38%  9.07%     231 304490 scala.collection.TraversableOnce.nonEmpty
   6  1.29% 10.36%     216 304480 scala.collection.TraversableOnce.nonEmpty
   7  1.27% 11.64%     214 310531 java.util.regex.Pattern$5.isSatisfiedBy
   8  1.27% 12.90%     213 304484 scala.collection.TraversableOnce.nonEmpty
   9  1.26% 14.16%     211 304473 scala.collection.TraversableOnce.nonEmpty
  10  1.13% 15.29%     190 310542 scala.collection.IndexedSeqOptimized.prefixLengthImpl
  11  1.07% 16.36%     180 310551 java.util.regex.Pattern$5.isSatisfiedBy
  12  1.02% 17.38%     171 310533 java.util.regex.Pattern$5.isSatisfiedBy
  13  0.96% 18.34%     161 310545 java.util.regex.Pattern$5.isSatisfiedBy
  14  0.93% 19.27%     157 310538 scala.collection.IndexedSeqOptimized.prefixLengthImpl
  15  0.61% 19.88%     102 310562 scala.collection.IndexedSeqOptimized.segmentLength
  16  0.59% 20.47%      99 310543 scala.collection.IndexedSeqOptimized.segmentLength
  17  0.44% 20.91%      74 310968 java.security.AccessController.doPrivileged
  18  0.42% 21.34%      71 310977 java.security.AccessController.doPrivileged
  19  0.33% 21.66%      55 310535 java.util.regex.Pattern$CharProperty.match
  20  0.25% 21.91%      42 310549 java.util.regex.Pattern$CharProperty.match
```

Certainly the regex stuff has dropped down the list, so it's progress, but I guess my module tickles a different part of FIRRTL into action which is still dominating the runtime we're experiencing.

Anyway keep up the good work! Performance improvements always welcome ;)

jackkoenig, 2018-11-27:
Ahh bummer that it's not a bigger change. Well we'll keep trying to speed things up!