chipsalliance/firrtl #935: Interval subtraction problem

chick, 2018-11-07:
## Bug Report
Using intervals with subtraction gets wrong result with verilog
This was found while testing PR #870

## Example
The following chisel3 test
```scala
class IntervalChainedSubTester extends BasicTester {
  val intervalResult = Wire(Interval())
  intervalResult := 17.I - 2.I // -& 2.I -& 2.I -& 2.I -& 2.I -& 2.I
  printf("Interval result: %d\n", intervalResult.asUInt)
  assert(intervalResult === 15.I)
  stop()
}
```
generates the following Chirrtl
```
;buildInfoPackage: chisel3, version: 3.2-SNAPSHOT, scalaVersion: 2.12.6, sbtVersion: 1.1.1
circuit IntervalChainedSubTester : 
  module IntervalChainedSubTester : 
    input clock : Clock
    input reset : UInt<1>
    output io : {}
    
    wire intervalResult : Interval[?, ?] @[IntervalSpec.scala 336:28]
    node _T = sub(asInterval(UInt<6>("h011"), 17, 17, 0), asInterval(UInt<3>("h02"), 2, 2, 0)) @[IntervalSpec.scala 337:26]
    intervalResult <= _T @[IntervalSpec.scala 337:18]
    node _T_1 = asUInt(intervalResult) @[IntervalSpec.scala 338:50]
    node _T_2 = bits(reset, 0, 0) @[IntervalSpec.scala 338:9]
    node _T_3 = eq(_T_2, UInt<1>("h00")) @[IntervalSpec.scala 338:9]
    when _T_3 : @[IntervalSpec.scala 338:9]
      printf(clock, UInt<1>(1), "Interval result: %d\n", _T_1) @[IntervalSpec.scala 338:9]
      skip @[IntervalSpec.scala 338:9]
    node _T_4 = eq(intervalResult, asInterval(UInt<5>("h0f"), 15, 15, 0)) @[IntervalSpec.scala 339:25]
    node _T_5 = bits(reset, 0, 0) @[IntervalSpec.scala 339:9]
    node _T_6 = or(_T_4, _T_5) @[IntervalSpec.scala 339:9]
    node _T_7 = eq(_T_6, UInt<1>("h00")) @[IntervalSpec.scala 339:9]
    when _T_7 : @[IntervalSpec.scala 339:9]
      printf(clock, UInt<1>(1), "Assertion failed\n    at IntervalSpec.scala:339 assert(intervalResult === 15.I)\n") @[IntervalSpec.scala 339:9]
      stop(clock, UInt<1>(1), 1) @[IntervalSpec.scala 339:9]
      skip @[IntervalSpec.scala 339:9]
    node _T_8 = bits(reset, 0, 0) @[IntervalSpec.scala 340:7]
    node _T_9 = eq(_T_8, UInt<1>("h00")) @[IntervalSpec.scala 340:7]
    when _T_9 : @[IntervalSpec.scala 340:7]
      stop(clock, UInt<1>(1), 0) @[IntervalSpec.scala 340:7]
      skip @[IntervalSpec.scala 340:7]
``` 
This fails when run with verilator through the chisel basic tester
If one takes the above firrtl and run it with treadle repl then it works and stops correctly

## Speculation
There maybe a problem emitting verilog from intervals.

## Impact
We can't deploy intervals until this gets fixed.

azidar, 2018-11-08:
For reference, here is the generated Verilog:

```
module IntervalChainedSubTester(
  input   clock,
  input   reset
);
  wire [6:0] _GEN_0; // @[IntervalSpec.scala 337:26 IntervalSpec.scala 337:26]
  wire [4:0] _GEN_1; // @[IntervalSpec.scala 337:26 IntervalSpec.scala 337:26]
  wire [4:0] intervalResult; // @[IntervalSpec.scala 337:26 IntervalSpec.scala 337:26]
  wire [4:0] _T_1; // @[IntervalSpec.scala 338:50]
  wire  _T_3; // @[IntervalSpec.scala 338:9]
  wire  _T_4; // @[IntervalSpec.scala 339:25]
  wire  _T_6; // @[IntervalSpec.scala 339:9]
  wire  _T_7; // @[IntervalSpec.scala 339:9]
  assign _GEN_0 = $signed(6'sh11) - $signed(6'sh2); // @[IntervalSpec.scala 337:26 IntervalSpec.scala 337:26]
  assign _GEN_1 = _GEN_0[4:0]; // @[IntervalSpec.scala 337:26 IntervalSpec.scala 337:26]
  assign intervalResult = $signed(_GEN_1); // @[IntervalSpec.scala 337:26 IntervalSpec.scala 337:26]
  assign _T_1 = $unsigned(intervalResult); // @[IntervalSpec.scala 338:50]
  assign _T_3 = reset == 1'h0; // @[IntervalSpec.scala 338:9]
  assign _T_4 = $signed(intervalResult) == $signed(5'shf); // @[IntervalSpec.scala 339:25]
  assign _T_6 = _T_4 | reset; // @[IntervalSpec.scala 339:9]
  assign _T_7 = _T_6 == 1'h0; // @[IntervalSpec.scala 339:9]
  always @(posedge clock) begin
    `ifndef SYNTHESIS
    `ifdef PRINTF_COND
      if (`PRINTF_COND) begin
    `endif
        if (_T_3) begin
          $fwrite(32'h80000002,"Interval result: %d",_T_1); // @[IntervalSpec.scala 338:9]
        end
    `ifdef PRINTF_COND
      end
    `endif
    `endif // SYNTHESIS
    `ifndef SYNTHESIS
    `ifdef PRINTF_COND
      if (`PRINTF_COND) begin
    `endif
        if (_T_7) begin
          $fwrite(32'h80000002,"Assertion failed    at IntervalSpec.scala:339 assert(intervalResult === 15.I)"); // @[IntervalSpec.scala 339:9]
        end
    `ifdef PRINTF_COND
      end
    `endif
    `endif // SYNTHESIS
    `ifndef SYNTHESIS
    `ifdef STOP_COND
      if (`STOP_COND) begin
    `endif
        if (_T_7) begin
          $fatal; // @[IntervalSpec.scala 339:9]
        end
    `ifdef STOP_COND
      end
    `endif
    `endif // SYNTHESIS
    `ifndef SYNTHESIS
    `ifdef STOP_COND
      if (`STOP_COND) begin
    `endif
        if (_T_3) begin
          $finish; // @[IntervalSpec.scala 340:7]
        end
    `ifdef STOP_COND
      end
    `endif
    `endif // SYNTHESIS
  end
endmodule
```

azidar, 2018-11-08:
Do you have a verilator waveform?

chick, 2018-11-08:
dump.vcd
```
$version Generated by VerilatedVcd $end
$date Wed Nov  7 14:13:02 2018
 $end
$timescale   1ns $end

 $scope module TOP $end
  $var wire  1 # clock $end
  $var wire  1 $ reset $end
  $scope module IntervalChainedSubTester $end
   $var wire  1 # clock $end
   $var wire  3 % intervalResult [2:0] $end
   $var wire  1 $ reset $end
  $upscope $end
 $upscope $end
$enddefinitions $end

#0
0#
1$
b111 %
#1
1#
#2
#3
#4
#5
#6
0#
#7
#8
#9
#10
```

closed 2018-12-10 by azidar