chipsalliance/firrtl #939: Translate increments of dummy iterators to Galois shift registers

nhynes, 2018-11-16:
**Type of issue**: feature request

**What is the use case for changing the behavior?** Hardware efficiency

**Impact**: no functional change

**Development Phase**: request

Is there any work on translating counters whose values are not used as literals (i.e. dummy loop vars), to Galois shift registers? Apparently the implementation is more efficient than using an adder:
[image]
[Source: section 5 of *A Memory Encryption Engine Suitable for General Purpose Processors*](https://eprint.iacr.org/2016/204.pdf)

Perhaps it would even be better to do this at the Chisel level where more semantic information is available?

seldridge, 2018-11-16:
This is an interesting idea... It's likely easier to do it at the Chisel level and hide it behind the `Counter` object.

I'm extrapolating here, but I think what you're saying is: Use an LFSR in place of a counter. If a user wants to count up to some value N and fire, then mangle the value that the user is counting to to be N places in the future from some starting value of the LFSR (say 1). Force the LFSR to reset when it hits the value the user wants.

Seems doable, but as you mention, the user can't be using the counter value directly. Perhaps this could be the underlying implementation behind a new `Counter` API that only returns a `Bool`.

aswaterman, 2018-11-16:
Yeah. This is something I’ve thought about doing, but it has always been low priority for me because most counters where you don’t care about the underlying representation are small, so don’t need carry-lookahead. (This technique is still beneficial in that case, just not significantly so.)